clough42/electronic-leadscrew #90: Lock while spindle is running

C4fishfood, 2020-09-27:
My setup must have a bit of electrical noise- I have had a few cases where the thread pitch setting changed on its own, or it switched from threading mode to feed mode. It is intermittent, but only seems to happed when the machine (lathe spindle) is running and ELS is in threading mode. Is there a way to add a lock so that the settings cannot be switches while the RPM is greater than 0? I'm not the greatest coder, and would be happy to try to modify to code to make this happen, but would appreciate any advice on how to pull this off.

Hobby-Lofty, 2020-10-04:
hi, I had a similar problem with EMI, mine sometimes changed settings but the spindle was at 0.
I fitted the launchpad and Led&Key module in the headstock of the lathe.
It did suffer from switching noise, occasionally getting its knickers in a twist
when threading with the half nuts engauged all the time, switching from reverse to stop 
via the forward -off- reverse switch, rarely it would change all the settings, from Forward to reverse, thread to feed, metric to TPI.
there is a 240v fan built into the the 3ph motor, this fan is switched via the forward -off- reverse switch so I changed this so it has its own switch
and I can leave it on all the time when when switching from reverse to stop,
While I was curing the swithing noise I also :
Added a EMI Filter to the 240v supply to the lathe,
(Yunpen YE10T1L2 Two Stage Filter 4 x 1.2 mH 250V AC 10A )
3 x clip on ferrite beads to the wires from the VFD to the motor
Used shielded cable from:
Launchpad to Led&keys to buttons
Lauchpad to stepper driver 
Added a few clip on ferirte beads to the 5v wall wart
Check out the video's James made when he fitted a 3ph motor and VFD to his lathe.
Curing EMI Noise:   https://youtu.be/DJfiOqaeFDg

C4fishfood, 2020-10-04:
Hi Lofty,
Thank you for responding. I also fitted the launchpad and Led&Key module in the headstock. I was only getting the problem of EMI while the motor was running, so it made it a bit easier for me to address the problem with code rather than shielding/filtering (I already was using properly grounded shielded cables but had not put chokes on them).

I modified the code in the UserInterface.cpp file to have an extra "if" statement within each of "if" statements that check for button presses, and only allows key press to be passed it the spindle is stopped:
        if(core->getRPM() == 0)
        { 

This doesn't effect the functionality of the lathe for the way that I use it. Changing the feed rate while cutting *might* be nice, but I absolutely do not want changes made while threading:).

Hobby-Lofty, 2020-10-04:
Very nice, I'll add this when I do my next update, better safe than sorry.
Cheers
Lofty

C4fishfood, 2020-10-04:
Out of curiosity, are you in NZ or Aus?

Hobby-Lofty, 2020-10-04:
I wish.
Wiltshire in the UK
Lofty

clough42, 2020-10-05:
I'm finally getting back to the code and incorporating some of the ideas people have submitted.  I think this one is good and I'll be adding it into the code soon.  I had been planning on doing something much more exotic to handle changes while the spindle was running, like gradually ramping the speed or slowing to a stop before reversing, but that was to solve the problem of inertia overwhelming the stepper or servo.

A few people have been talking about struggling with noise, and I think in general, this change could fix both the noise problems and avoid the sudden changes at the same time.

I think losing the ability to change the speed while the spindle is running is acceptable, since that's typical with a real gearbox anyway.

gusosborne, 2020-10-05:
Perhaps this could be an optional flag..?
I do rather like being able to dial in on a perfect feed rate..

C4fishfood, 2020-10-05:
The way I implemented it was to have an "if" check on the RPM for each of the bottoms- not very efficient in terms of coding but it is effective. Certainly there is a better way... but it also allows you to let there be some button functions that still work while the spindle is turning (if you wish). I'm not certain, but would imagine you could change the feed table if you wanted and not have that cause problems if it was in threading mode (which is where a part will be scrapped if the emi does come up).

btw, James, thank you very much for developing this- very awesome

clough42, 2020-10-06:
I just put a few more changes into the dev branch that I hope will address most of the reliability issues people have been seeing.  Here are the changes:

- Fix overflow that caused problems with large metric threads (thanks @hugosman95)
- Slow down the SPI bus from 250Kbps to 98Kbps
- Increase SPI timing delays slightly
- Aggressively filter bad key reads:
  - Require multiple, identical, consecutive reads before acting
  - Validate that only one key is pressed at a time (thanks @ngreenou)
  - Require all keys to be released between key presses
- Do not allow changes to mode or direction while the machine is in motion (thanks @C4fishfood)
- Add configuration option to ignore all keypresses when the machine is in motion (thanks @C4fishfood)

You can either pull the dev branch or you can download a zip of the code with these fixes here:

  https://github.com/clough42/electronic-leadscrew/archive/dev.zip

If you try this, please report back.  I would like some positive feedback before I mark it as a global release.

clough42, 2020-10-07:
I just made a new release (v1.2.00) that likely fixes this issue.  Please try it and let me know if it helps.

https://github.com/clough42/electronic-leadscrew/releases/tag/v1.2.00

Hobby-Lofty, 2020-10-07:
I posted this uder the wrong heading earliar. Hi James, I have tried V1.2.00, the mode buttons can't be changed when the spindle is running, the up and down buttons still work in thread and feed mode when the spindle is running.
Should the up and down buttons be turned off for thread mode or is this something I have to configure?
Cheers Lofty

clough42, 2020-10-07:
The default is to allow pitch/feed changes while the spindle is running.  I often change the feed rate to tune chip formation, even in the middle of a cut.  There is a new define in Configuration.h you can uncomment to lock all buttons while the spindle is turning.

You raise a good point about locking the pitch while threading.  I can't think of a good reason to need to do that.  I would need to add a little more state data to make that happen, but it could be useful.

My hope is that the false button reads are now fixed with aggressive validation, so the real benefit of the lock is to prevent user error.

clough42, 2020-10-11:
This is now resolved, so I'm closing this issue.

closed 2020-10-11 by clough42