clough42/electronic-leadscrew #91: Lathes with 2 Leadscrews

robotwizard, 2020-10-01:
[image]

Planning an electronic leadscrew for my PM1127 lathe, but ran into a big obstacle. This lathe has an 8TPI leadscrew for threading and a separate drive shaft for feed. The problem is that while the leadscrew moves the carriage .125"  per revolution for threading, the feed rod only moves the carriage approx .080".  I looked at config.h , but don't see anything that would allow separate rates for threading and feed. Has anyone made this work for lathes with 2 leadscrews?

Blue35mm, 2020-10-01:
Hello,
Just use your lead screw for both. You would change the feed on the ELS to get the feed you want for the carriage. Omit using the feed shaft.

robotwizard, 2020-10-01:
Thanks for the reply. That is an idea, except the leadscrew is only engaged when the half nuts are engaged and would wear the half nuts out prematurely.... would also lose the power feed to the cross slide.

clough42, 2020-10-01:
Has anyone considered or tried using the ELS to thread using the feed rod
and gearing on a lathe so equipped?  I don't have one, so I don't know if
this is practical, if the rack is accurate enough, if the RPMs are
reasonable, etc.

Blue35mm, 2020-10-01:
One thought might be to put a second pully on the drive shaft ( Could do this where it , and probably the lead screw are driven from) then drive both from the leadscrew motor. Do you have a picture of the end of the drive box?

matthew-gd, 2020-10-01:
This lathe looks similar to mine. If that is the case then the gearbox in the lower left of your picture is driven by a single shaft on the left side (out of the picture) where I mounted a stepper motor. The stepper motor is driven by stepper driver module (from eBay) which is controlled either by an ELS (Thank you, John) or via a changeover switch connecting to an Arduino configured as a variable frequency source of pulses. The cross slide is also driven by a stepper motor which can be connected to the ELS or disconnected electrically. 

To turn a thread I connect both stepper motors to the ELS and use the thread leadscrew on the lathe with the 'C' setting on the gearbox. 

To turn manually with a powered feed then I disconnect the cross slide motor driver and use the Arduino to drive the gearbox stepper motor and configure the gearbox to use the normal lead screw. From the lever on the carriage either the carriage movement or cross slide movement can be selected.

Zymurgy42, 2020-10-02:
See: https://github.com/clough42/electronic-leadscrew/pull/43 

This would seem to cover the issue.

Hobby-Lofty, 2020-10-04:
Hello, My lathe, a Warco 280VF 3mm pitch leadscrew also has a leadscrew for cutting threads and the power feed lower slotted shaft used for turning and facing, I leave the gearbox feed rate selector on the A setting (1:1) for turning and treading. For threading, the threading and turning selector (pictured above) is moved to the Left, this engages the leadsrew and you select the thread required on the ELS, you then use the half nuts and thread dial or leave the half nuts engaged stopping and reversing the lathe.
For turning move threading and turning selector to the Right and select the feed rate you want on the ELS, I can change the feed rate up and down whilst turning on the fly. I have successfully turned large lumps of metal into chips and cut metric and inch threads. I am very happy with my setup. 
[https://youtu.be/ooxJIqEBvXU](url)
hope this helps
Lofty

robotwizard, 2020-10-05:
There is only 1 input shaft and a selector on the gearbox to switch between thread leadscrew and feed rod. Input side is very similar to G0602 and the shaft is even 14mm except the top shaft is on banjo and banjo is longer. The software mods shared by ttabbal should work I think, with a few minor adjustments to source code in configuration.h and tables.cpp. His machine is same model # but newer version... my leadscrew pitch is 8TPI instead of 10TPI and I measure slightly different gear ratios in box. I have 1:1 in the 'A' position which should work well. The spindle has a 40 tooth gear and I have mounted a 60 tooth gear on the banjo as you can see in the photo. Ordered a 60 tooth 14mm bore GT2 pulley to go on intermediate shaft with the gear and a 40 tooth 6mm bore pulley to go on the encoder. The encoder will mount lower on the banjo. Planning to mount the hybrid servo to the left of the input shaft. Will have to drill a hole in the cover and it will stick out slightly, but I'm ok with that as there aren't many options to mount the motor.

[image]

[image]

[image]

[image]

Hobby-Lofty, 2020-10-06:
I just did a quick eyeball measurement on the metric power feed rate and found that the ELS set at 0,05mm the carriage moved 0.035mm and at 1mm the carriage moved 0.62mm, the carriage moves a bit slower than the ELS setting. For my use this works ok, there are so many variables anyway, type of material, spindle speed, turning tool type, HSS, carbide, tool nose radius..., I go more on the sound the cut makes and surface finish. as long as I can cut accurate threads and don't have to use change gears I can live with it.
Good luck with your project.
PS with mine another variable, if you use a closed loop stepper system you may find when taking a too heavy a cut in tough metal the single main drive belt will slip and grip again, the sudden change will cause the stepper driver to go into alarm state.
Maybe James could add 43 into his code.
Lofty

robotwizard, 2020-10-07:
@Hobby-Lofty, if you use the modified source code for dual leadscrews created by ttabbal you should be able to change the parameters to work on your machine, then recompile, and reinstall on your controller . Can't give you any advice about changing the parameters in the source code tho, as I haven't got that far yet.

clough42, 2020-10-07:
Merging #43 is on my list.

I wanted to tackle the random keyboard read issues first, though, so this went on the back burner.

Hobby-Lofty, 2020-10-07:
Hi James, I have tried V1.2.00, the mode buttons can't be changed when the spindle is running, the up and down buttons still work in thread and feed mode when the spindle is running. 
Should the up and down buttons be turned off for thread mode or is this something I have to configure?
Cheers Lofty

clough42, 2020-10-11:
I just merged this functionality to the dev branch.  It will be in release 1.3.01.

clough42, 2020-10-12:
1.3.01 is now released.

closed 2020-10-12 by clough42

robotwizard, 2020-10-12:
Thanks James! I know you're busy, really appreciate you taking the time to do this

 
 

Closed #91.