clough42/electronic-leadscrew #92: Leadscrew flips randomly into reverse

PaulJ25, 2020-10-06:
I have an issue the motor runs fine for about 20mins then flips into reverse or stalls due to flipping into reverse anyone else had this issue

PaulJ25, 2020-10-06:
Thanks for the quick reply all cables are short, my 5v does not have an earth as it's 5v DC UK plug, I will try the dev branch to see if it makes a difference.

clough42, 2020-10-06:
What kind of cable are you using for the control panel?

I'm using unshielded CAT5e network cable (stranded).  I have it wired in
pairs, so every pair has one logic ground and one of the other signals (+5,
STB, CLK, DIO).  This has been pretty bulletproof for me.

Even with the 5V DC plug packs, I've seen people intentionally connect the
5V ground to the 12/24/48V ground of the stepper system, thinking it's
needed.  That's the surest way to get into trouble with noise.

James

PaulJ25, 2020-10-06:
Please can you explain the wiring logic ground as I have not heard this term before I am currently using cat 6 cable but have cat5e available any chance of a wiring diagram?

clough42, 2020-10-06:
The 5V power supply has a + and a -.  The + wire is the 5V supply.  The -
wire is the "logic ground".

I use the term "logic ground" to differentiate it from the "motor ground"
which is the - side of the power supply for the stepper or servo motor.

The logic ground and the motor ground should be kept separate, since the
motor power is very noisy, and you want the 5V power for the electronics to
be quiet and stable.

The stepper or servo driver has optical isolators on the input, so the two
circuits do not need a common ground (-) to function.

PaulJ25, 2020-10-06:
So out of the 8 cables from the cat5e 4 would to ground at each end 4 connected as a single to the gnd on the els board and the 4 the other end as one connected to the ground on the display board would this be right sorry to be an inconvenience but have been struggling with this issue for 3 months

clough42, 2020-10-06:
There are four "pairs" in the cable.  Typically they are:

- white/orange and orange/white
- white/blue and blue/white
- white/green and green/white
- white/brown and brown/white

I would tie all of the white wires together at the LaunchPad end and connect them all to GND, and then use the four colored wires for the other signals (+5, STB, CLK, DIO).

At the control panel end, you can just use one of the white wires for GND.  There's no need to connect them all at that end.

clough42, 2020-10-06:
I just put a few more changes into the dev branch that I hope will address most of the reliability issues people have been seeing.  Here are the changes:

- Fix overflow that caused problems with large metric threads (thanks @hugosman95)
- Slow down the SPI bus from 250Kbps to 98Kbps
- Increase SPI timing delays slightly
- Aggressively filter bad key reads:
  - Require multiple, identical, consecutive reads before acting
  - Validate that only one key is pressed at a time (thanks @ngreenou)
  - Require all keys to be released between key presses
- Do not allow changes to mode or direction while the machine is in motion (thanks @C4fishfood)
- Add configuration option to ignore all keypresses when the machine is in motion (thanks @C4fishfood)

You can either pull the dev branch or you can download a zip of the code with these fixes here:

  https://github.com/clough42/electronic-leadscrew/archive/dev.zip

If you try this, please report back.  I would like some positive feedback before I mark it as a global release.

clough42, 2020-10-07:
I just made a new release (v1.2.00) that likely fixes this issue.  Please try it and let me know if it helps.

https://github.com/clough42/electronic-leadscrew/releases/tag/v1.2.00

PaulJ25, 2020-10-10:
Hi I tried the lead as you suggested this made the issue worse so I got a cat 6 shielded cable I put 2 of the 4 white wires to Gnd along with the shielding on both ends with your updated firmware it seems to have fixed the issue I will continue to test and update if anything changes thank you for your help with this issue.

clough42, 2020-10-11:
Thank you.  I'll mark this issue as closed.

closed 2020-10-11 by clough42