clough42/electronic-leadscrew #100: Default metric/ imperial at power on

asm308, 2020-10-18:
A little quality of life suggestion. I've modified my setup to start up in metric mode as I have. Metric lathe, but the firmware could determine this based upon whether a metric or imperial leadscrew is defined in the configuration.h

J642, 2020-10-28:
How did you modified so you start in metric

asm308, 2020-10-28:
Hi 
If you look in the UserInterface.cop file at around line 64 you will see a line that says
this->metric=false
Change false true and recompile.
If I was doing it for a general solution I’d have set it depending on whether LEADSCREW_HMM was defined, but being a bit lazy

clough42, 2020-11-01:
This is on my list of features to add.  Though rather than making it configurable, I was planning on making the ELS remember its settings in the EEPROM flash so if you set it to metric, it'll still be in metric next time you turn it on.

asm308, 2020-11-01:
Yeah that would be a better solution. I’ll keep an eye out for it.  Thanks