clough42/electronic-leadscrew #101: Metric lathe with 2mm pitch leadscrew has two different gears for dial indicator

asm308, 2020-10-18:
My mini lathe has two thread indicator gears and I have to change the gear for some thread pitches.  Currently I’ve put in a little mod to display a symbol in the spare digit position of the pitch part of the display. This reminds me if I need the 40 or the 42 tooth gear for the dial indicator.

I don’t suppose there is a way the els could deal with this so I have a totally gear change less@ els?

0.4, 0.5, 0.8, 1.0, 1.25, 2.0, 2.5 require a 40 tooth gear
0.3, 0.6, 0.7, 0.75, 1.5, 3 require a 42 tooth gear
These are instead in the dial indicator and mesh with the leadscrew

Does anyone else have the same issue?

davidgleach, 2020-10-23:
I also have a Seig mini lathe.  After installing the ELS I have dispensed with the thread indicator as it was worn out anyway.  The trick is to keep the lead screw permanently engaged.  
When you reach the end of the thread pass, stop the lathe-retract the cross slide- reverse the lathe motor- return the cross slide-stop the motor-adjust the thread depth-set the motor to foward.  Under no circumstances do you disengage the lead screw so that back lash is longer an issue.
Once I got used to the new routine it works like a charm.

asm308, 2020-10-23:
Hi David. Yeah I was aware of the technique (I eventually figured that out) but obviously (well I say obviously now 😁)you can’t change anything on the els either.  My lathe is wired up a bit weird so if I use the forward / reverse switch it trips out the main on/off / emergency stop switch as well. So it’s rather a faff to reverse the motor reset the main switch, reverse the motor reset the main switch for each pass.  I was hoping there was a more friendly way.  I guess I’ll have to try and reverse the wiring so the main switch is after the main on/off,  but as you know on these mini lathes everything is rather tight on space.
Thanks for the tip, I’m sure it will help people.

davidgleach, 2020-10-24:
> <electronic-leadscrew@noreply.github.com>
> <comment@noreply.github.com>
> leadscrew has two different gears for dial indicator (#101)

I see your problem
;-)

David Leach

Home: 03 5310 6313

Mob:    0416 119 381

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

carneeki, 2020-10-31:
To be able to get rid of the indicator would require tracking the position of the carriage, which means James would need to track more encoders in the ELS, and a DRO isn't really what he's wanting to do.

An alternative might be to change the indicator and the gear or 'star wheel'. I had a play around on [this calculator](https://www.cgtk.co.uk/metalwork/calculators/threaddialindicator), and discovered that the chart on my lathe (a 2mm metric version of James' lathe) might in fact be wrong, or at least, misleading. My chart indicates that I need to change from 42T to 45T for about half of the threads; but in fact, if the chart were drawn just a little differently, I can keep the 42T wheel in for all pitches (except 1.25mm and 2.5mm); admittedly I lose some of the numbers I can engage the half nut on, but waiting a few seconds extra is way quicker than swapping the star wheels. I might use that calculator to print up a new chart to cover the old one.

asm308, 2020-10-31:
Hi 
Thanks for your detailed reply.  I will definitely check out the calculator. If I can do most threads without changing the tdi I gear that would be great. Pity about the 1.25 pitch though as I think that’s an M10 pitch, but not the end of the world. It will still save a lot of changes.