clough42/electronic-leadscrew #102: Can't compile with 28004x_generic_flash_lnk.cmd

LotusElan, 2020-10-21:
Since I had made changes to the previous version I set that copy aside and did a fresh clone from GitHub. I compiled with no problem but it would not flash.  I changed the linker script from 28004x_generic_ram_lnk.cmd to 28004x_generic_flash_lnk.cmd. But now when I compile I get 38 errors like these:
"../28004x_generic_ram_lnk.cmd", line 32: error #10263: BEGIN memory range has already been specified
"../28004x_generic_ram_lnk.cmd", line 33: error #10263: RAMM0 memory range has already been specified
"../28004x_generic_ram_lnk.cmd", line 33: error #10264: RAMM0 memory range overlaps existing memory range RAMM0
"../28004x_generic_ram_lnk.cmd", line 35: error #10263: RAMLS0 memory range has already been specified
"../28004x_generic_ram_lnk.cmd", line 35: error #10264: RAMLS0 memory range overlaps existing memory range RAMLS0

Any guidance would be appreciated.
Thank You
John

clough42, 2020-11-01:
Make sure you have only one of the link commands in the build at a time.  Those errors look like you have both files, and that's why it's complaining.  Of course, you can just switch to "release" and you'll automatically get the flash command.

If the issue is that your constants are too large to fit in the RAM bank, you can try changing line 108 of the ram link file from:

`.econst          : > RAMLS5|RAMLS6,    PAGE = 1`

to

`.econst          : >> RAMLS5|RAMLS6,    PAGE = 1`

This should allow the constants to be split into RAMLS5 and RAMLS6.

cad-quark, 2021-02-27:
If you flash with computer power alone remove the 5v jumper, leaving the gnd and 3.3v jumpers.  LaunchXL does not need 5v to flash.  the clough42 boards use more power than the pc can provide.  PULL all the jumpers when on 5v power.