clough42/electronic-leadscrew #113: ELS appears to work...but display flickers

TreatHull, 2020-11-27:
After several efforts to troubleshoot, I got to a place where when I powered up the unit, one red LED illuminated on the control board....but nothing else happened.

After tracing, tracing and tracing again, I found that the ground wasn't connecting with the control panel. When I tested the cable (an old VGA cable), the ground lead was open. 

I carefully made up a new cable only 12" long to minimize noise and when I powered up - the control panel was sort of working. The alpha-numeric display was seemed very dim, but I could make out Clough42 and the version number. After the bootup information was done, the RPM showed zero unless I spun the encorder, when it produced plausible RPM numbers except they seemed so dim it was almost impossible to see.

Likewise, when I press the up and down buttons for feed, the value in the feed screen went up and down like it was supposed to. However it seemed so dim that it was almost impossible to see.

While this was going on, several of the red LEDs at the top of the control panel were lit up. LED's 1 and 2 were brightly lit, while LEDs 4 and 7 were lit but not as brightly. (See image below.)

I have 100MHz Rigol oscilliscope which I could use if i knew what I was looking for.

Any thoughts on what might be happening would be welcome. 

[image]

stahlous, 2020-12-27:
I had similar issues with flickering on my LED & KEY board. I had very reasonable `signal voltages on my oscilloscope with the LED & KEY board disconnected, but they went to heck when I reconnected it. I figured I screwed up the soldering when I removed the straighter header pins to connect the supplied 90deg header pins. Turns out that was the issue -- after I re-soldered all the connections it runs solidly. I figure I probably had a whisker of solder nearly shorting out +5V to GND.

TreatHull, 2020-12-31:
December 31st update:

Thanks to @stahlous, one problem solved.
 
I went back and examined the solder job I did on the header and I must have been really careless when I de-solder and removed the old header because a bit of the copper via was lifted and torn away from the Vcc pin. Other parts of the board were getting power, but the torn via appears to have cut off the supply voltage to the TM1638. No wonder nothing was happening.

I did a real kludge repair which involved a blob of solder on the end of a piece of #24 wire and then hooked it up to my Arduino, but still nothing happened. 

As I checked the sketch (program) for the Arduino, I finally noticed that the pin assignment for the output pins was wrong in the code I copied online. One minute correcting and uploading the sketch and -bingo- the control panel lit up at full brightness, no flickering and worked the way it was supposed to with the little Arduino test program.

When I switched cables and connected the control panel to the ELS, the control panel was dead once again.

When I hooked up the scope, I found that the STB and CLK waveforms were good, but that the voltage difference from high to low was only about 2.5 volts:

[image]

However, when I put the scope on STB and DIO, I got a different result. The voltage swing on DIO was just over 5 volts - what you'd expect, while STB was still only about 2.5 volts. 

[image]

I got the same waveforms with and without the control panel connected.

So now (1) the control panel works correctly with signal input from my Arduino which is progress; and (2) I trying to figure out what would limit the high-low voltage swing on STB and CLK, but not DIO.

JonWoellhaf, 2021-01-01:
I suggest you carefully check the switch settings and jumper configuration on the TI board. Good luck!

forrestlw, 2021-01-01:
It appears that he is using a Arduino board. I didn't respond because the Clough42 project it based on a TI board.  "I tried two different Arduino sketches (programs) to drive the TM 1638 LED & KEY board "

TreatHull, 2021-01-01:
@forrestlw  Sorry if I wasn't clear in my explanation. The ELS uses the same components as James at Clough42: a TI Launchpad, the Clough42 booster board and a TM 1638 control panel from Clough42.  

The Ardunio board was only used to test whether the TM 1638 control panel was functioning correctly, which it was.

TreatHull, 2021-01-09:
Hurray!! The slow boat from China arrived this week and my new control panel board was delivered.

Connected it up, fired up the system....and it works perfectly.

The problem was a bum control panel board, probably due to my rough-handling the removal of the original header and installing the new one.

Even before I hook it up to the lathe, I want to say that this has been a terrific project. This is a hobby that I pursuit to have fun and learned new things. Though it required some patience, I had fun and I learned buckets, one of the most rewarding projects I've done in long time.  I encourage anyone else who's considering it to give it a go.

Couldn't have done it without the excellent videos and components from Clough42 and the advice received. Spec thanks to @stahlous who hit the nail on the head. Thanks all!!

[image]

closed 2021-01-09 by TreatHull