clough42/electronic-leadscrew #114: Dividing head function

Monchichack, 2020-11-28:
Additional feature that would be very handy to have:

Dividing head function.
I'm currently trying to cut a spline positioned in the lathe with the spindle indexed using a dividing plate. What a faff!
I'd love to see an indexing function to the ELS, then I could use a mtb disk brake to lock the spindle in place. Happy days.

Very basic version when selected simply gives an encoder position readout on display.

Basic version is the same as above but has separate configuration settings within the code to adjust spindle : encoder ratio.

Deluxe version, when selected has spindle : encoder ratio configuration settings in code & allows the operator to select how many divisions the spindle should be divided into & calculatethe steps required to achieve this. Once setup, as the spindle is rotated the display will count up/down the number of divisions selected, incrementing each time a new division is reached, then loop back to 1 & count again.

What a lovely xmas present this would be :)

gusosborne, 2021-01-03:
This would actually be pretty cool - I just had the same idea, figured I’d search for previous posts.  Perhaps a semi-separate project with an alternate boot mode..?

Monchichack, 2021-01-03:
Following my post, I noticed an older thread on this topic too, suggesting a few people are interested.
Separate boot mode's would be the best way of navigating the additional functions of the ELS as it evolves. The changes would need to be adopted & integrated or easily copied into the main code as it progresses to remain 'futureproof'.

Wish I could write the code but it's beyond my limited programming experience.