clough42/electronic-leadscrew #115: Stepper Drive Starts at a resolution of 400

ofer1973, 2020-12-16:
If the driver of my stepper starts with a resolution of 400, Something needs to change in the code ?
[image]

Hobby-Lofty, 2020-12-18:
Ofer, I would leave the code as:
// Stepper driver configuration
#define STEPPER_MICROSTEPS 8
#define STEPPER_RESOLUTION 200
and set the dip switches on your DM542T driver to 1600, off off on on

for a 1.8 degree stepper motor = 200 steps per rev,  times this by 8 = 1600
Lofty

ofer1973, 2020-12-18:
And if I want to change the poly of the stepper to a ratio of 1: 2, what do I need to change ?

Hobby-Lofty, 2020-12-18:
try setting your driver to 800, this should double the speed of the stepper motor.

ofer1973, 2020-12-19:
My intention is just the opposite, I want to reduce the speed of the stepper to half.

Hobby-Lofty, 2020-12-19:
try setting your driver to 3200, this should half the speed of the stepper motor.

ofer1973, 2020-12-19:
I'll try it, thank you very much :-)