clough42/electronic-leadscrew #116: I'm bought Els Pc Board, Encoder, Els Control panel, it's time for the servo Kit...

andrea271068, 2020-12-17:
I as I wrote in the objective, it's time for me to buy the servo Kit, servo+driver+ power supply, I don't have 3 phase, is so important? I can't use my domestic 220V? I live in Italy it's normal for us 220V.

I can't find even in e-bay your servo and the Kl -5080H

I have a Chinese mini-lathe, w 

I can buy  those kind of servo?

Nema23 2N.m 57 Hybrid Closed Loop Servo Stepper Motor 2Nm 76mm+HSS57 Servo Driver Controller CNC Kit 24-50V DC For CNC Router Engraving Milling Machine

many thanks in advance

Hobby-Lofty, 2020-12-18:
Andrea,  I fitted to my 7x14 mini lathe:
eBay kit: 3Nm Nema 24 Closed Loop Stepper Hybrid Servo Motor 5A 86mm &Driver CL57Y CNC Kit
it works fine
Power supply from banggood : AC 110/220V to DC 0-60V 8A 480W Adjustable Regulated Switch Power Supply Driver Transformer W/ LCD Display for LED Strips, I've been running it at 30v .
In the UK we have 240v and it works fine
The nema 24 just has 10mm diameter shaft.
this may be overkill but I know it works for me.
Good luck
Lofty

forrestlw, 2020-12-18:
Andrea, The power supply does not require 3 phase to operate. Just 115v or 220v single phase house voltage. The phase is the type of motor/drive set you are using. A 3 phase set has a 1.2 degree step angle. A 2 phase has a 1.8 degree step angle. Most motors have a 8mm shaft and very few have a .250" (6.35mm) shaft.

andrea271068, 2020-12-18:
Dear Lofty, many thanks, really, I was confused in what to buy.

In case of problem I can write you again?

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

COMPOSAD s.r.l.

Sede Legale: Viale Lombardia, 29 - 46109 Viadana (MN) - Italy

ph: +39 0375 783611 | fax: +39 0375 7836695

Sede Operativa: Via Lombardi 1 - Z.I. Gerbolina - 46019 Viadana (MN) - Italy

C.F - P.IVA/VAT code e R.I (IT)02317920201

www.composad.com<whttp://www.composad.com>

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

Il presente messaggio di posta elettronica e i relativi allegati sono strettamente confidenziali, soggetti alle leggi vigenti, pertanto se ricevi questo messaggio di posta elettronica per errore, per favore contattaci: e' assolutamente vietato usare, copiare, o divulgare il messaggio e ogni informazione in esso contenuta, tranne che al destinatario originario. Le comunicazioni Internet non sono sicure e non ci riteniamo responsabili per eventuali abusi di terze parti, per alterazioni nel corso della trasmissione, né per danneggiamenti causati da virus o altre anomalie a noi non imputabili.
----
This message and its attachments are addressed solely to the persons mentioned above and may contain confidential information. If you have received this message by mistake, be informed that any use/copying/forwarding of any part of its content is completely prohibited. Please return it immediately to the sender and delete the message. Internet communication is not secure and therefore we are not responsible for any misuse of it by third parties; Alterations during the transmission or damages caused by viruses or other anomalies are not attributable to us"

'

andrea271068, 2020-12-22:
Hi, finally I bought a servo Kit, I follow your suggestion but come with 350W power supply 36 V, I hope is good.
I never do electronic configuration or job, is my first time, hope that I can do it, is a challenge
[image]
.
may I will need your help guys to go true to this adventure. 
Especially for the setup and configuration. 

Thanks for everithing.