clough42/electronic-leadscrew #118: Spindle gear Ration for the Omron encoder 1024P/R

andrea271068, 2020-12-22:
Hi, I'm a little lost this is my actual gear ration Spindle 56 teeth the idle gear wheel 80 !! is a mess, I don't know how to set for the encoder. 

Please can you help me to solve it, considering that I have a 56 teeth spindle gear what hind of pulley I have to buy to connect the encoder? 
I should connect directly the encoder to the spindle gear by buying another 56 teeth gear or I can buy a pulley to be place side by side to the 80 teeth idle gear wheel; that case what pulley dimension I should buy? I don't know the correct ratio to follow. 

[image]
[image]

Hobby-Lofty, 2020-12-22:
Adria, I would make or buy a 56 gear for the encoder the same size as the spidle gear, you can drive straight from the spindle gear to the encoder gear, or fit any size idler in between the 2 x 56 gears. Keep the gear module (metric) or DP the same.

andrea271068, 2020-12-22:
many thanks, I bought a 56 gear, maybe I have to adjust the bore diameter to fit in the encoder in a way, I don't know the bore diameter because the supplier didn't place it in eBay the required information, let's see. 
In case I will need an idler gear I can use any size? 
And if I want to fit a pulley to drive the encoder from the 80 idler 80 teeth gear, may you have some suggestions? what size pulley I should try to fit?
Many thanks for your help

TreatHull, 2020-12-22:
I had the same problem: I had a gear with the same number of teeth as a gear on the lathe - but the hole in the gear was 10 mm while the encoder shaft was closer to 6 mm. I bored out the hole in the gear, then made up an adapter with the ID at 6 mm to fit the encoder and the OD sized to fit the bored out hole in the gear. The adapter also had a flange which held the adapter onto the gear with machine screws.

The pictures are probably easier to understand than the explanation. Incidentally, I wanted to make the whole thing out of brass, but didn't have any brass stock with a large enough diameter. In the end, I made the adapter out of aluminum with a section of brass press fit in the center.
[image]
[image]

andrea271068, 2020-12-22:
Nice, very nice job, I think is the only way to adapt, thanks for the suggestion and sample :)
My gear is in metal, but will arrive end of Jan :( 
I can't wait, but anyway also the encoder is in the way, now I'm building up everything in mind to be ready.

andrea271068, 2020-12-22:
I can't wait

Hobby-Lofty, 2020-12-22:
Andrea, you need a 1 to 1 ratio spindle to encoder, 56 to 80 to 56, fit a pulley to the side of 56 and a pulley to encoder 
both pullies must be the diameter. you don't have to use the 80 tooth.

On my Warco 280VF I used a couple timing gears and belt, via gear directly driven gear from the spindle gear.
utube about 3mins in
[https://youtu.be/ONOhrxajeRc](url)

Lofty

TreatHull, 2020-12-22:
I had a bunch of gears that I got with my lathe. I wasn't going to need them once I had the ELS running so I figured I could sacrifice them. I found one gear that had the same number of teeth as the gear on the lathe I wanted to drive the encoder from. Before working on that gear, I made a couple of prototypes of my adapter with other gears. Once I was sure the design and my skills were good enough, I did the adapter shown in the photos.

andrea271068, 2020-12-22:
>Many thanks, Lofty, I will go to YouTube t check, I'm very curios 
> Thanks TreatHull, unfortunately, I don't have I order from China, and I have to wait

andrea271068, 2020-12-22:
Lofty, your link is not available
?

andrea271068, 2020-12-22:
Would me nice to see it, but is not available your you tube video, I don't know why. 

"On my Warco 280VF I used a couple timing gears and belt, via gear directly driven gear from the spindle gear.
utube about 3mins in
https://youtu.be/ONOhrxajeRc"

andrea271068, 2020-12-22:
TreatHull, The closest gear that I have as spare is 60 teeth, but my spindle is 56 teeth

Hobby-Lofty, 2020-12-22:
try copy and paste into your browse. https://youtu.be/ONOhrxajeRc

andrea271068, 2020-12-22:
Okay, I found it, now I will check your solution, very curious

andrea271068, 2020-12-22:
Nice, as the idler gear you place the 56 teeth? to run 1:! with your spindle, and for the pulley what ratio did you follow? what pulley set did you buy?

Hobby-Lofty, 2020-12-22:
Andrea, I used these from RS Components
Aluminium Timing Belt Pulley, 6mm Belt Width x 2.5mm Pitch, 60 Tooth, Maximum Bore Dia. 35mm Stock no.:744-558 Qty:2 
-- | --
Timing Belt, 122 Teeth, 305mm Length X 6mm Width Stock no.:474-5117 Qty:1
Contitech 6 / T2.5 / 305 SS, Timing Belt, 122 Teeth, 305mm Length X 6mm Width Stock no.:474-5117

andrea271068, 2020-12-22:
thanks, did you bought true eBay or true RS Components website directly?

andrea271068, 2020-12-22:
I found it true RS, Thanks :)

Hobby-Lofty, 2020-12-22:
yes form RS Components website directly, you may find similar at less cost on the web.

Lofty

andrea271068, 2020-12-22:
I will look around now, is clear that you start with spindle gear 56 teeth and idler 56 teeth, and there you addon two 60 teeth pulley. 
Did you see the Clought42 video, he starts with something similar with a different ratio in the spindle was 40 and idler 60 if I remember well, to that he addon side by side to the 60 idler a 40 teeth pulley, this is difficult to me to understand.

Hobby-Lofty, 2020-12-22:
Andrea, I seen all of Jame's videos, his gear ratio still ended up as 1:1 from spindle to encoder, , with the belt driven version you have 1:1 for the 2 gears and 1:1 for the timing pulleys belt driven,

Lofty

andrea271068, 2020-12-22:
he has 40 teeth gear in the spindle and 60 in the idler, he place (if I remember correctly) next to the 60 gear a 60 teet pulley (???) the diameter is different from the gear 60 (???) may have the same ratio even if the diameter is different ?, and to the Encoder I 40 teeth pulley, but in this way, the ration is not more 1:1, right? here is where I miss my self.

andrea271068, 2020-12-22:
for these reasons I start to think but instead to buy a metal gear 56 teeth to replace my 80 teeth if I will add to 80 gear a pulley 60/80 teeth and to the encoder 40 teeth will not work?

andrea271068, 2020-12-22:
may I should stop to think about it and do as for your solution and that it!

Hobby-Lofty, 2020-12-22:
Andrea, keep it simple.

Merry Xmas
Lofty

andrea271068, 2020-12-22:
> Andrea, I seen all of Jame's videos, his gear ratio still ended up as 1:1 from spindle to encoder, , with the belt driven version you have 1:1 for the 2 gears and 1:1 for the timing pulleys belt driven,
> 
> Lofty

How is Jame, Clugh42?, if yes,  the only thing as I try to explain the pulley that he adopts clearly have a different diameter from the gear setup, here is where I miss my self 

[image]
[image]

JonWoellhaf, 2020-12-22:
Andrea,

I hope this will help your understanding.

James Clough is Clough42.

If you have a gear OF ANY SIZE that is turning at spindle speed, you can mount a timing belt pulley OF ANY SIZE to that gear. Then use the SAME SIZE timing belt pulley on the encoder and the encoder will turn at the same speed as the spindle.

Regards,

Jan (Yan)

From: andrea271068 
Sent: December 22, 2020 10:06
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Spindle gear Ratio for the Omron encoder 1024P/R (#118)

  Andrea, I seen all of Jame's videos, his gear ratio still ended up as 1:1 from spindle to encoder, , with the belt driven version you have 1:1 for the 2 gears and 1:1 for the timing pulleys belt driven,

  Lofty

How is Jame, Clugh42?, if yes, the only thing as I try to explain the pulley that he adopts clearly have a different diameter from the gear setup, here is where I miss my self

andrea271068, 2020-12-22:
> How is Jame, Clugh42?, if yes, the only thing as I try to explain the pulley that he adopts clearly have a different diameter from the gear setup, here is where I miss my self
>
> [image]
> [image]

**This is my question? check below very confuse now** 

[image]

andrea271068, 2020-12-22:
JAN, THANKS !!! Finally, I get it, have a great Christmas, sorry to bother all of you guys, please don't leave me alone for next problems, :) I do something completely different in my life, but mechanics was my study at a younger age that now I want to take back, I don't remember everything much time past.

andrea271068, 2020-12-22:
> Andrea, keep it simple.
> 
> Merry Xmas
> Lofty

:) 
You too, thanks for everything.

andrea271068, 2021-02-25:
WOW!!!!! We get it!!! The problem was un the Led&Key board!!!!! 

Now I have to place in the enclosure and make a run test in the machine!!! I'm soooooo happy 

In place there the photo of what I did for everybody   😅😅😅 thanks to everybody!!!  [image]

[image]

andrea271068, 2022-02-21:
> Hi, from the last time I changed my lathe, now I have a lathe that practically is like your but with a different brand, my is Bernardo lathe from Austria. My lathe is like your with a feed bar for feeding and a lead screw and gear box, with the three seclectors.
As James says in ELS Part 20, for the lathe with a feed bar I should to configure in Configuration.H. 
What I should configure? What do you do? 
My lathe has for feeding 1/0,075 ratio, this means that I have in the spindle a gear of 40 teeth + the gear train with (70-45/ 80-20 gears) and end up with 85 teeth in the shaft, so = gear ratio 1/0,075.
That means when the spindle makes 1 full turn the shaft bar makes only 0,075 fractions of a turn. The same from a different perspective: Every 10 spindle turn the shaft move the 85 tooth gear 0,75 of 1 turn) Or the spindle gear make 13 turns to spin the shaft 1 time (Spin Ratio 13:1)
Now assuming that what I should set (write) in configuration.h, what I should configure?

Some people did it? can you guys put me on the correct track, I don't know what to do, I'm not familiar with code, thanks