clough42/electronic-leadscrew #119: debug problem, I need help please

andrea271068, 2020-12-28:
Hi, 
I have an error message, I download from Ti the latest version CCSTUDIO_10.1.1.00004 I also download from Clogh the latest firmware version 1.3.01 and I don't know how to solve it.
Please I need help because is the first time that I manage firmware and programming. 

Hope to receive soon your kind feedback, many thanks in advance. 

[image]

JonWoellhaf, 2020-12-28:
Hi, Andrea.

Please see if you can download version 9.3 of Code Composer Studio. I believe James said he intends to make the necessary changes to get his code to run with version 10, but he hasn’t yet gotten to that.

Version 9.3 of CCS should work fine.

Happy New Year! (Let’s hope it will be a happy one, anyway.)

Jan (Yan)

From: andrea271068 
Sent: December 28, 2020 10:39
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] debug problem, I need help please (#119)

Hi,
I have an error message, I download from Ti the latest version CCSTUDIO_10.1.1.00004 I also download from Clogh the latest firmware version 1.3.01 and I don't know how to solve it.
Please I need help because is the first time that I manage firmware and programming.

Hope to receive soon your kind feedback, many thanks in advance.

forrestlw, 2020-12-28:
It's still available I just downloaded it. It's not easy to find. Full download for 64 bit only. 1010mb.  Code Composer Studio Version 9 Downloads. http://software-dl.ti.com/ccs/esd/documents/ccs_downloads.html

andrea271068, 2020-12-28:
Many thanks, happy new year you too.

But I’m running metric, hope will work too.

Thanks for all, let’s see

Ciao

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

COMPOSAD s.r.l.

Sede Legale: Viale Lombardia, 29 - 46109 Viadana (MN) - Italy

ph: +39 0375 783611 | fax: +39 0375 7836695

Sede Operativa: Via Lombardi 1 - Z.I. Gerbolina - 46019 Viadana (MN) - Italy

C.F - P.IVA/VAT code e R.I (IT)02317920201

www.composad.com<whttp://www.composad.com>

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

Il presente messaggio di posta elettronica e i relativi allegati sono strettamente confidenziali, soggetti alle leggi vigenti, pertanto se ricevi questo messaggio di posta elettronica per errore, per favore contattaci: e' assolutamente vietato usare, copiare, o divulgare il messaggio e ogni informazione in esso contenuta, tranne che al destinatario originario. Le comunicazioni Internet non sono sicure e non ci riteniamo responsabili per eventuali abusi di terze parti, per alterazioni nel corso della trasmissione, né per danneggiamenti causati da virus o altre anomalie a noi non imputabili.
----
This message and its attachments are addressed solely to the persons mentioned above and may contain confidential information. If you have received this message by mistake, be informed that any use/copying/forwarding of any part of its content is completely prohibited. Please return it immediately to the sender and delete the message. Internet communication is not secure and therefore we are not responsible for any misuse of it by third parties; Alterations during the transmission or damages caused by viruses or other anomalies are not attributable to us"

'

andrea271068, 2020-12-28:
?

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

JonWoellhaf, 2020-12-28:
I think forrestlw means CCS version 9 is still available at the site he references.

From: andrea271068 
Sent: December 28, 2020 13:12
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] debug problem, I need help please (#119)

? 

Andrea Pisani 
Export Manager 

Composad srl – Italy 
Mobile Phone: 0039 335 70 98 102 
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com> 

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg] 

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

andrea271068, 2020-12-29:
Thanks

Scarica Outlook per Android<https://aka.ms/ghei36>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

TreatHull, 2021-01-09:
I couldn't make it run at all with Code Compose Studio 10, but it worked right away as soon as I installed version 9.

andrea271068, 2021-01-11:
Hi, many thanks, s
o I will uninstall CCSTUDIO 10 and I will download 9.3.0.00012 (9, last version)

I didn’t try yet because I was scared to make something not good, I will install 9.3 it’s good?

waiting for your kind feedback, many thanks

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]