clough42/electronic-leadscrew #120: No TI XDC Debug Probe Support CCS 9.3.0 on MacOS

Jan-Andries, 2021-01-04:
Good evening,

Let me start by thanking James for his effort in this project. I am planning to install the ELS on my Myford lathes. Being no expert on this subject I am just trying to follow James as close as possible. I this stage I purchased the Control Panel Kit and Interface so I would not miss out. ;-)
 
In addition I just installed CCS 9.3.0 to get familiar with it on my mac OS High Sierra . During the pre installment setup I was not able to select the TI CDS Debug Probe Support as instructed by James. 

Will this be an issue in any way?

Best regards,

Jan-Andries

[image]

clough42, 2021-02-24:
Don't know.  Did you eventually get it programmed?  Maybe the probe support is installed by default?

I used version 9.1.

clough42, 2021-03-02:
I installed 10.2 on my system over the weekend, and it doesn't ask about the TI debug probes on this screen.  I think it's just a place where you can select *additional* probes.