clough42/electronic-leadscrew #121: It just won't run

xfuzz, 2021-01-13:
I have it all together assembled for testing. Servo wont spin. It gets energized so I can't turn by hand when powered. The control pannel lights up.when I spin the omron encoder the rpm's  show up on the control panel. The buttons show changes with each push ,  but servo does not spin. I have triple checked continuity all good. I am perty sure all dip switches are set correct as well. I tried going to the older version software 9.1 to up load the els software too. One thing when doing so is I do not get the red  pause break point icon on the TI  software like Jame's does in his videos. It just seems to complete to upload without stopping no break with red icon. The servo & encoder are from Automation Tecch. The number on the encoder is different it is ES-D508 , servo is ES- M32320S. I called as when I received them,  I ordered same as the ones James used. They told me they are exactly the same and numbered for a different market.I am lost.

mac651955, 2021-01-17:
Mine didn’t work when I initially set it up.  Flip the switches on the micro processor back and forth a couple times to make sure they make contact.  Additionally go ahead and make all your connectors up and use them as your problem may be in connections.

xfuzz, 2021-01-17:
Thanks for 5hevinout. I tried the switch flipping multiple times M I red all the post and someone suggested it prior to you. I have continuity on all of the connections. The servo is getting power as I stated if I flip one of the two dip switches on the servo encoder it will rotate back and forth one revolution non stop. It's  something simple , but what?

mac651955, 2021-01-17:
Are you using the servo encoder combo that James used or something different.  If it is the same double check your three leads from the servo to the servo controller.  Make sure they are correctly connected U, V, W. You might be getting your phases working against each other.

xfuzz, 2021-01-17:
Yes. As I indicated above  according to Automation tech it isi identical. . I will have to find the schematic,  I have have brown on "U", blue on "V", black on " W", I think that is correct.

xfuzz, 2021-01-27:
[image]

xfuzz, 2021-01-27:
[image]
[image]
[image]
[image]
[image]

JonWoellhaf, 2021-01-27:
Do you have the Pro_Tuner, hardware, and software manuals for your ES-D508?

If not, you can download them from https://mecheltron.com/en/product/es-d508.

I think you will need all three manuals.

Do you have an oscilloscope?

Your connections from James' Boost Board to the ES-D508 should be as shown at the top of page 8 of the hardware manual. The boost board has a "sinking" type output.

Do you have a jumper from pin 1 of the driver connector to pins 3 and 5 of that connector? In the photo, it looks like you do, but I want to be sure.

candera, 2021-01-27:
I'm actually having a similar problem (see #125), and am no expert, so take the following for what it's worth, which is "not much". 

I took a long look at the pictures you posted, and I can't see anything wrong. You've checked continuity, and it sounds like when you flip DIP switch #2 it runs the self test, which seems to indicate that the motor is running. The only things I can think of are either that continuity is not actually good or that there's a lot of noise in the system. But with noise you'd expect some movement, even if random.

The critical pin for you is the STP- pin. If ENA- were not pulled low, the motor wouldn't be energized. And DIR- doesn't really matter - whether it's high or low the motor is still going to turn, it's just which direction. Unfortunately it's a little hard to test, since in order to turn the motor it needs to go high and low a bunch of times. I did try wiring a pushbutton switch between ground on the TI board and the STP- pin so that I could press it a bunch of times and generate a pulse train. I even tied the ENA- and DIR- pins low to try to take out any variables. I *think* that was a good test (maybe someone else can tell me why it's a bad idea) so perhaps you could replicate it. If pushing the button doesn't rotate the motor at all then perhaps it is a motor controller problem. Iit might take a lot of pushes, since if you're configured for, say, 800 microsteps per revolution, each push is only going to move it like a third of a degree.

In my case, I believe perhaps the motor is bad, as I can get at best random movement from it. I've ordered a stepper to replace it. Hopefully I'll have more luck with that. The hybrid servo concept is very cool, though - I really wanted it to work, so I sort of hope it's just a bad motor rather than some weird thing with my setup.

xfuzz, 2021-01-27:
Jon et me start with a huge thank you for your assistance i downloaded the manuals. Not sure what I am looking at. I do not have an oscilloscope.Yes all the positives are connected via a jumper wire from one to the other two. Continuity on all, I do not have the required cable to enter data to the encoder. I will order one right away if that is what I need.  Actually I don't know a sinking output from one  that is not. I can follow instructions though.

mac651955, 2021-01-27:
In the photos it appears that (if you used the same wire colors as the encoder) your encoder is wired wrong.  The first pin from the bottom is 0v common and you have what appears to be the blue wire going to it.  Also you have 6 connections on your plug and that is a five pin header.  Check your wiring then maybe try a five pin connectors.

xfuzz, 2021-01-27:
Mac I triple checked and blue is marked common the omron. i up loaded pic. the 6 spot on the connector is  empty

JonWoellhaf, 2021-01-27:
xfuzz,

OK, let's see if I can help you get this working. Troubleshooting would be much easier if you had an oscilloscope. Can you borrow one?

Candera (above) offered some good information. I'd like to simplify the problem as much as possible.

First though, I think it's important you understand the schematic at the top of page 8 in the driver's hardware manual. Do you?

xfuzz, 2021-01-27:
[image]
[image]
[image]
[image]

xfuzz, 2021-01-27:
I would not how how to use one ,  have no one to borrow from. I uploaded more pics of the wiring I think I do understand page 8.

mac651955, 2021-01-27:
Did your server come with the signal cable, if it did are you using it.  Just covering all bases.

xfuzz, 2021-01-27:
Mac yes it did I am using it

xfuzz, 2021-01-27:
[image]

JonWoellhaf, 2021-01-27:
OK.
I am confused by the statement at the top of page 8. It reads, "For the enable signal, apply 0V between ENA+ and ENA- or leave them unconnected to enable the drive." If left unconnected, the LED in the drive's opto isolator will be off. I would expect that to be interpreted as disabled, but the text says off is enabled. I expect the sense of this input can be changed in the software.

So please try this:
1) Power off.
2) Disconnect the ENA- wire from pin 6 of the driver. Tape its end or attach a wire nut so it doesn't short against something.
3) Power on. Does the control panel operate normally? When you spin the encoder do you see RPM displayed? Does the motor turn?

candera, 2021-01-27:
That is pretty weird: definitely seems like powering the LED should enable the drive. It should be easy to tell whether enable is active high or low, though, because when not enabled the shaft should spin pretty freely. Again, wiring ENA+ to 5V and ENA- to ground should be a good way to verify. 

If it turns out the logic is inverted (and perhaps it's that way to allow the motor to be enabled without requiring the input to be driven) then it may be simply that you need to change Configuration.h to set `#define INVERT_ENABLE_PIN true`, or to `false` if it's already true. See [this line](https://github.com/clough42/electronic-leadscrew/blob/master/els-f280049c/Configuration.h#L78).

It's easy to tell whether you're driving the DIR and ENA inputs pins high or low, BTW. Just use an multimeter and check the DC voltage across the two pins (say, ENA- and ENA+). If you're driving ENA- low, you'll see about 5V across them. I would think that means "motor is enabled" but again that wording in the manual suggests otherwise.

xfuzz, 2021-01-27:
Jon, ok did that .the rpm shows on display when I spin by hand the omron.

xfuzz, 2021-01-27:
candra I get 5.3 volts between ena+ and ena

JonWoellhaf, 2021-01-27:
As Craig said, with ENA- disconnected, can you turn the motor by hand?

xfuzz, 2021-01-27:
jon, not easy i get a resistance like it is fighting me but can not complete a half turn

JonWoellhaf, 2021-01-27:
OK. That indicates the drive is enabled with the opto isolator off. As Craig said, you will need to change INVERT_ENABLE_PIN true in James' software. Or, after you get the RS-232 cable and run ProTuner, make the change there.

New thought. I remember others having a problem because they connected the encoder cable directly to the ES-D508 WITHOUT using the supplied extension cable. Take a look at page 11 in the manual. It clearly shows the extension cable being used and says "* must be used." I think it's more than just an extension cable, and also makes some wiring changes. Are you using the extension cable?

candera, 2021-01-27:
One other check you could do: connect the motor control wires up to the TI board again, power everything up and give the encoder a spin. That should force the board to enable the motor. Measure the voltage across ENA+ and ENA-. It should NOT read 5V. If it does, that means you need to invert the enable pin in the firmware, recompile and flash it onto the board. 

If you hit the power button on the control panel, it should disable the motor. So while you're measuring voltage, you can see if it flips from 0 to 5V and back again when you toggle the power button on the control panel. Just as one more verification.

candera, 2021-01-27:
But of course if the voltage is 5V when the motor should be DISABLED, then your firmware setting is already correct and we'll have to look elsewhere for the problem.

xfuzz, 2021-01-27:
Jon yes I musing the iextention cable. I looking at james software to see what it say now

xfuzz, 2021-01-27:
this is what it says in config h
#define INVERT_ENABLE_PIN true
#define INVERT_ALARM_PIN true

candera, 2021-01-27:
Which is probably correct, at least as far as the enable pin goes. You can verify it as described above if you want to make sure.

xfuzz, 2021-01-27:
now i am getting different reading o volts. around .158 volts

xfuzz, 2021-01-27:
think i should try to flash the software again?

JonWoellhaf, 2021-01-27:
Yes! It certainly could be incorrect TI board switch settings. As others have said, the switches are a bit flakey and may appear to be in the correct position but not actually making contact internally. Some advise to move the switch back and forth several times.

You said your board did NOT have tape over the switches. I am very surprised to hear that. I wonder if you got a used, and possibly defective board, but can't imagine TI doing that intentionally.

xfuzz, 2021-01-27:
I know it seemed strange to me that there was no tape on the dips .Especially after James made so much mention of how hard they were to remove. I could order another one at this point I am chasing dollars anyway. So do you think  I need that programing cable and should I order a new TI board? I upgraded a pic of the placement of the dips as I have them positioned.

JonWoellhaf, 2021-01-27:
If you intend to keep your ES-D508, I think you will need the RS-232 programming cable and the free ProTuner software.
With the Enable input disconnected, does spinning the encoder -- like with a drill -- cause the motor to turn?

xfuzz, 2021-01-27:
i tried using a drill to spin it with no result. I just re-flashed and now the rpms do not shown on the display.

xfuzz, 2021-01-27:
the only thing i did was flip the dips back nd forth a few times and re flashed
 now no reading of rpm ondisplay

JonWoellhaf, 2021-01-27:
OK. No RPM display. Do the control panel buttons work normally?

xfuzz, 2021-01-27:
yes

xfuzz, 2021-01-27:
Jon are you in the USA

JonWoellhaf, 2021-01-27:
OK. Double check the encoder wiring and see if something broke loose. I'm in Utah.

candera, 2021-01-27:
I'm trying to remember if my dips had tape on them. I think they did not - perhaps TI made a change where they aren't taping them any more. One of my mistakes was not flipping the switches to the correct position. For me that manifested as the control panel not working, since S9 enables GPIO32/33, which are connected to CLK and STB on the control panel. The other switches affect only the encoder header, but since you're no longer seeing RPM it sounds like maybe they are no longer set correctly. 

Check them vs: 

```
S3.1 = 0 Connects eQEP1 to J12
S3.2 = 0 Connects eQEP2 to J13
S4 = 1 Connects eQEP1 to J12
S9 = 1 (default) -- Connects GPIO32/33 to BoosterPack site 2 pins
S6 and S8 can be in any position. 0 is the default.
```

candera, 2021-01-27:
It's not surprising that you would see readings like 0.1V, BTW. When those pins aren't being driven to ground they float. There's probably some very high resistance path to ground through them at that point, which would divide the 5V signal between the internal resistor and wind up with a 5V to 4.9V (relative to ground) reading across the two pins. Or something like that. I certainly saw low non-zero readings when I was doing the same test.

xfuzz, 2021-01-27:
nope they are all good

xfuzz, 2021-01-27:
the dip switch have dot on one end and a double zero symbol at the other which is on

xfuzz, 2021-01-27:
ok got the rpm to show again but no servo movement

xfuzz, 2021-01-27:
OK wold anyone be up for me sending the whole set up. I pay for shipping both ways of course plus your time and any parts if needed?

xfuzz, 2021-01-27:
you could try my boards on your set up see if they work.

JonWoellhaf, 2021-01-27:
Are you on Practical Machinist?

xfuzz, 2021-01-27:
The only thing I changed in the software is I have a 8tpi leedscrew

xfuzz, 2021-01-27:
Jon do you mean do I subscribe to the forum?I get theri emails but have not paid for membership yet

JonWoellhaf, 2021-01-27:
There is no fee to join Practical Machinist at <https://www.practicalmachinist.com/>. I am known there as Yan Wo. If you join you can send me a private message and I can respond with my address. I don't want to send it in the clear. I'll be happy to look at your setup. Where are you located?

xfuzz, 2021-01-27:
Roseville Michigan just outside of Detroit

JonWoellhaf, 2021-01-27:
OK, a little too for of a drive. :)

candera, 2021-01-27:
You're a good guy Jon. Good luck xfuzz! For my part I'm hoping a new motor does it for me. If not I imagine I will be mentioning my woes. :)

xfuzz, 2021-01-27:
Just a tad.  Ok I signed up. My user is Xfuzz

xfuzz, 2021-01-27:
Craig, your a good guy as well. I truly appreciate the huge block f time  you gave me thank you ever so much and good luck

xfuzz, 2021-01-27:
Jon how do I find you on PM the search for Yan Wo bring up a lot f Chinese writting

JonWoellhaf, 2021-01-27:
Thanks, guys.

Let's try a little more remote troubleshooting before you send anything.

Do you have any resistors in the range 1000 to 10,000 ohms?

I sent you a private message. When you log into Practical Machinist, Right at the top to the right you'll see NOTIFICATIONS. There should be a dot there indicating you have a PM (private message). Click it and you should see my message to you.

You should be able to find me as Yan Wo by clicking Search and Search Forum Discussions. That'll bring up a form where you can type my user name with Exact name checked. When you click Search Now, you'll see my user name underlined. Click it and you'll see my profile, and under that Send Private Message.

xfuzz, 2021-01-27:
OK, No resisters I never do any electronics I built several computers but that was mostly assembling parts  .

xfuzz, 2021-01-27:
Yes I have your message now.

[40 more replies]

closed 2021-02-11 by xfuzz