clough42/electronic-leadscrew #123: Re-using plastic gears

TreatHull, 2021-01-15:
This isn't an issue, instead I wanted to share my no-cost solution to driving the optical encoder and lead screw in case it helps someone else.

Any time I order anything on ebay, it seems to take months to get here in Canada. Although James used belt drive for his ELS project, I decided I wanted to do mine with materials already on hand so I didn't have to wait forever for belts and pulleys to arrive.

What I had on hand was a lot of plastic gears. I was able to come up with a 1:1 gear ratio for the encoder and 2:1 for the lead screw. Apart from brackets, all I had to do was to turn hubs so the gears would mount on the shafts.

Enjoy!

[image]
[image]
[image]
[image]
[image]

xfuzz, 2021-01-27:
Can you tell how to post a picture 9n here. I can find no reference as to how.

TreatHull, 2021-01-27:
Just drag the image from your file onto the box where you’re writing your message.

Treat Hull, B.A. (Hon), M.A.Sc. (Management)
Broker of Record
Treat Hull & Associates Ltd., Brokerage
8 Jasper Avenue
Picton ON K0K 2T0
e: treat@treathull.ca
m: 613-503-0027
o: 613-645-2033
We don’t take listings.
We serve only buyers.

xfuzz, 2021-01-27:
Oh, I am using a tablet.  It's not like a pc not sure I can drag them, I'll
try .  Really appreciate your help. Thanks aga8n

xfuzz, 2021-01-27:
Got it, had to email from tablet to self then open on my pc. Worked like a
charm. Thanks again

clough42, 2021-02-24:
Very cool.  Thanks for sharing.