clough42/electronic-leadscrew #124: How to wire Encoder Omron E6B2-CWZ6C 1024 P/R to the TI launchXLF280049C

andrea271068, 2021-01-17:
Good morning,  
I bought the suggested Encoder, now I have to wire to the TI board, I have 5 wire + the Shield, but I can't fine the wiring diagram and I'm scared to mess up everything, the brown is simple because is the 5 volts, but for the rest, I don't know from where to start, also the shield has to be connected to the GND pin on the TI board?

The Omron has 5 cables: 
   Brown= 5V > this is clear where to be connected.
   Blue= OV (Common) > Where I should connect this? (have to be connected to TI board GND pin??)
   Shield= GND >  Where I should connect this? (but on the TI board there are only 5 pins) 
   Black =  OUT A ? > Where I should connect this? (Does this correspond in the TI board to 1A or 1B or 1I?)
   White = OUT B > Where I should connect this? (same as above).  
   Orange = OUT Z > Where I should connect this? (same as above). 

TI Board:
   GND Pin > where?
   5VO Pin
   1I Pin > Where?
   1B Pin > Where?
   1A Pin > Where?

I'm a little lost here, I can't find in youtube or here the wiring diagram to be followed. 

Please somebody can be so kindled to help me, thanks

JonWoellhaf, 2021-01-17:
Hi, Andrea.

James’ Wiki page at <https://raw.githubusercontent.com/clough42/electronic-leadscrew/master/docs/images/WiringDiagram.JPG> says this about the encoder wiring: “The encoder usually five wires, plus a shield. Check the markings on the LaunchPad board to determine the correct pins for 5V power, ground, A, B and I (index) wires. Connect the shield to a chassis ground, if practical."

Here's a schematic of the encoder connection to the TI board:

Hope this helps.

Happy New Year!

Jan

From: andrea271068 
Sent: January 17, 2021 03:08
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] How to wire Encoder Omron E6B2-CWZ6C 1024 P/R to the TI launchXLF280049C (#124)

Good morning,
I bought the suggested Encoder, now I have to wire to the TI board, I have 5 wire + the Shield, but I can't fine the wiring diagram and I'm scared to mess up everything, the brown is simple because is the 5 volts, but for the rest, I don't know from where to start, also the shield has to be connected to the GND pin on the TI board?

The Omron has 5 cables:
Brown= 5V > this is clear where to be connected.
Blue= OV (Common) > Where I should connect this? (have to be connected to TI board GND pin??)
Shield= GND > Where I should connect this? (but on the TI board there are only 5 pins)
Black = OUT A ? > Where I should connect this? (Does this correspond in the TI board to 1A or 1B or 1I?)
White = OUT B > Where I should connect this? (same as above).
Orange = OUT Z > Where I should connect this? (same as above).

TI Board:
GND Pin > where?
5VO Pin
1I Pin > Where?
1B Pin > Where?
1A Pin > Where?

I'm a little lost here, I can't find in youtube or here the wiring diagram to be followed.

Please somebody can be so kindled to help me, thanks

andrea271068, 2021-01-17:
Hi, Jan, thanks a lot to help me in this venture.

I check the chart but is very generic,

Okay I will connect the shield to the chassis ground, and this is also clear now.

Brown to 5V > OKAY
Shield to chassis ground, OKAY

But for the rest?

Blue= OV (Common) > Where I should connect this? (must be connected to TI board GND pin??)
Black = OUT A ? > Where I should connect this? (Does this correspond in the TI board to 1A or 1B or 1I?)
White = OUT B > Where I should connect this? (same as above).
Orange = OUT Z > Where I should connect this? (same as above).

[cid:image002.jpg@01D6ECC5.149C7A40]
I check also (I think was your video in YOUTUBE) but the image move and I can’t recognize cables color in TI Board connection, ☹

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

COMPOSAD s.r.l.

Sede Legale: Viale Lombardia, 29 - 46109 Viadana (MN) - Italy

ph: +39 0375 783611 | fax: +39 0375 7836695

Sede Operativa: Via Lombardi 1 - Z.I. Gerbolina - 46019 Viadana (MN) - Italy

C.F - P.IVA/VAT code e R.I (IT)02317920201

www.composad.com<whttp://www.composad.com>

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

Il presente messaggio di posta elettronica e i relativi allegati sono strettamente confidenziali, soggetti alle leggi vigenti, pertanto se ricevi questo messaggio di posta elettronica per errore, per favore contattaci: e' assolutamente vietato usare, copiare, o divulgare il messaggio e ogni informazione in esso contenuta, tranne che al destinatario originario. Le comunicazioni Internet non sono sicure e non ci riteniamo responsabili per eventuali abusi di terze parti, per alterazioni nel corso della trasmissione, né per danneggiamenti causati da virus o altre anomalie a noi non imputabili.
----
This message and its attachments are addressed solely to the persons mentioned above and may contain confidential information. If you have received this message by mistake, be informed that any use/copying/forwarding of any part of its content is completely prohibited. Please return it immediately to the sender and delete the message. Internet communication is not secure and therefore we are not responsible for any misuse of it by third parties; Alterations during the transmission or damages caused by viruses or other anomalies are not attributable to us"

'

JonWoellhaf, 2021-01-17:
The schematic I included in my last email clearly shows the connections, but the schematic doesn’t appear in GitHub under your issue. I don’t know why nor how to include it.

It shows the connections to J12 EQEP1 as:

    A Black pin 1 EQEP1A
    B White pin 2 EQEP1B
    I Orange pin 3    EQEP1I
    5 VDC Brown pin 4 +5V0
    GND Blue pin 5 GND
    Shield chassis ground

Jan

andrea271068, 2021-01-17:
Jan many thanks, for 1000 time.

Now I can go forward.

Have a great Sunday Jan,
now I go to my home hobby shop to do try the connection.

Thanks again

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

TreatHull, 2021-01-19:
The Omron instructions cover several different encoder models. I've circled the connections for the encoder you said you had.
[image]