clough42/electronic-leadscrew #125: Problems driving hybrid servo motor using James's kit

candera, 2021-01-25:
Posting this here because I'm not sure where else to go and I'm a bit stuck.

I've been working on getting the Electronic Leadscrew set up using James's kit. I've got the display working after a bunch of dumb mistakes (whoops plugged the boost board into the wrong set of headers), and when I spin the encoder it shows a reasonable RPM, so I think that's working as well. 

What I can't get working is the hybrid servo motor I'm using. I've triple-checked all the wires are correct and continuity is good. But when I spin the encoder the motor just sort of stutters and doesn't do much. Sometimes it turns a bit, but it's definitely not doing what it's supposed to.

I've got some basic electronics knowledge, but I'm pretty new to motor control. I've never used a hybrid servo before. I bought this one: https://www.ebay.com/vod/FetchOrderDetails?itemid=124061452016&transid=2351269567002&ul_noapp=true

I've tried every combination of DIP switches and inverting the drive pins I can think of. I've got it wired the way James' instructions suggest, with +5V to the PUL+, ENA+, and DIR+, and the STP-, EN-, and DIR- outputs of the boost board to the corresponding inputs on the motor driver. I *think* perhaps there's something up with that, though, because when I put a logic analyzer on the pins, it shows a boatload of noise. Like, even if I'm not spinning the encoder I'm seeing lots of high/low and low/high transitions on the step and direction lines. 

At one point I was able to get a clean signal on the logic analyzer, where I'd see pulse trains on STP- like I'd expect, including the direction line flipping appropriately. I can't quite remember at this point what I was doing, though. I may have been measuring the pins with all the outputs inverted.

So I wonder if I should be driving it differently? Like, should I be driving the PUL+ pin instead of the PUL- one? Again, totally new to motors. Sorry I can't put my question more intelligently - I'm a n00b. Any input would be extremely appreciated.

xfuzz, 2021-01-26:
I too am having trouble getting the servo to spin. I requested help here too. There is this one fellow Jon  who seems to be very knowledgeable and has helped others but has not committed on my post as of yet. I am praying he is well and will chime in. James has not commented in some time. I follow his channel and he is busy with his new mill. Maybe he will look and take pity on us. Good luck sorry I have no clue how to help you.

candera, 2021-01-26:
Right now I can't tell if it's noise or a bad motor. The logic analyzer seems to show a clean signal when I separate the cables, but it might be triggering on a level that's wrong for what the motor controller is expecting. There is certainly plenty of noise when I measure with the cables in close proximity. I don't have an oscilloscope to see what the signal actually looks like. My next step is to wire up a simple motor driver to test the motor in isolation so I can tell whether it's the control signal or the motor itself.

xfuzz, 2021-01-26:
I have no clue how to do any of that. I am thinking it is the ti launchpad itself or the servo encoder as everything else seems to be ok.  I'll give it a few more weeks to see if anyone sqn help me figure it out . If not I'll try to sell what I have to recoup some money and just use myntaps and dies a usual.

candera, 2021-01-26:
Well, I'm pretty clueless myself, but if you want to give any more details on the problem you're having I can maybe throw some ideas your way.

xfuzz, 2021-01-26:
Thanks I appreciate any ideas. I believe I covered everything in item #1.1 just a few post below. Nothing has changed in my setup. I think it might be something simple but no idea what that might be.

candera, 2021-01-26:
Can you post a picture of your setup, particularly the hookups to the motor controller?

xfuzz, 2021-01-26:
Well sure if I knew how. I tried to do that in my original post but couldn't figure how to. I even looked in the get hub help section.
I got you replys in a email maybe I can reply with attachment pics. I'll try now.

xfuzz, 2021-01-27:
I sent a reply via email with pictures. I don't see it here in gethub. So I have no idea if you received it.

xfuzz, 2021-01-27:
The file came back to large. I just sent 3 separate ones mayb that will get to you

candera, 2021-01-27:
Does not seem like it came through. That's a shame. I'm sure it's quite frustrating - I know computers quite well and I'm fairly frustrated with this project at this point.

For my part, I've ordered a new motor. A bit expensive but I think I can still return the old one if it turns out the new one works out. And if not, well, I guess maybe it's the power supply?

xfuzz, 2021-01-27:
Damn the bad luck. I sent several with one pic each to cut down on the
size. I know they can be posted on get hub just cant figure out how. I am
feeling very ignorant now. I do wish to thank you for the time you gave me.
Hope you get your working soon.

xfuzz, 2021-01-27:
Got the uploaded with a little help they are on under open items #121

xfuzz, 2021-01-29:
Did the new servo and encoder work for you?

candera, 2021-01-29:
I wound up buying a stepper instead of a servo. Arrived today. Hooked it up and it worked perfectly straightaway. Definitely a bad motor or bad driver. Given the weird behavior, my money is on the servo encoder being bad. Anyway the new motor works a treat. Hopefully I can return the bad one. 

I may still try to pick up a hybrid servo at some point, since I like the concept and I have some other projects I want to do, but a stepper should work fine for me and I've got it all hooked up.

closed 2021-01-29 by candera

xfuzz, 2021-01-29:
Congratulations! Good to hear. I am sending my ti board to Jon so he can
test it. That should narrow the search for a solution.

JonWoellhaf, 2021-01-29:
It seems several people have had trouble with the hybrid servos, but not with the steppers. Is there something simple we’re missing? If you can’t return the bad stuff, I’d love to examine it.

From: Craig Andera 
Sent: January 28, 2021 19:24
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Problems driving hybrid servo motor using James's kit (#125)

I wound up buying a stepper instead of a servo. Arrived today. Hooked it up and it worked perfectly straightaway. Definitely a bad motor or bad driver. Given the weird behavior, my money is on the servo encoder being bad. Anyway the new motor works a treat. Hopefully I can return the bad one.

I may still try to pick up a hybrid servo at some point, since I like the concept and I have some other projects I want to do, but a stepper should work fine for me and I've got it all hooked up.

candera, 2021-01-29:
Unfortunately I have a somewhat limited window to return it. Otherwise I'd be totally game for sending it your way.

There's a pretty good chance I will pick up another one as part of another project I have in mind, though. I'll try to remember to report back here on whether it works. If not, we can see about analyzing it.

Thanks a ton for all your help with this. I thought I was going bonkers, having tried everything I could think of. I'm glad it was just a bad motor, but it really helped me not feel so out in left field when I was troubleshooting.