clough42/electronic-leadscrew #126: Twisted Pair Question

Geoff-McCombe, 2021-01-26:
Hope this is the correct place to post as it's not really an issue but a question.

When running separate ground for each wire in a twisted pair, where should each ground be terminated at each end, just group them at each pcb? I have ordered a locking aviation style connector to enter the control box so I assume that it's best to take each ground straight through and continue the balanced line to the interface board.

Thanks for any help offered!

Geoff

mac651955, 2021-01-27:
I only connected mine at one end, the electronics enclosure end and went to chassis ground.  I was cautious about grounding anything to the electronics except where shown.

clough42, 2021-02-24:
In general, twisted pairs are used for differential signals.  I often abuse them to get *some* noise immunity for digital logic.  I typically use the solid color wires for my signals and the white wires for ground.  I typically connect all of the white wires to the ground at the source end, and use one of the white wires as my ground at the destination end.  This is how I have my control panel on my ELS wired.  I don't have anything connected to the chassis ground in this scenario, but since the metal box is screwed to the lathe casting, it is grounded.

closed 2021-02-24 by clough42