clough42/electronic-leadscrew #127: Sharing my implementation

JoeSchmoo, 2021-01-31:
Just sharing my implementation of ELS on Sieg SC4 / Little Machine Shop HiTorque 7500 lathe in case someone finds this helpful.
These are still a few finishing touches I need to put on it but it works without any issues so far.
Thank you James for making this open source.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

egapelr, 2021-02-03:
Thanks for sharing photos of your ELS install.  I have just finished the install on my Atlas 10F and will share some photos as well when I get the opportunity.  As with you the install went well and I thank James for his excellent work, videos and valuable information.

haiqu, 2021-03-24:
Nice work!