clough42/electronic-leadscrew #129: Servo not slowing with lathe

mac651955, 2021-02-10:
Just wondering if anybody else experiencing this.  Set my controller to 8tpi on slowest spindle speed,  turned spindle speed to maximum servo matched speed.  When I turned speed back down servo kept running at max.  Reset system and everything returned to normal.  I understand you would never turn threads at high speed but was wondering if anyone else experienced this or might have an explanation.

clough42, 2021-02-24:
Does it run forever?  Or does it eventually stop?

I wonder if it's stepping as fast as it can and getting behind.  If so, it will keep stepping until it catches up.  This is obviously not what you would want and would be a candidate for a code change to automatically stop under these circumstances.

mac651955, 2021-02-25:
James, When I turn the rpm’s down the servo runs at high speed for a couple of seconds then returns to normal.  It is probably maxed out and just takes the couple of seconds to cycle.  Did the same test at 11 TPI and had same results.  Not an issue that you would ever have under machining operations since I don’t think anyone would try to turn threads at 800 rpm.

clough42, 2021-02-25:
I can reproduce this behavior on my test bench.  You are correct; it isn't likely a real issue since no one will want to thread this fast, but I still think it should be fixed.  Any threads cut under these conditions will not be accurate.  It's also not good if the operator expects the leadscrew to stop and it doesn't.

This will happen whenever you ask the ELS to output step pulses faster than about 100KHz.  In my test configuration, this happens at 2700RPM when threading at 8TPI.  It could happen at slower speeds and finer pitches for systems with higher resolution on the stepper/servo.

Here's a screenshot of the logic analyzer output on my test bench:

[image]

The bottom two traces are the quadrature pulse inputs.  The green bar is the step output.  They're zoomed out so you see a solid bar instead of individual pulses, but you can see that I stopped the input and it took just over 1.4 seconds for the output pulses to stop.

I'm looking at this now, but it will likely take some refactoring.

There isn't an awesome solution.  **I think the safest approach is to have the code keep track of the delta between the desired and current position and stop the leadscrew if it's getting behind.**  This isn't ideal because stopping the leadscrew might scrap a part or break a cutter if it happens during a pass.  Then again, getting behind will cause the same problem, and the buffered steps could become a safety issue.

clough42, 2021-02-25:
I have a candidate fix on the dev branch.  If you know what that means, feel free to try it out.  I have only run it on my test bed at this point, and not on the actual lathe.

There is a new configuration item in Configuration.h to set the maximum step backlog.  If the system gets behind and this threshold is exceeded, the ELS will stop with a message flashing on the display ("TOO FAST / RESET").  To reset the system, stop the lathe and power cycle the ELS with the power button and it will start working again.

clough42, 2021-02-26:
I just released version 1.4.00, which fixes this bug.  Please let me know if this fixes the issue.

mac651955, 2021-03-01:
James,
That works.  Kicked out at around 500 rpm on the spindle set at 8 tpi.

Thanks
Mac

clough42, 2021-03-01:
Awesome!

That is slower than I would have expected.  I have to go to nearly 3000RPM
before I hit the limit.  I assume your drive resolution is quite a bit
higher than mine.

What's your setup (leadscrew pitch, stepper steps and microsteps, encoder
resolution)?

James

mac651955, 2021-03-01:
Leadscrew 8 tpi, 1000 steps, 12 micro steps, encoder is same as yours.  In fact all my components are the same as yours.  Running 3 to 1 servo to lead screw 24 tooth to 72 tooth.  I did have to invert my direction to get the leadscrew turning the correct direction.  Thought since it was three phase I could switch one phase but that didn’t work.

clough42, 2021-03-01:
That's 1000 steps on the servo, and 12 microsteps, also on the servo?  I'm
not sure I understand.  I'm running 1000 and 3 with a 3:1 drive.  With that
configuration, it sounds like you have a total of 12:1 between the servo
and the leadscrew?

That would explain why you're cutting out at 500rpm and I'm getting closer
to 2500rpm.

James

mac651955, 2021-03-01:
James,
I knew with a 8 tpi leadscrew I had to turn the servo 3 times to get 1 rev at the leadscrew when I set the controller to 8TPI.  I originally set the micro steps to 3 and the steps to 1000. With one rev of the encoder I didn’t get a full rev at the servo.  I increased microsteps to 4 and got one turn.  So I increased microsteps to 12 and got the 3 turns I needed.  I’m running a 24 tooth gear on the servo and a 72 tooth gear on the leadscrew.  There is a picture n GitHub of my setup. In issues.

Mac

mac651955, 2021-03-01:
Tried to send a video of it running but file to big for here.

clough42, 2021-03-01:
Interesting.  I see that your leadscrew has a gearbox.  How are your gear knobs set?  Is the gearbox running 1:1?  Do the input pulley and the shaft turn 1:1?

If they are, there's still a factor of 4 missing.

Is your RPM readout accurate?  If you're using the 1024-line encoder, and it's running 1:1 with the spindle (looks like this in the photos) then your encoder resolution should be set to 4096.

Actually, if you could paste in your entire Configuration.h file, that might help.

mac651955, 2021-03-01:
Here you go.
Configuration.h

mac651955, 2021-03-01:
Here you go And the gearbox is 1:1

[Configuration.txt](https://github.com/clough42/electronic-leadscrew/files/6064782/Configuration.txt)

clough42, 2021-03-01:
Thanks.  This all looks correct to me.  The only thing I can imagine is
that maybe your servo driver is configured for 4000 pulses per revolution.
Do you have the cable and software to look at your drive parameters?  Here
is how I have mine set:

[image: image.png]

clough42, 2021-03-01:
Looks like the image didn't get copied in from the email gateway:

[image]

mac651955, 2021-03-01:
This is the servo and controller that I have:
https://www.automationtechnologiesinc.com/products-page/nema23-closed-loop-stepper-motor-system-hybrid-servo-kit/hybrid-servo-drive-kl-5080h

mac651955, 2021-03-01:
I see quadrature steps per rev 4000 but pulses per rev 1000.  What advantages would I get if I changed my steps to 4000 and microsteps to 3?  More torque?

clough42, 2021-03-02:
That is exactly the same setup I have.  I still don't understand why you have to have a microsteps value of 12.

The encoder that comes in that motor should be 1000 lines per revolution.   This results in 4000 quadrature pulses per revolution, meaning that the controller can detect 4000 steps in one motor revolution.  The Pulses/Revolution value is the resolution that the controller will interpret from the ELS.  If it's set to 1000, then it will take 1000 pulses from the ELS controller to cause the motor to make one revolution.

For my setup, which seems identical to yours, the ELS has to send 1000 pulses to make the motor turn one revolution, but since the belt drive is 3:1, it has to send 3000 pulses to make the leadscrew turn one revolution.  That's why I have:

#define STEPPER_MICROSTEPS 3
#define STEPPER_RESOLUTION 1000

If you have STEPPER_MICROSTEPS set to 12, then the ELS is outputting 12000 steps for a single leadscrew revolution.  That doesn't affect torque--it just affects the rate at which you have to send the pulses.  That's why you're hitting the 100KHz limit at only 500RPM.

But I don't understand why you have to set it to 12.  From everything I can see, you should only need 3.

One other possibility is that your encoder resolution is off by a factor of four, but your photos show a 1024-line encoder, and the configuration file you uploaded showed 4096 for the encoder resolution, which should be correct.  And if your encoder configuration was off by a factor of four, your RPM readout would be off by a factor of four as well.

I don't get it.

Any chance you could post a screenshot of your servo driver configuration?  I don't know if I would notice anything you didn't.  It's still got me confused.

mac651955, 2021-03-02:
Not sure what you mean by servo driver config.  Do you mean how it is wired up?  Otherwise if it is driving you crazy you could always drive over on a weekend with test equipment and check it out.  Southwest Montana.  I will do anything within my capabilities to help you figure it out.  It works and I love it which is all that matters to me.
Mac

mac651955, 2021-03-02:
Just thinking out loud.  I really didn’t notice but are there dip switches on that servo controller?

mac651955, 2021-03-02:
After watching your video on stepper, servo for the third or forth time I think I have the answer.  Mine is out of the box with what ever they they configure it when they ship it.  You say in your video that you reconfigured yours to 1000 steps or something like that.   So out of the box it must be configured to 4000 steps per rev.  That is an old mans interpretation of what I think is going on.  Your way better educated on this subject. What do you think?

clough42, 2021-03-02:
That sounds right to me.  I thought it was set to 1000 from the factory, but I just looked back at the video, and I think you're right.  It's probably 4000 from the factory.

The only downside of having the resolution set to 4000 is that it means you'll be outputting 12,000 steps per revolution, and you'll hit the 100KHz sooner than you would if it were set to 1000.  That said, you're still talking about threading 8TPI at 500RPM, which is faster than you need to run it.

If it's working for you and it can go as fast as you need, I think you're good.

Thanks for all the info.  I will be publishing a video on Saturday and referencing an unnamed person who reported the issue.  I didn't want to use your name without permission.  I did say repeatedly in the video that this occurs for me around 3000rpm, but after this conversation, it seems that it can occur at much lower speeds if you have a high servo resolution.

mac651955, 2021-03-02:
I ordered the cables after I figured it out.  It may cause other unforeseen problems.  Use my name it doesn’t bother me.  I’m helping a couple other people on theirs so any problems like this is knowledge in my pocket.

mac651955, 2021-03-07:
James, Just finished resetting the parameters on my servo drive to the same settings as you provided previously.  First issue was it didn’t like the 4096 on the encoder resolution.  Changed it back to 4000 and that worked somewhat.  Reflashed microprocessor to 1000 steps and 3 microsteps.  Timing is perfect.  Turned revs up and servo stopped at 900 rpm on the leadscrew but no error message.  Checked servo driver and it had faulted out.  Several attempts with same results.  I’m thinking the first two lines on the parameters might be the problem but I don’t fully understand what they do.   What do you think?

clough42, 2021-03-07:
If the servo drive is faulting, it's probably just not able to generate
enough torque at the desired speed.  That shouldn't have anything to do
with the ELS.  If you're running the leadscrew at 900 rpm, that means the
servo is spinning at 2700rpm.  That's really fast for a closed-loop
stepper.  Most of them lose almost all of their torque when run that fast.

The encoder resolution needs to match your encoder.  If it's a 1024-line
encoder, it needs to be set to 4096 or everything will be off.  What do you
mean when you say "it didn't like the 4096"?

James

mac651955, 2021-03-08:
James, 
First off my servo driver is a ES-508, alittle research shows that the KL-5080H was rebranded as the ES-508.   Don’t quite understand how you could turn your servo at around 8100 rpm, 2700 rpm spindle speed with 3:1 ratio without faulting the driver but I only get 2,700 and yours didn’t fault the driver it just got behind in steps.
When I set the Encoder Resolution to 4096 on the servo driver it immediately faults out when I turn the spindle by hand.  On the specs for for the servo it list Quadrature digital signals: 4000 steps per rev, I am assuming that the driver has to match that.  I tested the factory settings, current loop Kp and Ki, and holding current while only changing Pulses/Rev and got the same results, driver faults out at 900 rpm/8tpi.  It works and I am happy with it.  There are just too many different combinations of drivers, motors, lathes etc, ect, ect for them all to work the same.

clough42, 2021-03-08:
Oh, I see the confusion.  Maybe I didn't explain well enough.

There are two things that can limit your speed.  The first is the limitations of the servo.  Most closed loop stepper systems lose most of their torque by 1000 or 1500rpm.  If you try to push past this point with a load (or even rapid acceleration) the driver will fault.  Mine included.  I can't go anywhere near 8100.  Mine dies at about the point you're reporting.

The second is the absolute step limit that the ELS will output.  This is the issue you first reported, and my comment is that with my settings, even at my max spindle speed of 2500rpm, I can't reach this point, so I had to reproduce it on my test bench.  On the real lathe, while it isn't possible to reach the 100KHz step frequency, it also isn't possible to go this fast because the servo will fault.

If the motor has a 1000-line encoder on it (I think it does) then yes, you have to set the "Encoder Resolution" to 4000 to match this, so the driver knows what to expect from the motor.  But you can use a different value for the "Pulses/Revolution" value, which controls how many pulses from the ELS controller will cause one rotation of the servo.  The driver translates between these.  I'm using 1000, and that's what I would recommend.  It'll keep you away from the 100KHz pulse limit.  You can see my settings, above.

As for the servo and driver, I'm currently using the KL05080H driver and NEMA21 LK-23-2N-1000 motor.  This setup has been working fine on my lathe, but it's probably just barely enough.  I haven't had an issue while cutting, but I would feel better with more torque margin.

I have parts ordered to experiment with two other options.  Automation Technology has another NEMA23 closed loop system now that claims 3.1Nm (439oz-in).  This one wasn't available when I bought mine, or I would have bought it instead.  The torque curve on it looks pretty good.  I spent a bunch of time searching to see if I could find another source for the motor from StepperOnline, or Leadshine or someone else, and I couldn't find it.  StepperOnline has a ton of stuff that claims 3Nm or even 4Nm "holding torque" but when you look at the torque curves, both produce much less torque than the Automation Technology options.

I also ordered a ClearPath servo that should produce quite a bit more torque than either of the closed-loop systems, but of course for a considerably higher price.

mac651955, 2021-03-08:
Thanks James that clears it up a bit.

Mac

mac651955, 2021-03-09:
James, Should I close this issue or leave it open for others to follow along.  Makes me no difference.

steelbarz, 2021-03-10:
> Leadscrew 8 tpi, 1000 steps, 12 micro steps, encoder is same as yours. In fact all my components are the same as yours. Running 3 to 1 servo to lead screw 24 tooth to 72 tooth. I did have to invert [...]

 I ordered the exact same part number off eBay, however, I missed the detail about the PR number being important on the listing I bought and got a 50.  I made it work and will be upgrading the encoder with one with 1000 PR or what I see on one of James' videos number-wise.

koenbro, 2024-06-25:
> Clearpath

Hi James any suggestions on how to wire up the Clearpath servo? Thank you !