clough42/electronic-leadscrew #130: Precision Matthews 10 inch variants installation

mac651955, 2021-02-11:
My install.
[image]
[image]
[image]

clough42, 2021-02-24:
Looks good.  Several people have mounted the servo inside the foot on the Grizzly, but it looks like the structure is a little different on the PM.

xfuzz, 2021-02-25:
I have the same lathe, PM1022V.  would love to see more pics.  Specifically what pulleys did you use and where did you purchase them. I am unclear especially about the spindle pulley how does it go on ? Are spacers required or does the pulley width need to be reduced. What belts did you use for both the omron and spindle as well as the servo and lead screw.. I would greatly appreciate your reply. I have all the rest hooked on  for testing and it seems to be working well. However its 19 degrees and I will have to wait several more weeks for a reasonable warm up. So once I am armed with your info I can order and hit the ground running  Thanks.

mac651955, 2021-02-25:
Xfuzz, I have been doing a email collaboration with Blondihacks so if you could forward your email to me I could just forward all of the emails I sent to her with the info you are requesting.  Additionally there is a YouTube video on the PM variants, just search electronic lead screw, that is where I got my idea from.  Just don’t order the parts he referred to they won’t work.

mac651955, 2021-02-25:
Don’t want to load GitHub down with all those emails.

xfuzz, 2021-02-25:
So much apreciated  my email

 xfuzz1993@ gmail.com

I seen the YouTube videon you mentioned He did not use the Clough42 boost
board. I did 3d spring the servo mount he has on thinigiverse though.

steelbarz, 2021-03-10:
> Xfuzz, I have been doing a email collaboration with Blondihacks so if you could forward your email to me I could just forward all of the emails I sent to her with the info you are requesting. [...]

Someone complained the parts don't work?  Mine did I got all of them today in the mail.  I followed the instructions.  There are tons of details and the wiring has to be perfect.  We're talking precision here this is the most powerful tool in the shop.

dsgroninger, 2021-03-16:
mac,

Where in the code did you make the adjustment for the feed rates?  The Precision Matthews lathe has the gearbox for both Z axis feeds and cross feeds separate from the half nuts, and that gears things way down.  Did you just edit the feed tables, or did you find a more elegant solution in the calculations somewhere?

dsgroninger, 2021-03-17:
If I've measured correctly, it looks like the gearing ratio is about 1:3.3.  On turn of the lead screw with the feed gearbox engaged moves the tool 0.0375", so about 1/3.3 as fast as the half nuts engaged.

dsgroninger, 2021-03-17:
Seems to me that if I take these lines in tables.cpp:
#if defined(LEADSCREW_TPI)
#define THOU_IN_NUMERATOR(thou) ((Uint64)thou*LEADSCREW_TPI*STEPPER_RESOLUTION_FEED*STEPPER_MICROSTEPS_FEED)
#define THOU_IN_DENOMINATOR(thou) ((Uint64)ENCODER_RESOLUTION*1000)
#endif

and edit to this:
#if defined(LEADSCREW_TPI)
#define THOU_IN_NUMERATOR(thou) ((Uint64)thou*LEADSCREW_TPI*STEPPER_RESOLUTION_FEED*STEPPER_MICROSTEPS_FEED*10)
#define THOU_IN_DENOMINATOR(thou) ((Uint64)ENCODER_RESOLUTION*1000*3)
#endif

that would give me the 3.33 to 1 ratio I'm looking for and keep the math all in integer.  I then do the same in the metric definition section.

Is that what you did?

dsgroninger, 2021-03-17:
Just tested the changes in the last post and that worked perfectly.  Feed rates are updated and accurate, thread rates unchanged and still accurate.  Unfortunately, I see that all the asterisks for multiplication were stripped out in the text of the post.  :(  

What I did was add an additional multiply by 10 in the numerator for both feed inches and metric, and multiply by 3 in the denominator to get a 3.33 to 1 ratio while sticking to integer math with a 10 / 3 ratio.

I did see that I hit James's new speed limit somewhere near 900 RPM on the spindle while in the 0.040" per turn mode.  Not a problem on my lathe, I don't believe that it's capable of a cut like that anyway.  It did stop the feed as expected, and reset with toggling the power button on the control panel.

Also needed to un-comment the reverse direction define in configuration.h to suit my setup.

All set now for feeds and threads.

mac651955, 2021-03-17:
DS,
I didn’t adjust anything on feed rates.  I concentrated on getting the servo to turn exactly 3 times to one rev of the spindle when the ELS was set to threading at 8 tpi which is the lead screw pitch.  My gearbox is set with both knobs to the left which is one to one pulley to lead screw.  My servo to lead screw ratio is 3:1, 24 tooth gear to 72 tooth gear.  I used the HTD 15 mil wide gears and pulleys.  I was assuming, always dangerous, that if the threads were correct the feed rates were correct.  Just ran downstairs to the shop and put a dial indicator on the carriage and It moved exactly .00025 with one rev of the spindle when the ELS was set at .010.  That indicates there is a 4:1 ratio leadscrew to carriage.  Will have to do some digging to determine how to have the ELS show correct feed rate.  I know the cross feed rate is even a higher ratio because it fed slower even with the change gears.  I know we probably can’t get the ELS to show cross feed rates but can extrapolate that once we get the carriage feed rate to show correctly.

mac651955, 2021-03-17:
DS,
What feed rate did you have set on the ELS when you did your measurements?  You also are assuming incorrectly that I know anything about coding.  LOL. I too hit a max speed around 900 when ELS is set to 8 tpi but we have determined that it is not the same Max speed that James corrected but a loss of torque in the servo.  I don’t get the message on the ELS, the servo driver faults out.  I too had to change direction in config file to work on my lathe.  3.33:1 seems really odd but you could be correct.  I will do some further checking but I got 4:1 as I mentioned previously.  I tend to not be so scientific but use a feed rate that cuts well.  I hope I can figure out your coding with the asterisks missing.  If not you will hear me asking for help.

dsgroninger, 2021-03-17:
MAC,

It took  me some fiddling to figure out my gear ratio.  There is considerable backlash through the gearbox setup, but I did something similar to what you did with the dial indicator, but I used the DRO.

When James designed the software, it looks like he assumed that all movement would be done using the half nuts directly on the lead screw, which for us is 8 TPI and set in configuration.h.  Everything worked fine in that mode, and one turn of the lead screw is 0.125" of travel.   When I disengaged the half nut and engaged the power feed gearbox on the saddle, now one turn of the lead screw made on average 0.0374" or thereabouts, which works out to a 1:3.33 ratio.  That means that when I'm in feed mode, I'd like to be spinning the lead screw 3.33 times for each 0.125" of feed, or 3.33 times faster than Jamse's calculations would yield.

When I found how he defined the calculations in tables.cpp, it came clear where to make the change the best way.  There are four tables, one for inch thread, one for inch feed, metric thread and metric feed.  The tables have a line for each rate, and right above each table are #defines with the numerator and denominator for the speed calculations.  Even though in configuration.h it looks like James has set to use floating point math, each of the numerator and denominator macros seems to use care to work in integer math.  That's why I didn't simply multiply the numerator by 3.33.  However, 3.33 is conveniently 10 divided by 3, so multiplying the numerator by 10 and denominator by 3 does the same thing.  Attached is a copy of my tables.cpp file, with a .txt extension so the forum would take it.  Have a look at my comments and the two lines under each set of comments to see the change - once for the inch feed and once for metric feed.
[Tables.cpp.txt](https://github.com/clough42/electronic-leadscrew/files/6155633/Tables.cpp.txt)

dsgroninger, 2021-03-17:
Well, shoot.  After all of that, I remembered these couple of lines in configuration.h:

// Separate step and microstep settings for feed rates.  Redefine these if your
// lathe has a separate feed drive train with a different ratio.
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION

That is the way that James intended we deal with this problem of a separate rate for feeding and threading.  The only hitch here is that both of these factor in the numerator of his calculations.  Again, since we need to multiply by 10 and divide by three, the easy path here is this:
#define STEPPER_MICROSTEPS_FEED 1    // for threading, this is 3 for us for the timing belt pulleys, so divide 3/3=1
#define STEPPER_RESOLUTION_FEED 10000   // normally 1000 for us, multiply by 10 here.

That gets us our 10/3 ratio, and is far fewer lines of code to change.  It carries through Jame's design better as well.  As they say, many ways to skin a cat.  I'll test this later today and let you know the results.

dsgroninger, 2021-03-17:
Forgot to say, if making the above change to configuration.h, we'd leave tables.cpp as we get it from James.

dsgroninger, 2021-03-17:
My apologies, I'm posting like Robin Williams doing a stand up routine, stream of consciousness.  I'm used to having these discussions standing at a whiteboard with another engineer in my team, and we've worked together for 25 years.

Anyway, for the PM lathes and our motor setup, most of us have these lines in configuration.h set thus:
// Steps and microsteps
#define STEPPER_MICROSTEPS 3
#define STEPPER_RESOLUTION 1000

And by default, these two lines are:
// Separate step and microstep settings for feed rates.  Redefine these if your
// lathe has a separate feed drive train with a different ratio.
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION

This uses the same step settings for thread and feed.  With the Precision Matthews saddle gearing, we need to multiply the feed rate steps by 3.33 to get things to work out.  Since that's 10/3, and our micro steps for the timing belt pulley ratio is 3, multiplying 3 by 10/3 is simply 10.  So changing this one line of code:
// Separate step and microstep settings for feed rates.  Redefine these if your
// lathe has a separate feed drive train with a different ratio.
#define STEPPER_MICROSTEPS_FEED 10
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION
 
has the effect we want, needing 10000 total step pulses per revolution of the spindle and moving the lead screw one rev.

There are no changes needed anywhere else in the code, jus this one line changed in configuration.h to accommodate the gearing in the PM saddle for both saddle travel and cross feed.

I tested this approach just now, and it works perfectly, giving the same results as my previous set of changes.  This seems much more in keeping though with the spirit of James's code design, and keeps the changes to one line in configuration.h.  Luckily the math worked out to a nice, round integer ratio that fit with our other settings.

mac651955, 2021-03-17:
DS,
Did you hit the speed limit at a lower RPM?

dsgroninger, 2021-03-17:
I'll try to get a more exact number this evening.  I think it was in the high hundreds.  I have the lathe belt set for 50 - 1000 rpm at the spindle.  I did have feed set all the way up on the highest inch setting, 0.040" per turn.  It tripped out somewhere near the top speed.  That's totally unrealistic as a cutting setup.  At normal cut rates we shouldn't have any issues.

If it becomes an issue, next step would be to change our motor controllers to something less than the 1000 steps per turn, and make the corresponding adjustments in the defines in configuration.h.  For now, I don't think that we'll need to do that.

xfuzz, 2021-03-17:
Mac, I have my cable and usb adapter and software installed. Was going to do the changes you offered but now I see there is a new issue. So once again I need your help I see the discussion regarding adding and changing lines in the files  to account for the PM lead screw timing. Do you have all that figured out? If you could provide it and where and how to input it I would be most appreciated of your kindness

mac651955, 2021-03-18:
Xfuzz,
The changes to the servo driver remain the same.  James posted his settings in my other thread about servo not slowing with lathe.  Match his settings with the exception of encoder resolution, 4th line down from the top, and that should be 4000.  The other change is in config h file, which is what we have been discussing here. You will have to make that change in code composer studio.  Under the Steps and microsteps lines you will find a line STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS   just add a 10 to the end of this line with a space.  Save the file and reflash your microprocessor.  That will adjust the actual feed rate with what is shown on the control panel.

xfuzz, 2021-03-18:
I did this and get 44 errors after I click the hammerto compile
// Steps and microsteps
#define STEPPER_MICROSTEPS 3
#define STEPPER_RESOLUTION 1000

// Separate step and microstep settings for feed rates.  Redefine these if
your
// lathe has a separate feed drive train with a different ratio.
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS 10
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION
I take out the 10 and no errors. What am I doing wrong

dsgroninger, 2021-03-19:
xfuzz,

mac got you 95% of the way there.  Where you have this line:
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS 10

change it to this:
#define STEPPER_MICROSTEPS_FEED  10

Everything will compile, and it you have your lead screw pitch set to 8 TPI (that's right for the PM-1022 and 1030), all of your feed rates and thread rates should be correct.

dsgroninger, 2021-03-19:
Big evening in the shop this evening.  I finished my stepper motor mount and have everything up and running.  The only part of the lathe that I had to modify was to trim about 1/2" x 3" notch in the bottom surface of the gear train guard sheet metal.  It doesn't show when assembled and there's no way to get fingers in there, so I'm OK with that.  I can dismount the encoder, leave the stepper in place and be back to the old change gears in minutes if something breaks in the ELS.

I did confirm the feed and thread rates that result from the code settings above are correct.  Thread rates are using the half nuts directly on the lead screw to move the saddle, while the feed rates are using the feeding shift lever on the saddle.  That's why the need for the different rates.  For the feed rates, at the highest rates of 0.040" per turn of the spindle, Jame's speed limit does kick in right around 900 RPM at the spindle.  No harm here, the feed just stops until the spindle is stopped and our ELS controller reset by pressing the power button on the control keypad twice - once to turn the stepper off, then another time to turn back on.

**_Now, a really important caveat._**  Just because the ELS setup will spin the lead screw fast enough to do a 0.040" per turn cut, that does **NOT** make it a good idea.  I can tell you that at 900 spindle RPM, that lead screw is spinning at frightening rate.  I went back and checked the lathe manual, and the fastest feed rate using coarse feed change gears is just 0.015" per turn.  The ELS has a couple of feed settings available that are just not appropriate for our class of lathe.

If we stick to 0.015" per turn or slower, we should never hit the speed limit.

Anyway, the math is all working out in the code, and I have a running setup.  Next step is to clean up all the borings from making the mount and put tools away, then this weekend I'll work on a writeup of the build from front to back and get that posted.

xfuzz, 2021-03-19:
Thank you . That worked like a champ. I am still waiting on a pully that
should be here Saturday. If it,s warm enough in the garage I will start the
install. My much appreciated thanks to all of you.

xfuzz, 2021-03-19:
Oh I forgot, one more thing .where do I change the servo direction in
config h I will be placing it in the cavern in the lathe bed like Mac did
      #define INVERT_ENABLE_PIN true                    I think i change
the true to false is that correct?

mac651955, 2021-03-19:
That is correct.

mac651955, 2021-03-19:
No wait on that response.

mac651955, 2021-03-19:
Xfuzz, had to look at code on my computer to make sure:

change
// #define INVERT_DIRECTION_PIN true
To
#define INVERT_DIRECTION_PIN true

xfuzz, 2021-03-19:
Got it Thank again

dsgroninger, 2021-03-21:
mac, James,

Please have a look at the attached writeup that I did on my implementation and give me your honest feedback.  Would this be useful to someone trying to implement the ELS on a PM lathe, or too much?  Did I leave out anything critical?
[Clough42 Electronic Leadscrew.pdf](https://github.com/clough42/electronic-leadscrew/files/6176695/Clough42.Electronic.Leadscrew.pdf)

mac651955, 2021-03-21:
DS,
Great write up and pics to clarify any areas of concern.  Although I chose to eliminate the change gears your implementation works as well.  One thing I did notice once I got mine up and running without all those gears meshing and spinning was the lathe runs much quieter and seems to have a tad more power.  My only concern with your installation is grease from the change gears you left installed interfering or contaminating electronics parts and the belts.  Believe you covered all aspects of the install and provided valuable information to those embarking on this project.  Amazes me that two totally removed installations used numerous identical ideas and materials.

dsgroninger, 2021-03-21:
mac,
I don't think that I really had a original idea in there anywhere other than making it work with materials that I had.  It's all stuff that I found in reading these threads, ThingiVerse and YouTube.  Everything is out there if one looks long enough.  My hope was that I could get it all into one document, with pointers to more details and sources for everything and maybe give a little something back to the community.

I take your point on flung grease.  I have probably overdone the grease in there in the past, and cleaned that up when doing this project and tried to be a bit more judicious with this stack.  This setup is a lot quieter, even though there are still gears left.  I think part of that is there is almost no torque load on them, just the friction of two bearings.

One other thought that I had was looking to the future - what happens a couple years from now when something breaks?  I assume most of the big parts like the encoder, motor and controller would be readily available in some form, but what of the Launch, James's interface board and the display?  I ordered a second set of the Clough42 boards last evening, and will send Digi-Key an order today for another Launch.  Since that's basically an evaluation board for the processor, those rarely seem to be available for  more than a couple of years, so it'll be on us to keep spares.

dsgroninger, 2021-03-25:
Cut my first threads tonight.  I started to work on a drawbar for an ER40 collect chuck that's been laying in the lathe drawer for a year.  MT4 to fit the PM-1030 spindle, 5/8-11 drawbar thread.  Cut, test fit, cut, test fit... ended up with a nice thread, but...  

As I was turning the chuck onto the freshly threaded rod, it was making an oddly eccentric pattern.  Turns out my fine import ER40 chuck has the drawbar hole drilled and threaded off angle.  It's not horrible, but obviously visible when threading the two together.  Maybe 3 or 4 degrees off axis.  Harrumph!

mac651955, 2021-03-25:
I did the same with a Chinese ER-32 chuck, end result was about 3 thou runout.  Just made an arbor for some 7/8 arbor size involute cutters and cut a real nice 13/16 x 20 threads external and internal.

MattA986, 2021-03-31:
mac651955, I also have a PM10 variant lathe and I'm very interested on how you drive the encoder with the spindle.  Would it be possible to also forward me the emails you sent to Quinn, please? My email is matt98699@yahoo.com I would really appreciate it! Thanks!

dsgroninger, 2021-03-31:
Matt,

Here's a writeup that I did on my implementation on a PM-1030.  Hopefully you'll find some useful tips in there.  One thing that I noticed in one of Quinn's videos is that the banjo on her lathe is different than mine.  Turns out hers is about two years older, from 2017 as compared to 2019 for mine.  The mount could be simpler with hers, but would need a tensioning mechanism of some sort to keep belt tension as there would be just the one mounting bolt.

Dan
[Clough42 Electronic Leadscrew.pdf](https://github.com/clough42/electronic-leadscrew/files/6237144/Clough42.Electronic.Leadscrew.pdf)

BlueManRising, 2021-03-31:
You have no idea how much time you just saved me, my 1030V is showing up
later this month and was wondering about the settings people used for the
gear box/cross feed.

Just a massive thank you is in order here!

Nathan

xfuzz, 2021-03-31:
Execelent write up. Thank you for doing it. I am sure it will be most
useful to other PM owners.

dsgroninger, 2021-03-31:
You are all most welcome.  Any suggestions on where/how I might post it to be easier to find?  I don't want to steal anyone's thunder - mainly just trying to wrap up all the details I could to make it a one stop shop for the PM equipment.

dsgroninger, 2021-04-01:
I put my design for the LaunchPad mount and large hole/small cable grommet on ThingiVerse along with my writeup.  It's at  https://www.thingiverse.com/thing:4813034/files  

I joined the Hobby Machinist forum and will try to post there as well.  Looks like they have some limits on posting until a history is established, so not sure how quick I can get things posted there.