clough42/electronic-leadscrew #131: Encoder Problem?

JPL-Dx, 2021-02-12:
Trying to install the ELS on a 14 X 30 Hendey Lathe.
Per bench test, everything with the ELS Controller seems to work except the RPM display.
Spinning the encoder shaft does not change anything.
Stepper Servo Driver is powered up without any faults.
The servo motor is locked with power on, can be spun with power off.
I have completed all suggested trouble shooting and  looked through all posted issues hoping for enlightenment, all without success. 
 Continuity Checks all OK
 Voltage Checks all OK
 Wiring Checks all OK

I tested my encoder using an LED and all channels seem to be generating a pulse.

This should be simple, what am I missing?

Stumped!

mac651955, 2021-02-17:
Double check your dip switch settings first.  Flip them back and forth several times.  Second double check your solder connections.  Connection issues seem to be the root cause of the problem you are encountering.

JPL-Dx, 2021-02-19:
Thanks for the suggestions. My encoder is no good.  Checked it with a scope and discovered the output signal is under 1 volt.  That is not high enough to trigger anything.  I have another one on order.  I am pretty sure that will solve my problem.  Waiting on delivery.

JPL-Dx, 2021-03-08:
Finally received the new encoder. Hooked it up and it works as expected. Next step is to install on the lathe.

closed 2021-03-08 by JPL-Dx