clough42/electronic-leadscrew #132: Debug problem, what a sad day, I don't know waht to do.

andrea271068, 2021-02-14:
Hi, finally today I complete my set-up, I was so happy, the time to try everything was ready.
I run code composer studio 9 on Mec Book.
I click the hammer, the debug, after that, I click the bug icon, but there the problem comes, they force me to download an update.
The control panel never show-up, when I try to do again the debug, another message shows up. 
I'm very lost, I don't understand please somebody can help me, I'm not a programmer, I use a computer every day but as a normal user, for the business not as a programmer.
Really I don't know how to fix this issue. 

Please check the attached photo, may somebody can help me to solve this thing. 

[image]
[image]
[image]
[image]
[image]
[image]

andrea271068, 2021-02-14:
I forgot to add, I click resume, but when I click nothing happens, and if I try to do resume gain a message shows up and says that I have to complete the debugging process, really don't know??
Can be a wrong connection? can be that I have to do something on the program?

poulbo, 2021-02-15:
You have to update XDS110 firmware with xdsdfu.exe at the suggestions  location (search for the location) 
also read the TI. 's pdf file

andrea271068, 2021-02-16:
Many Thanks,
I can’t find xdsdfu.exe, or I find many things regarding XDS110 firmware on TI, please can you send to me the right link?, to many information and options find.

Waiting for your kind feedback, thanks is advance.

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

COMPOSAD s.r.l.

Sede Legale: Viale Lombardia, 29 - 46109 Viadana (MN) - Italy

ph: +39 0375 783611 | fax: +39 0375 7836695

Sede Operativa: Via Lombardi 1 - Z.I. Gerbolina - 46019 Viadana (MN) - Italy

C.F - P.IVA/VAT code e R.I (IT)02317920201

www.composad.com<whttp://www.composad.com>

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

Il presente messaggio di posta elettronica e i relativi allegati sono strettamente confidenziali, soggetti alle leggi vigenti, pertanto se ricevi questo messaggio di posta elettronica per errore, per favore contattaci: e' assolutamente vietato usare, copiare, o divulgare il messaggio e ogni informazione in esso contenuta, tranne che al destinatario originario. Le comunicazioni Internet non sono sicure e non ci riteniamo responsabili per eventuali abusi di terze parti, per alterazioni nel corso della trasmissione, né per danneggiamenti causati da virus o altre anomalie a noi non imputabili.
----
This message and its attachments are addressed solely to the persons mentioned above and may contain confidential information. If you have received this message by mistake, be informed that any use/copying/forwarding of any part of its content is completely prohibited. Please return it immediately to the sender and delete the message. Internet communication is not secure and therefore we are not responsible for any misuse of it by third parties; Alterations during the transmission or damages caused by viruses or other anomalies are not attributable to us"

'

poulbo, 2021-02-16:
Hi Andrea

 

I have the same problem with mine, and  I just do what they tell to do,

Here is the link  C:\ti\ccs930\ccs\ccs_base\common\uscif\xds110

I am using   firmware_3.0.0.14.bin     

The  XDS110SupportReadMe.pdf  tell you how to do it.

 

Hope it’s works

 

poulbo

 

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg] 

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

andrea271068, 2021-02-16:
Thanks, but the link doesn’t work >
C:\ti\ccs930\ccs\ccs_base\common\uscif\xds110

May there is an error and the typing

But I have to buy this hardware and install it?, if yes, how you did it? Please.

[cid:image002.jpg@01D70445.AF57CB60]

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com<mailto:a.pisani@composad.com%3cmailto:a.pisani@composad.com>>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

poulbo, 2021-02-16:
Hi 

 

Andrea

Nothing  wrong with link , if you don’t have the xds110 directory ,something wrong with your installation .

Xds110 is the debugger coming with the Code Composer studio 9.3.0.

(I am using windows 10)

 

poulbo 

 

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg] 

[http://www.grupposaviola.com/intranet/loghi/piede.jpg] 

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg] 

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

andrea271068, 2021-02-16:
Soo, I don’t have to buy that device, nice? Because is already inside the LounchXL and CCS 9.3.0.

I should find the C:\ti\ccs930\ccs\ccs_base\common\uscif\xds110
Inside Code Composer studio 9.1.0 installed?

I run this software in Mac Book Apple, I think the directory is not C, may this is why I can’t find it, now I will find the directory letter for Apple.

I don’t have experience in programming, this is the problem, really, I walk in the dark, any help to see the light will be very welcome, other wise I think I spend a huge money for nothing, and I will be forcing to dismantle everything and put on the side, really I don’t know what to do.

Sorry for my way out, but I need a person that drive me step by step on this thing.

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com<mailto:a.pisani@composad.com%3cmailto:a.pisani@composad.com<mailto:a.pisani@composad.com%3cmailto:a.pisani@composad.com%3cmailto:a.pisani@composad.com%3cmailto:a.pisani@composad.com>>>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

andrea271068, 2021-02-16:
I findXDF110 inside the finder in mac book, and I found the folder XDF110 and also the readme file mentioned.

Now please what I must do?

[cid:image002.jpg@01D7044F.209370A0]

[cid:image006.jpg@01D7044F.209370A0]

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

andrea271068, 2021-02-16:
Dear Poulbo, this is the screen shoot, I don’t know the controller don’t go, but after I resume I don’t receive any error message.
I change cables from the controller to the logic, I disconnect all others connection from the servo to the driver and I disconnect also the encoder. Just to make sure that didn’t was a cable problem.
James in his video install the firmware in front his computer, and just the controller was connected to the logic (TI-XL +Interface board)
I download from TI, the XDS emulation software (Emu pack) for mac. And I try to download inside the 3.1.0

But this is the result. Check below screen shoot.

[cid:image001.jpg@01D7045C.7D010480]
Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[cid:image005.jpg@01D7045C.7D010480]

[cid:image007.jpg@01D7045C.7D010480]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

poulbo, 2021-02-16:
Hi Andrea

 

I can’t see any pictures ,the first mail you send was ok and with pictures off you  system

Something wrong now, mabe you have to talk with a Mac user,

 

poulbo 

From: andrea271068 [mailto:notifications@github.com] 
Sent: 16. februar 2021 12:13
To: clough42/electronic-leadscrew
Cc: poulbo; Comment
Subject: Re: [clough42/electronic-leadscrew] Debug problem, what a sad day, I don't know waht to do. (#132)

 

Dear Poulbo, this is the screen shoot, I don’t know the controller don’t go, but after I resume I don’t receive any error message. 
I change cables from the controller to the logic, I disconnect all others connection from the servo to the driver and I disconnect also the encoder. Just to make sure that didn’t was a cable problem. 
James in his video install the firmware in front his computer, and just the controller was connected to the logic (TI-XL +Interface board) 
I download from TI, the XDS emulation software (Emu pack) for mac. And I try to download inside the 3.1.0 

But this is the result. Check below screen shoot.

andrea271068, 2021-02-16:
This is the image, do the same also with window

Try to see now, please don’t leave me alone.

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

poulbo, 2021-02-16:
Hi Andrea

 

The first mail in this tread was ok with all your picture , but after that you are sending lot of stuff on Italia in your mail 

I don’t what to do with it, and there is no pictures, sorry.

 

Every time I answer you I hat cut all that stuff out from the mail.

 

poulbo

 

From: andrea271068 [mailto:notifications@github.com] 
Sent: 16. februar 2021 13:10
To: clough42/electronic-leadscrew
Cc: poulbo; Comment
Subject: Re: [clough42/electronic-leadscrew] Debug problem, what a sad day, I don't know waht to do. (#132)

 

This is the image, do the same also with window 

Try to see now, please don’t leave me alone. 

Andrea Pisani 
Export Manager

andrea271068, 2021-02-16:
Let’s try now.
I attached the photo here
May did you have a personal e-mail contact?

I found on the program the > C:\ti\ccs930\ccs\ccs_base\common\uscif\xds110
But now I have to invent how to manage 😊 I don’t know nothing about programming, I check also the PDF is like Chinese language for me, 0 😊

Andrea Pisani
Export Manager

Composad srl – Italy
Mobile Phone: 0039 335 70 98 102
E-mail: a.pisani@composad.com<mailto:a.pisani@composad.com>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

poulbo, 2021-02-16:
Hi Andrea

 

 

In the PDF file it  says:       

Mac and Linux systems require libusb to be installed. Go to libusb.info for more details.

(you have to read information in the file)

 

My problem is that I am not using the Mac system, and don’t  how I works, that’s  why I think you have to find a 

Mac user in this forum for more help, or get help from a Mac user in your country ,an if you also have to change parameters in the code,

Maybe a person who familiar about programming in cpp.

 

poulbo

 

From: andrea271068 <notifications@github.com> 
Sent: 16. februar 2021 15:24
To: clough42/electronic-leadscrew <electronic-leadscrew@noreply.github.com>
Cc: poulbo <poulbo@poulbo.dk>; Comment <comment@noreply.github.com>
Subject: Re: [clough42/electronic-leadscrew] Debug problem, what a sad day, I don't know waht to do. (#132)

 

Let’s try now. 
I attached the photo here 
May did you have a personal e-mail contact? 

I found on the program the > C:\ti\ccs930\ccs\ccs_base\common\uscif\xds110 
But now I have to invent how to manage 😊 I don’t know nothing about programming, I check also the PDF is like Chinese language for me, 0 😊 

Andrea Pisani 
Export Manager 

] Debug problem, what a sad day, I don't know waht to do. (#132) 

Hi Andrea 

The first mail in this tread was ok with all your picture , but after that you are sending lot of stuff on Italia in your mail 

I don’t what to do with it, and there is no pictures, sorry. 

Every time I answer you I hat cut all that stuff out from the mail. 

poulbo 

roblem, what a sad day, I don't know waht to do. (#132)

andrea271068, 2021-02-17:
Hi, Thanks, 
I update the firmware, but still stocked, the control panel didn't work. 

this is the page where I'm stuck after clicking the resume (blue arrow), nothing happens to the controller. 

I try using CCS in windows and in Mac computer, both are updated, but is the same. 

As you can see, from the attached screen shoot photo,  I don't have any error message, I just stuck.  

I think That I have to do something, but...

**I need Mr. James Help,** 
I can go forward, everything is installed but...

[image]

poulbo, 2021-02-17:
Hi Andrea

 

Well it seems that debugger part works now, do you have access to an oscilloscope, 

I want you to measure on pin STB, CLK, DIO on James PCB card, 

to see signals on all 3 pins, and if not move cable and measure again, 

if there is a signal now the control card is defective (luckily they are cheap) 

I have 2 control boards that are defective. 

 

poulbo

 

 <https://user-images.githubusercontent.com/75749307/108160197-ec0e0f00-70e8-11eb-97cc-e3ee989ae206.jpg> Screen Shoot debug

poulbo, 2021-02-17:
Hi Andrea 

 

I forgot to say I got your picture this time.

 

poulbo

andrea271068, 2021-02-17:
Thanks :) Pouldo, I don't have Oshilloscope but I have the multimeter tester, I can check the continuity, sure.

Wow two control panel defective..., the major problem is that takes a lot of time to arrive, I don't know why. 

Sure today after business when I back home, I will check.
I will let you know ASAP, thanks a lot. In a way, I prefer to know that the control panel is defective than having a logic problem that I can't solve because I'm completely out of programming competence.  
Thanks :) let's see

andrea271068, 2021-02-17:
Hi Pouldo, I check the continuity in the PCB card and also the connection within the PCB card and connection cable to the Led&Key 5 pin connector to the board, and is working too. 
The only strange thing is when I control the continuity just in the Led&key from 5 input pins and the related connection in the board if I test STB, CLK, DIO doesn't, blink, no sound, but maybe is normal.
I decided to buy again the PCB and Led&Key board, never mind. 
Pouldo is normal that the led light in the Led&key board is on, even if is just connected to the PCB and of course the 5V, sometimes if I move the cable this led light switch to off, the 5V connection from the transformer pin and the PCB is not so fixed, but the receiver is well soldered, very strange, for this reasons I decide to buy again this two components. Anyway when I try to debug that light in the Led&key was on, and also about that, I'm not so sure, is normal that the led light is switch on from the start?  
Please tell me what you think to, fill free to give me your opinion by your experience, 1000 thanks

poulbo, 2021-02-17:
Hi Andrea

 

It is normal for LED lights on the control board to always light up.

if you take 5 pin plugs off and measure on CLK, STB, DIO on James

board with your multimeter in 10 or 20V mode measure first between 

GND and 5V+ then of course it shows 5V, measure then between GND and DIO,

and it should show approx. 3.5V-4.8 V does the same with STB and CLK,

should also show approx. 3.5V-4.8 V. If this is ok try again with the plug inserted.

You should also now see the same measurements. If some of the 3 pins are close

to 0V- 0.5V there is something wrong with the control board. 

One of my control boards pulled the DIO connection to 0V. input on TM1638 chip

was short-circuited. 

 

You can buy control board in Germany and you will get it faster.

 

Yours sincerely

poulbo

 

From: andrea271068 <notifications@github.com> 
Sent: 17. februar 2021 19:22
To: clough42/electronic-leadscrew <electronic-leadscrew@noreply.github.com>
Cc: poulbo <poulbo@poulbo.dk>; Comment <comment@noreply.github.com>
Subject: Re: [clough42/electronic-leadscrew] Debug problem, what a sad day, I don't know waht to do. (#132)

 

Hi Pouldo, I check the continuity in the PCB card and also the connection within the PCB card and connection cable to the Led&Key 5 pin connector to the board, and is working too.
The only strange thing is when I control the continuity just in the Led&key from 5 input pins and the related connection in the board if I test STB, CLK, DIO doesn't, blink, no sound, but maybe is normal.
I decided to buy again the PCB and Led&Key board, never mind.
Pouldo is normal that the led light in the Led&key board is on, even if is just connected to the PCB and of course the 5V, sometimes if I move the cable this led light switch to off, the 5V connection from the transformer pin and the PCB is not so fixed, but the receiver is well soldered, very strange, for this reasons I decide to buy again this two components. Anyway when I try to debug that light in the Led&key was on, and also about that, I'm not so sure, is normal that the led light is switch on from the start?
Please tell me what you think to, fill free to give me your opinion by your experience, 1000 thanks

—

cad-quark, 2021-02-19:
Andrea271068.   I do not know enough about this design to be dangerous but . . .    
 I got the same message on a Windows10 desktop, and on a Windows8.1 desktop.  It also told me “Target not found” which persisted after several reloads of CCS, and ELS 1.301.   In addition, the USB device was not being recognized by either machine.  In Windows, the system reports a USB device with a sound on insertion, and a different sound as its removed.  Neither machine made the USB noise.  
In frustration, I removed my 5 volt power supply, and put back all three isolation jumpers next to the USB card connector.  The Win8.1 machine then made both insert and remove noises one after another.  I pulled the 5 volt jumper and the ‘connected’ sound occurred once only.  The card was then powered by the computers USB outlet.  I was then able to flash Ver 1.301.   
I think its was a voltage problem.   The USB computer jack is not able to power the LaunchXL card and the 5v LEDs  but can power the card alone enough to flash 1.301 to memory.   My power supply measured 5.4 volts, which must have been too much.  I later used an adjustable power supply set voltage to exactly to 5v after I removed the jumpers and again the computer and CCS recognized the card.  
Also I suggest you use the USB cable supplied with the LaunchXL card since its nice and short and has a lower voltage drop.

I used CCS 9.1 build 10, (CCS 10 choked) ELS Ver 1.301 and have them on a USB drive since I reloaded them so often.  I checked all the connections with the drawings provided.  I tried reading the CCS  and documentation and watch the TI videos.  Not much help, especially the one in Chinese (actually all of it is a language I do not know) .  I am no embedded controller engineer or C programmer but I have seen a lot of useless messages kicked out by programs when the programming cannot figure out what failed.  There is no way CCS or XDS110 can measure operating voltage, so it obviously must be the debug that failed.  That was a fib in my experience.

 I have just added a BEI encoder and can display RPM 
[image]
when spun by a drill.

cad-quark, 2021-02-19:
And yes, the one power light is on with 5v from any source.  The important point is you do not need 5 volts to build, compile, link or flash the Ti Launchxl-F280049c, C2000, Piccolo, Icepick, XDS110, TMS320F280049c microcontroller.  3.3 volts will do just fine.  What you could use is fewer f——- names for it.  The 5 volts is only required to use the hardware of James’s brilliant design and careful programming.  Spend some time looking at Youtube Electronic Lead screw systems and it soon is obivious how amazing the Clough42 system is.  All that function from a single card and the work to do 20 videos.  
      Bravo JC.

andrea271068, 2021-02-19:
Thanks to everybody, today I was in the office, I will try Pouldo, thanks also to Bravo JC, I will read it, is complex and I try to configure-out , if I understood you say that 5v transformer is not stable enough for the pcb and can create conflict, today I bought Dc-Dc step out module,my is the cheep transformer that gave me also problem, I will connect now to the servo 36V  power supply and I will transfor to 5v true the DC-Dc step out module that stabilize also the power, let's try. Hope to solve, thanks guys please continue to support me 🙂

my 36V piwersupply

cad-quark, 2021-02-19:
You can eliminate all booster connections ( green module you built ) and run the LaunchXL with the USB cable,  only leave the 3.3 volt and ground jumpers.  Remove the 5v middle jumper.  Your computer and CCS should detect a USB device to  flash ELS 1.301 on to.  The Windows machines make the insert USB noise and add an XDS110 com port to the Device Manager.  I do not know what MAC does.  ( maybe you can help me load CCS to the MAC M1 AIR ).
  Don't forget to hit the resume arrow after the flash.  If that works, you have eliminated any possible problems with your soldering or crimping.  When you add your booster board and hookups to the Launch XL, on adding 5V power, the blue LED 3 should glow and the program is running.  I do not remember if this LED is active before the flash, but when the program is screwed up, and hangs at ‘message1’, it blinks erratically like a tiny blue screen of death.  I had to hit the white reset button on the LaunchXL.   What do you expect when the icon for “Build” is a hammer and flash is a bug.  We need a pickaxe for debug.  ⛏
On flash, LED2 will light up and blink
 
![A60DEAC3-A48F-48D4-9F07-6731A23707A6](https://user-images.githubusercontent.com/75825422/108546872-596b9c80-72af-11eb-88c2-77aa71b65bd7.
 Any one know CCS well enough to trace program execution or insert break points on the PC?

cad-quark, 2021-02-19:
[image]

cad-quark, 2021-02-19:
What I am saying is the 5V you supply must be close to 5 v that is all.  You do not need anything but the base LaunchXL card and a USB cable to program.  You can add the Clough42 parts later.   I just now tweaked the program and changed my encoder to 4000 pulses rather than 4096.  I used only the Usb cable and was powered by the computeer's cable only.  The Clough42 parts were removed.  This is how most users of controller cards program anyway.Doing so eliminates any bad construction or missed solder joints.  -----Original Message-----
From: andrea271068 
Sent: Feb 19, 2021 11:43 AM
To: clough42/electronic-leadscrew 
Cc: cad-quark , Comment 
Subject: Re: [clough42/electronic-leadscrew] Debug problem, what a sad day, I don't know waht to do.  (#132)

Thanks to everybody, today I was in the office, I will try Pouldo, thanks also to Bravo JC, I will read it, is complex and I try to configure-out , if I understood you say that 5v transformer is not stable enough for the pcb and can create conflict, today I bought Dc-Dc step out module,my is the cheep transformer that gave me also problem, I will connect now to the servo 36V  power supply and I will transfor to 5v true the DC-Dc step out module that stabilize also the power, let's try. Hope to solve, thanks guys please continue to support me 🙂

my 36V piwersupply

andrea271068, 2021-02-19:
Get it! You say debug only connection the TI via USB to the computer, because I don't need the rest. After that I can still plug in the Clught42 control board?

andrea271068, 2021-02-19:
Okay, I will try, I will debug, plug-in the TI directly via usb to the computer without PCB, I will debug and only after I will plug the Pcb board. 
Thanks, I will let you know 🙂

cad-quark, 2021-02-19:
Right.  You win a bucket of Iowa Snow.  Let us know if it works for you

cad-quark, 2021-02-19:
Oh yes, Bravo JC is just hooray for James Clough and his design.

andrea271068, 2021-02-20:
Good morning, I did it, I bluild up flashing the code true computer Usb whitout the  PCB, saying buildup completed and running. After that I disconnect usb and I remove the three jumpers, and I connect again the PCB, but notting happen, I don't know, see the photo. 
How I can delete now the flash in the memory? And try again to build up? When I try to do it, gave me some errors. [image]

andrea271068, 2021-02-20:
Hi, How I can erase previous flashed code on in the memory? i wan to debug again from 0

cad-quark, 2021-02-20:
Make your changes and hit debug.  I only had "els-f280049c" under the debug.  I changed steps and encoder resolution and flashed the card again.  It erases the previous flash.  Do not forget to hit resume.   After that, you might want to exit CCS or you will get an error when you pull the USB cable...  -----Original Message-----
From: andrea271068 
Sent: Feb 20, 2021 7:28 AM
To: clough42/electronic-leadscrew 
Cc: cad-quark , Comment 
Subject: Re: [clough42/electronic-leadscrew] Debug problem, what a sad day, I don't know waht to do.  (#132)

Hi, How I can erase previous flashed code on in the memory? i wan to debug again from 0

cad-quark, 2021-02-20:
Were you able to program the LaunchXL processor?

cad-quark, 2021-02-20:
Did you hit RESUME after the flash? if you remember James said it always halts at the first instruction when you flash debug or release.  Look for the Little green arrow.  

Did your MAC show the USB device?

cad-quark, 2021-02-20:
I just hooked up my new M1 Mac Air and I see the Mac does not make any sound when you connect the LaunchXL USB cable.  I did find it listed under SYSTEM INFORMATION It lists a USB 2 as XDS110.  It lights up LED 0, 1 and 3.  This new Mac has only a 3.1 round USB jack so I am using a USB adapter.  I do not have CCS9.1 installed on the Mac (can not figure out how to load it), since I used PCs to flash my LaunchXL.

andrea271068, 2021-02-21:
Hi, thanks, yesterday, I make huge disaster, I have many errors now, and now the blue led in the LaunchXL is off. I can't find the way to reset and cancel. I wan to reset the LounchXL, please you know how to do it? In one of your previous e-mail you told that you was force to reset many time for some reasons, and you told about a white button, ? I wan also delete the CCS and the software yo run the ELS, I wan to start again from 0. Please can you drive me on this? Many thanks

andrea271068, 2021-02-21:
I think that LounchXL is burned, when I connect directly to PC usb no light are one, and the computer say that he was force to disabilitate the USB because an accessory absorb to much energy. Also when I plug in the James PCB the LounchXL is death!! 
Now I purchase again a new LounchXL a new James PCB and a new Led&Key, I will remove everything from Pc and start everything again from 0. Please when the new goods will arrive, can somebody drive me true from step 1, now I have to wait few days

andrea271068, 2021-02-21:
I'm going crazy, now is running again, but still found 38 errors, I din't know what to do, completely lost now. How I can erase the project in CCS and start again? Please drive me true. [image][image]

cad-quark, 2021-02-21:
The white reset button is on the edge just up from the center jumpers.  If your main processor were a clock, it would be at 11:00.  Lay a rule along the top in your photo and its at 4cm.

Again, I am no programmer, but it looks like some punctuation has been removed from main.cpp.  Did you edit it?   If it were mine, I would re-unzip the ELS files.  I did so several times when I got the “target not found” or XDS110 failed messages before I found my 5v problem.

I am sorry I cannot give you better support.

cad-quark, 2021-02-21:
Its not void (void)
It should be void main (void)

cad-quark, 2021-02-21:
At the first red x

cad-quark, 2021-02-21:
Line 92.  Where did that “=“ come from?  I do not see it in my main.cpp

cad-quark, 2021-02-21:
For power, the USB from the Mac or a PC CANNOT power any Clough booster boards and the LED board must take over 100 mA.  In fact I removed the middle ( 5v ) isolation jumper when I loaded the program to the bare LaunchXL-F280049c board.  

Build ( if the Hammer symbol is dark) Debug ‘Release’ or els-f280049c,  whichever you have AND hit RESUME.  Disconnect USB AND the isolation jumpers for ground, and both voltages if present.  Then put on the Clough42 boards and plug in 5v power to the booster board.  The LED board should show Clough42, ELS 1.301, then 0 rpm, 005 feed.

If you know all your assembly is good, and your 5v supply is correct, then you can load the program like James does in part 10 video with the LED board so you know if its working.  

The last picture you sent shows the LaunchXL powered.  Is it now burnt out?

cad-quark, 2021-02-21:
As Export manager, you might know if its possible to send a computer board/development kit through the international mail.  The USA has rules about technological exchange in ITAR and EAR.  If we could figure that out, I could get another card from TI and flash it with your parameters and ship it to Gerbolina MN.

andrea271068, 2021-02-21:
Ciao, many thanks for your support. I never touch the program, not even for mistake, my problems yesterday come when I remove the middle jumper, I did when the power was on for all three jumper, after that many errors start to come out. For this reasons I would like to erase everything, even the CCS and the ELS 1.3..., everything!in this way if something is corrupted I will jump the problem.

andrea271068, 2021-02-21:
Regarding the LounchXL, I bought a new one from US, TI web site, should arrive within two weeks, if I will find same problems, sure man, Would be great if you can send to me a preinstalled board. i will pay for everything also for the bother that I gave to you, I'm force to find any solution to pass true this, I spend a little fortune for my hobby and I'can't stand with the machine not running with the ELS

andrea271068, 2021-02-21:
Today I did others staff, I was little I've had enough! I say to my self give little time to you on this issue, is drive me crazy, and this kind of panic is not good to see clear. Today I start to make a back plate for my lathe for a 4 jaws chuck and other little fixing on my mini mill machine. 
🙂

cad-quark, 2021-02-21:
if the jumper was removed under power, it might have introduced some errors, I do not really know.  I would reload everything.  I did on my PCs a number of times.  I have been working on this for weeks, and gave up several times.    We will get there, somehow. 
 Can you tell me how to load my new Mac with CCS and ELS?

cad-quark, 2021-02-21:
by the way, what in on your roof at the plant?  Solar panels?

cad-quark, 2021-02-21:
just for fun, what is your encoder and stepper resolution and your thread pitch?

andrea271068, 2021-02-21:
I found the white button, but how to do it? I push but notting happen. I open code composer studio, I open ELS project, and I push the white button in the TI board, but notting happen, may that is not the way to do it... 
many new thighs for me, I never deal with programs and good electronic level jobs. But I have good manually almost for everything and even if in my life I did something completely different, my school education was for mechanic engineering. 🙂

cad-quark, 2021-02-21:
When I had the LED2 blink I pushed the white button and the LED stopped it irregular flashing.  Other than that, I do not know.  If the LEDs still come on on the LaunchXL board then it still may be good.  Remember isolation jumpers on when using USB power from the PC/Mac but only if all the other boards and cables are disconnected.   If you use the 5V power plug, then the isolation jumpers must be pulled off the LaunchXL board.  That is the way James loaded the program so he could immediately see the LED board show this program running after he flashed and hit RESUME.

andrea271068, 2021-02-21:
On top my ruff? Solar panels. 
I just download the CCP from It for Mac, and for the ELS I sace in the mac book in a folder, just that! Was easy, may too easy 🙂

andrea271068, 2021-02-21:
I have a nema 23 close loop, for the encoder same a James, I have a metric lead-screw 2 mm pitch

andrea271068, 2021-02-21:
Manuale nema 23 close loop 
￼
HSS60驱动器-说明书.pdf

andrea271068, 2021-02-21:
[image]

andrea271068, 2021-02-21:
[image]

andrea271068, 2021-02-21:
[image]

andrea271068, 2021-02-21:
[image]

[112 more replies]