clough42/electronic-leadscrew #133: Who can help me with the configuration file  Hybrid stepper JMC 86J12156EC-1000-60-H; stepper driver JMC 3 HSS 2208H

Schaublin12, 2021-02-15:
Hi all,
I come from Germany and I`m excited about the ELS that I want to build one. I have ordered a hybrid stepper JMC 86J12156EC-1000-60-H 
https://www.jmc-motor.com/product/896.html
and a stepper diver JMC 3 HSS 2208H. 
https://www.jmc-motor.com/product/888.html
Do I have to define microstpes?
the stepper resolution of this motor is 200?! in the datasheet I can read:  
Step angle(deg):1.2
Position accuracy :1.8±0.09 

[image]
what do I have to define in Line 76,77,78,79,82 for my motor/driver configuration

TIA

Schaublin12, 2021-02-15:
okay direction and alaram pin is clear. 
In which case do I have to invert the signal of my step pin?
In which case do I have to invert the enable logic?

mac651955, 2021-02-18:
I would leave things as they are in the default config file until you get it all connected and working.  Steps and micro steps need to be adjusted based on your your results after you get it running.  First you should Flash your microprocessor with the control board to ensure that works.  Next hook up your encoder and ensure you get RPM displays.  Lastly hook up your servo and get it working.  Then once you have everything working you can the adjust the config file to work on your lathe.

mac651955, 2021-02-18:
Additionally that appears to be a 1000 step motor.  After you get the encoder working set your steps to 1000 and your micro steps to 3.  That should be a good starting point.  Depending on your gear ratio servo to lead screw you will probably have to adjust those micro steps.  It also appears that you don’t need a power supply as there is a 220 volt connection on that driver.

Schaublin12, 2021-02-18:
Okay thanks a lot. Now I have only modified the 
leadscrew pitch to 600
Microsteps 3
and resolution 1000
I I was afraid of damaging the stepperdriver if parameter 76, 78,79 is wrong
[image]

Can you explain me that fig.?
[image]

Now I have to wait for the parts.

mac651955, 2021-02-20:
The diagrams are for reference to another use I presume.   Notice PUL + and minus, Dir + and minus ,and ena + and minus.  Look at James wiring diagram and those wires will come from the TI board.    The TI board, microprocessor is the controller in our case.  Did you comment out the TPI line in the program.  It will not recognize both a imperial and metric lead screw.  By the way where are you in the process?
Step one-  build the control board
Step two- hook control board to microprocessor 
Step three- flash the microprocessor 
Step four-  wire in the encoder and test to ensure you get rpm display 
Step five -  wire in the servo driver and servo to test your revs to ensure your leadscrew will be turning the correct direction and correct speed.
Step six -  build electronics enclosure 
Step seven - install on lathe and enjoy.

Schaublin12, 2021-02-22:
Today I flashed the TI LaunchPad with Code Composer Studio 10.2.0 without any problems. I had to do this without the Interface PC Board mounted because it is not there yet. I´ll pull off all of these three jumpers now. 
When the package arrives I`ll make the next steps.
Thank you so far
Kind regards Christoph

Schaublin12, 2021-02-24:
Yesterday I got post from the USA. I soldered a little bit... I think everything works fine. The encoder and the stepperdriver comes from China. Because of chinese new year holiday a I have to wait for step four. 
 I'll keep you posted if anything changes
[image]
[image]

mac651955, 2021-02-24:
Looking good.  Remember those steps it will be easier to troubleshoot later.

Schaublin12, 2021-02-27:
Encoder is working, but I ordered one with only 360 p/r instand of 1024p/r. 
[image]

Schaublin12, 2021-03-04:
can I also use an incremantal encorder Heidenhein ROD 426.009 interface is TTL 
[image]

Schaublin12, 2021-03-05:
Chinese Bud Box is 1mm too long :-(
[image]
[image]

Schaublin12, 2021-03-21:
I cut the box into two pices. So I didn't had to deal with the draft angle. After a little bit of milling and glueing everything is fine. 
Stepper driver arrived yesterday after nearly a month. I wired everything together. It seems that works
[image]