clough42/electronic-leadscrew #134: Waiting to solve with ELS, I need a  momentary solution to power-feeding

andrea271068, 2021-02-23:
Hi, I have a question, if for this moment I wan no use the servo to run just the leadscrew for power feeding what I can do? I need just to buy a simple pulse generator with forward and reverse? Or I need to buy also other stuff, Arduino etc.  Can somebody list to me what I should buy for my servo? I wan a simple momentary solution, at least I can power feeding and use my lathe. I don't wan to disassemble the ELS set up and put back gears. 
I bought a nema 23 hybrid servo for the Clought42 automatic lead-screw project. 

IS NOT MY SURRENDER!!!

clough42, 2021-02-24:
There are lots of published arduino projects to speed control a stepper motor and there are also stepper speed control devices available on eBay.  I've never worked with any of them, but they might provide a solution for you.

closed 2021-02-24 by clough42

andrea271068, 2021-02-24:
But this is not what i'm looking for in truth I really wan to solve my ELS problems, dis you read about  my problem? I flesh the code LounchXl (I had sever problems but not look like ia working) but look like that  cannot communicate to display.
I bought again a new PCB and led&key to check again with a new pair. 
Can you help me to solve? I'm little lost, there are one gentle persons that is trying hard ti help me there, without him I' lost in this issue. Please can you check to my story?