clough42/electronic-leadscrew #136: Bedug problem issue solved? Or communication problem within LunchXl, Pcb to Led&key?

andrea271068, 2021-02-24:
Please, 

1) How I can make sure that the Luachxl have debug problem solved? 
In case yes 

2) How I can solve the communication problem within LounchXl, Pcb and Led&key?
 
[image] 

[image]

clough42, 2021-02-24:
It looks from your screenshots like you got it programmed and running.  If you have the "Release" configuration selected, and you saw the dialog saying it was erasing the flash, it should be permanent.  (until you flash it again)

Take a look at the troubleshooting section of the wiki.  In particular, you should probably start by checking continuity of all of the connections between the boards as described there.  The most common cause of a blank display is a bad solder joint.

mac651955, 2021-02-25:
It also appears in the photo that you have not disconnected the power tha separates the debug section from the microprocessor.  Might just be the way I am looking at it.  If not that needs to be done before something bad happens.

andrea271068, 2021-02-25:
Dear James, 
many thanks, may we found a problem in the LED&KEY board,  check, we didn't found any signal within that references point. 
Now I will make some bridges 
And I will see. Meanwhile I order a new led&key board, but will take several days to arrive. 
I can not wait to run your Fantastic ELS!!! I so exited about this project, will bring my machine at different level!!! 
Thanks to give this project to as as an open source and specially thank to drive us step by step in this venture true your really amazing youtube video!! I will let you know soon if that led&key was my problem. Thanks to support me, to share with us your ideas 💡 

[image](https://user-images.githubusercontent.com/75749307/109129648-29117b80-7751-11eb-898a-6acfc4d32a95.jpeg)

andrea271068, 2021-02-25:
WOW!!!!! We get it!!! The problem was un the Led&Key board!!!!! 

Now I have to place in the enclosure and make a run test in the machine!!! I'm soooooo happy 

In place there the photo of what I did for everybody   😅😅😅 thanks to everybody!!!   

The code was flashed, but before to say done let's wait the installation in the machine enclosure, never know... 

[image]

[image]

andrea271068, 2021-02-25:
It's running but is not stable, I have interferences may I have some noise, or some not stable connections and with the vibration move, I will check everything, but I'm very happy now!!! 

Here you will find the video 
[image]. 

[image]

[image]

James Thanks again for your fantastic project!!!! 🙂

steelbarz, 2021-03-10:
Hi James and everyone.  1st I would like to say thank you to James and his amazing project.  I found it on youtube just a few weeks ago and all the parts showed up today.  I did get the wrong rotary encoder did not order the right # PR encoder.  Mine came as a 50.  not even good enough for the lower limit in the software James wrote of 100.

And that's where I came into the same problem as the OP.  My daughter was over and we set the darn thing up on the coffee table and watch James set his up.  Imagine that 2 woodworkers upgrading the lathe I use as a mill...lol  Putting on this electric lead screw since the last part, the encoder, showed up.  The sanity check said I could only go as low as 100.  So I had to adjust some settings and I wanted a way to verify the update.  so I am sorry James my coding skills are nowhere near good as yours I can only edit in C++, I'm a wiz with visual basic.  

In the user interface config file I changed the Clough42 logo at start-up to a time stamp.  I hope I didn't violate any copyright, I thought this was the purpose of open source.  To help others with similar problems.

Perhaps James or some other wiz, he's busy...lol, could; have added to the flash some kid of 3rd startup screen that includes a rolling number or something each time the LaunchXL is updated during such a setup scenario while debugging.  I know in visual basic this would be easy to do...lol

steelbarz, 2021-03-10:
Oh and in case you do read this and plan on making more of your boards and have the time, I would have paid more money if you had of included the 5pin headers (wire lead ends??) or screw terminals for those pins.  I try had to make things look neat, so now I am waiting another month to get those little connectors I should have bought months ago when setting up my CNC..lol

Thanks again for all the amazing work.

steelbarz, 2021-03-10:
The guy with all the pictures, the reason you have noise, your shielded cables are useless unless you have a metal box in a noisy environment like a shop.  Switch that fancy white plastic box for a proper one and stay on topic will yah...lol

andrea271068, 2021-03-10:
Hi, thanks, the problem is that in EU we can't find and we can't use for electronics metal enclosure box, so taking in consideration that, for your experience would be better if I will remove all my shield cable  with standard one? 
Many thanks

steelbarz, 2021-03-10:
> Hi, thanks, the problem is that in EU we can't find and we can't use for electronics metal enclosure box, so taking in consideration that, for your experience would be better if I will remove all my shield cable with standard one?
> Many thanks

May I suggest do what James and many others do and suggest, put the "brains of the system" - low voltage stuff in the metal box using shielded cables connected properly  (if you find a small enough one it can even go in the plastic box).  And the high voltage - power supply,  in the white plastic box.  Shielded cables to connect everything together with the appropriate considerations for electrical ground loops etc.

andrea271068, 2021-03-10:
Thanks I will try to split the enclosure, and may I can find a small metal box for low voltage. For low voltage is required ti the fan? 
Thanks

andrea271068, 2021-03-10:
I din't have loops, but were you suggest to place, fasten the ground? Each shield drain separated to the machine work bench stand?

cad-quark, 2021-03-11:
You have the launchXL board on top of the stepper driver and switching power supply.  Those are very noisy neighbors.  I would move the launchXL board away from them

andrea271068, 2021-03-11:
Thanks, I will try move the LounchXl from the rest. But you know? I have little noise at the beginning just when I power on and turn the spindle, after few second is over the blinking,  Appear durning the spinning process when I accelerate fast on the spindle control knob.

cad-quark, 2021-03-11:
My little display flickers sometimes but I do not have any shielding and everything in the ELS system is mounted on a plywood base.  I will bend up some sheet metal covers.

The toughest thing to shield in an aircraft is Audio.  Audio picks up hum, crosstalk and switch noise.  What we did was route it away from Ethernet and heavy current users and only ground the shield at one end.  If a shield wire carries a current from a from one ground point to another, it is no longer a shield.  There is plenty of audio in an aircraft, Comm radio, Nav signals, ILS and TCAS warnings and all the autopilot and emergency voice overs.

andrea271068, 2021-03-11:
Very interesting and fashionable. 
Regarding my stuff,my shield are not connected to make the loop, only one side, now I will try to to cover with metal. 

My little display flickers sometimes, mainly at the start, after a little, fraction of second stabilize it self. It's strange. 

I can run even in that way, it is not a problem, but I will like to find the reasons and solve.

andrea271068, 2021-03-11:
You solve with your lift machine?

cad-quark, 2021-03-11:
No, I did not hear from Gianluca, nor did I expect to.  The company would much prefer I deal only with their dealer in Montreal.  There are a number of other issues in addition to Covid19 risks.

dsgroninger, 2021-03-11:
Just a suggestion, from looking at your pictures above.  The long, unshielded ribbon cable from the Launch to the keypad makes for a nice antenna to pick up all kinds of radiated crud from the stepper driver.  If your overall setup allows for that to be shorter, made from shielded cable properly grounded to chassis ground at one end only, or at least twisted to avoid a nice parallel long path, your should see an improvement.

The 5 volt supply to the Launch is also long with that adapter and its position in your box, and the 5 volt cable is strung out full length and goes right over the 48 volt switcher and right by the stepper controller.  I'd try to shorten that up and get it away from the motor circuitry if I could.

If you see flickering while spinning up the lathe, if your lathe has a variable frequency drive, or a DC motor driver, the heaviest currents to the main motor will happen when accelerating the motor from a stop.  If there's no cutting load on the spindle, the current draw should drop once the motor is at speed and the spindle is spinning freely.  Radiated and conducted noise from the main motor, wiring and the speed control will be worst when the current is at its peak during acceleration or a heavy cutting load.

My lathe and its factory DRO came with all kinds of warnings about keeping the DRO sensor wiring well clear of the motor and speed controller, giving it a separate power supply, being careful to use a star pattern grounding scheme to avoid loops, etc.  

In my setup, I used a larger metal box, and laid out the parts on an aluminum mounting plate that drops into the box on standoffs.  The plate and box are at chassis ground with the power inlet module, safety ground to mains.  The 48 volt supply frame grounds through the plate, and the stepper driver housing likewise grounds to the plate.  The stepper driver and it's supply are towards the bottom of the box, similar to yours, while the power is managed on a DIN rail terminal block as James suggested in his video.  Instead of the power adapter for 5 volts, I found a neat DIN rail 5 volt supply on Amazon and use that, keeping all my wiring as short as possible.  The Launch is located all the way at the top of my box, as far form all the noisy stuff as I could put it.

I have the box standing vertically behind the splash shield of the lathe, with a couple of inches extending above the top of the shield.  I mounted the display in the C124 box as James designed for, and mounted that to the face of the main electrical box facing the front of the lathe.  That meant that I only needed about ten inches of cable from the Launch to the display board.  So far, no flickering or noise issues with my setup.  

I'm still working on the mechanical assembly on my PM-1030 lathe, waiting for the right belts and making a motor mount.  The encoder setup I took from a posting on ThingiVerse from a guy named Nick Stahl (https://www.thingiverse.com/thing:4732492), and that seems to be working well.  Will post some pictures when all done and proven out.

andrea271068, 2021-03-13:
Many thanks, I will try to separate the enclosure and shorter the Led&key cable at the moment it's around  2,5 feet. 
Anyway I still wait the led&key new board, dis you see what happen to my, was defective and to make run I was force to make a huge rewiring job. 
Thanks for your nice and details explanation.

cad-quark, 2021-03-13:
I bought all the boards again and can ship them to you if you think it will be faster & send me your boards when they show up.
I finally got CCS loaded on the Mac M1 laptop.  Many people complain that the Apple M1 processor version of Big Sur quarantines apps and I could not open or "xattr" CCS91 to install.  CCS version 10 did load.  I just hope it will flash Version 1.3 or 1.4 Clough42 ELS.  I may just use it on the #2 Launchxl board.  Right now my stepper is running about 1/2 too slow.  10TPI gives me 20 threads per inch.  
Its still amazing: I am very happy with James's design.

andrea271068, 2021-03-13:
Thanks, I should receive the led&key this week. To configuration H i configure 1 because I have 1:3 to the lead-screw, but I think I have to i vert, I should put 3, now it's very slow, I like in the feed, but I din't know if can be yhe right set-up for threading, today I will try to make for the first time an M8, let's see if it is well configure or if I have to switch placing 3 in configuration H. I set at 1 because James set at 1 in his Metric lead-screw youtube video. And I follow him. But at 100 it's very slow and I don't know if it's correct or not, today I will try make my first screw, let's see... 🙂

cad-quark, 2021-03-13:
Ok I will be working on the Easylift today, but let us know how your threading works.  Go slow: it gets exciting at any speed.

andrea271068, 2021-03-13:
No, doesn't make any threads, is like feeding. I try to make M8-1,25 but doesn't make, eat very fine. Now I will try to set configuration H at 3:1 and not 1. For the occasion I will also download the new version.  You know, I'm little scared, hope to not face again previous debug problem, put I have to flash again, is a nightmare the scare to face again old problems, not reason to  happen. But scare is scare.

Scarica Outlook per Android<https://aka.ms/AAb9ysg>

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

COMPOSAD s.r.l.

Sede Legale: Viale Lombardia, 29 - 46109 Viadana (MN) - Italy

ph: +39 0375 783611 | fax: +39 0375 7836695

Sede Operativa: Via Lombardi 1 - Z.I. Gerbolina - 46019 Viadana (MN) - Italy

C.F - P.IVA/VAT code e R.I (IT)02317920201

www.composad.com<whttp://www.composad.com>

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

Il presente messaggio di posta elettronica e i relativi allegati sono strettamente confidenziali, soggetti alle leggi vigenti, pertanto se ricevi questo messaggio di posta elettronica per errore, per favore contattaci: e' assolutamente vietato usare, copiare, o divulgare il messaggio e ogni informazione in esso contenuta, tranne che al destinatario originario. Le comunicazioni Internet non sono sicure e non ci riteniamo responsabili per eventuali abusi di terze parti, per alterazioni nel corso della trasmissione, né per danneggiamenti causati da virus o altre anomalie a noi non imputabili.
----
This message and its attachments are addressed solely to the persons mentioned above and may contain confidential information. If you have received this message by mistake, be informed that any use/copying/forwarding of any part of its content is completely prohibited. Please return it immediately to the sender and delete the message. Internet communication is not secure and therefore we are not responsible for any misuse of it by third parties; Alterations during the transmission or damages caused by viruses or other anomalies are not attributable to us"

'

andrea271068, 2021-03-13:
Hi I have may a set-up co figuration H problem, I can't thread, and the leadscrew is very slow. Is like feed. 
My connection within spindle and encoder is 1:1. 
I set configuration H at: 
Stepper microstep: 1
Stepper resolution: 1000

May I have to set a different stepper resolution, below you will find my driver, look like that my driver default is 400, this means that I have to put in configuration H: stepper resolutions 400? 

Or any others set as for the driver grid setting?
Please check my driver photo with switch setting 

[image]

Check also my set up manual here below you will find the link: 

https://doc-04-9k-docs.googleusercontent.com/docs/securesc/21cocqnl5ekk0bo0mf9j4aoasiphhqm2/fra42nprkikdsdfls14bdq4v7ur77be5/1615662525000/02915991246204425624/11143975116392701136Z/134jbl9eqBQBOkDODQ6hUHt9WyAIOMW46?e=download&nonce=8ofg4u88ff4ks&user=11143975116392701136Z&hash=m0amtlafaam745fd34qu6isjjmdr31iu

andrea271068, 2021-03-13:
Just now I set my driver switch at 1000, but it's the same, no metric, no TPI! 🤷‍♂️😮

andrea271068, 2021-03-14:
When I turn the spindle 1 turn, the lead screw just move a very very little fraction, don't spin 1:1 with the spindle. I have 2mm lead-screw, I set Configuration H at 200=2 mm. I'm going crazy, I can't configure-out why is appending. Please some body can help me,know why and how to solve, an idea that will bring me to the right road. 
  [image]

[image]

mac651955, 2021-03-14:
Start increasing your micro steps several at a time until you get 2 rev of the lead screw with one revival the spindle.

mac651955, 2021-03-14:
Post a picture of your dip switch settings on your stepper driver also.

andrea271068, 2021-03-14:
Many thanks, now I set the driver deep switch like this at 1000. 
If I understand well you are say to increase the microstep in ci figuration H, at the moment I'm at 1, but I try also 3, but notting change may the change is so little impercettibile. may I should try to out at 30?, or I din't understand? 

     [image]

mac651955, 2021-03-14:
Try micro steps at 12 and see what you get.  Just turn the spindle by hand and mark your stepper pulley.  Watch the the pulley and see how many revs you get with one rev of the spindle with your els set to thread at a pitch at 2 mm, if that is the pitch on your lead screw.

mac651955, 2021-03-14:
After more looking you need to reset those dip switch settings back to 400.  Those are micro step settings not step settings.  With them set to 1000 with a motor with 200 steps (assumption that your motor is 200 steps),  it will take 200,000 pulses to turn one rev.  Post a picture or a link of the motor and motor driver that would help.  With 200 steps per rev with microsteps set at 400 that is still 80,000 pulses.  Seems awful high.  Need to determine how many steps are required for that particular motor.  Then set your steps to that number then you can dial in your micro steps.

andrea271068, 2021-03-14:
Many, many thanks for your kind help. 
Okay, I will bring back at 400, and I will try at 12. 

This is my motor: 
[image]

[image]

[image]

[image]

This is the manual 

https://doc-04-9k-docs.googleusercontent.com/docs/securesc/21cocqnl5ekk0bo0mf9j4aoasiphhqm2/fra42nprkikdsdfls14bdq4v7ur77be5/1615662525000/02915991246204425624/11143975116392701136Z/134jbl9eqBQBOkDODQ6hUHt9WyAIOMW46?e=download&nonce=8ofg4u88ff4ks&user=11143975116392701136Z&hash=m0amtlafaam745fd34qu6isjjmdr31iu

mac651955, 2021-03-14:
Try setting your config.h file with 1000 steps with 40 micro steps and check your revs.

andrea271068, 2021-03-14:
Means instead 1 I should try to put 1000 microstep. 
And instead of 1000 replace with 40 in configuration H. 

Define stepper_microsteps 1000
Define Stepper_resolution 40

It's correct, I well interprete?

andrea271068, 2021-03-14:
Or it is: 

Define stepper_microsteps 40
Define Stepper_resolution 1000
?

JonWoellhaf, 2021-03-14:
Andrea,

I hope this will help:

Everywhere in the code, James uses the product of STEPPER_MICROSTEPS and STEPPER_RESOLUTION.

So it would make no difference which one you change, as long as the result is 40,000. 

However, James has restricted the allowable values of both, so STEPPER_MICROSTEPS must be 1 to 256 and STEPPER_RESOLUTION must be 1 to 2000.

So you only have the option of 

    Define stepper_microsteps 40
    Define Stepper_resolution 1000

Jan (Yan)
From: andrea271068 
Sent: March 14, 2021 17:22
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Communication problem within LunchXl, Pcb to Led&key? (#136)

Or it is:

Define stepper_microsteps 40
Define Stepper_resolution 1000
?

andrea271068, 2021-03-15:
Thanks, At 40 still is not enough. 
After several set, I end-up with almost the maximum possible 
Microsteps 250 and resolution 1000, in this set my spindle need to make 7,5 turn to turn the lead-screw 1. 
Is 7,5:1. 
I reach the top of the configuration, and now? Is better but still is far to run in order to make is work.

andrea271068, 2021-03-15:
Many thank, now I set 40 and 1000, I set thread in the display mm 2, with this set I need to turn 3 time the spindle to make turn 1 time the lead screw. So I set at 120 microsteps and 1000 resolution! In this way 1 spindle turn will make more the leadscrew 1 time, but the problem is that when I stop to run the spindle the servo continue to run for a very very long time, may for other 3 minutes!! Now I reach to run 1:1 but the timing is not correct, the servo continue to turn the lead-screw, practically run stop 🛑 timing is not synchronized! And now?

cad-quark, 2021-03-15:
Sorry, Andrea, but I am confused.  Let me try to get back to basics.
1. What is your lathe encoder output?  That number, times 4, should be in Configuration H.
The Rattm encoder is 1000 but that number has nothing to do with the ELS its only used for feedback to the Rattm hybrid stepper system.
The Rattm “default” has NOTHING to do with ELS.  Its just how it is set when they send it to you.  
Its up to you to pick your own steps that work for your application.
2. Pick any number of steps.   For instance 1000 and set switches 3,4,5 on and 6 off.
3. If you gear down or change ratio of stepper motor to lead screw, use Micro steps to correct that. 
I believe you said you had 3 to 1.  Three turns of the Rattm stepper motor for one of the lead screw.
4. Now use “micro steps” to do this reduction.  Microsteps = 3.
5. NOW before you do anything, wait a while for all the other folks on Git-hub to comment on this approach.   I could be wrong.

cad-quark, 2021-03-15:
Setting 1000 steps and 40 micro steps gives you 40,000 steps per revolution of the lead screw.  You could walk halfway to Milan with that many steps.

andrea271068, 2021-03-15:
Hi thanks god you answer me, is all day that I'm there is drive ne crazy!! Should work but I don't know why is not turn in the correct ratio. 
If I set configuration H 40, and 1000 with default switch, the lead-screw pulley move just 1/4, should not, I know. Check the photo below.

I will answer to your point

- 1) my encoder is 1024 P/R and is set in configuration H as for Janes setup: 4086

-2) I did, is the setup use for the below photos.

-3) as you can see from the photo I drive 1/3 

-4) Now use “micro steps” to do this reduction. Microsteps = 3. =? WHAT YOU MEAN, I place 40, or maybe I don't understand this point? Were I should put the 3? 

-5) I hope that some also others will come to help me, I try hundreds time. Best configuration was:
Driver at 3200 at 24, in this configuration up to 250 Rpm work if I spin faster will loose the step, and wen I stop the spindle the lead-screw continue to run. 

Configuration below 3200 will lost the steps even at 100 rpm. 

Higher then 3200, work good 12800 at 12, but at 300 start to lost the steps. 
Anyway I try all, and is very unstable, I don't know what to do, I try all the set-up from driver to configuration H 😓. 

[image]

[image]

JonWoellhaf, 2021-03-15:
Andrea,

Assume all the hardware and software is working perfectly.

I will use the term pulses per revolution.

If the driver is set to move the motor one revolution after it has received 1600 pulses, then the ELS software must be set to generate 1600 pulses per revolution.

The 1600 pulses can come from any (allowed) product of steps and microsteps. For example, 200 steps times 8 microsteps. Or 800 steps times 2 microsteps.

The key issue is that the number of pulses generated by the ELS must match the number of steps expected by the servo driver.

Jan

From: andrea271068 
Sent: March 15, 2021 11:41
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Communication problem within LunchXl, Pcb to Led&key? (#136)

Hi thanks god you answer me, is all day that I'm there is drive ne crazy!! Should work but I don't know why is not turn in the correct ratio.
If I set configuration H 40, and 1000 with default switch, the lead-screw pulley move just 1/4, should not, I know. Check the photo below.

I will answer to your point

  a.. 
    1.. my encoder is 1024 P/R and is set in configuration H as for Janes setup: 4086 
-2) I did, is the setup use for the below photos.

-3) as you can see from the photo I drive 1/3

-4) Now use “micro steps” to do this reduction. Microsteps = 3. =? WHAT YOU MEAN, I place 40, or maybe I don't understand this point? Were I should put the 3?

-5) I hope that some also others will come to help me, I try hundreds time. Best configuration was:
Driver at 3200 at 24, in this configuration up to 250 Rpm work if I spin faster will loose the step, and wen I stop the spindle the lead-screw continue to run.

Configuration below 3200 will lost the steps even at 100 rpm.

Higher then 3200, work good 12800 at 12, but at 300 start to lost the steps.
Anyway I try all, and is very unstable, I don't know what to do, I try all the set-up from driver to configuration H 😓.

cad-quark, 2021-03-15:
Encoder steps 1024 x4 is 4096, not 4086  ok maybe a mis-type.

Now Configuration H, line 67,  # define STEPPER_MICROSTEPS 3  //not 40

SEE ELS  video PART 10 at 27:59 or read the note in configuration H above the MICROSTEPS setting,  line 60 or so.  this message is in version 1.301 and later, but not his first release.  

And I am wrong: 40,000 steps will put you in Rome.

cad-quark, 2021-03-15:
Clough42:
// NOTE: If you are using a servo drive with something other than a 1:1 drive
// ratio, you can use the STEPPER_MICROSTEPS to compensate.  For example, if you
// have a servo configured for 1000 steps/revolution and it drives the leadscrew
// through a 3:1 reduction, you can set STEPPER_RESOLUTION to 1000 and
// STEPPER_MICROSTEPS to 3.

cad-quark, 2021-03-15:
[image]
You may want to buy a  cheap digital tachometer ($20 US) to check your encoder setting.  

Once that matches it is a matter of steps per revolution of the lead screw.  Motor steps times reduction.  In your case  stepper steps times 3.

andrea271068, 2021-03-15:
> Clough42:
>
> // NOTE: If you are using a servo drive with something other than a 1:1 drive
>
> // ratio, you can use the STEPPER_MICROSTEPS to compensate.  For example, if you
>
> // have a servo configured [...]

I did in this way, hundreds tine, the driver switch is set at 1000, configuration H at 3 and 1000, but like this one tuen of the spindle is equal to a micro-metric movement of the leads-crew pulley :(

andrea271068, 2021-03-15:
[image]

You can't even recognize by eye the pulley movement.

cad-quark, 2021-03-15:
Oh sorry.   I assume your Clough42 RPM readout and your machine RPM match.  That would indicate that the encoder pulse count is correct at the Launchxl board.  Also that your stepper controller switch settings are at 3,4,5 ON, 6 And all other OFF for the 3,1000 settings.   I assume you press THD and the lower threads per inch to try it.   If so, I am out of ideas.  
My lead screw is 4mm per turn so I use  #define Leadscrew HMM 400 and  comment out  \\ Leadscrew TPI.  Running the lathe at 8 TPI it is a race to keep up with the carriage.   

You have the second shipment of Clough42 boards, and I received a second LaunchXL board.  Is this the time to replace all the electronics?
//Chuck

andrea271068, 2021-03-15:
Trust ne it's drive me crazy!! I run out if idea too, but I can't surrender. I follow all the instruction, hundreds time, to my eyeI can't see any error! I don't know why it's not turning at the right ratio! At 3-1000 in configuration H the lead screw move just a minimal fraction! I found play with the diver switch other possibilities, I found several time 1:1, but all the time with that setting when I switch off the spindle the lead-screw continue to spin, some time even for many minutes! But when this also the servo sound is not good, you can ear that he make thousand step to make a revolution. I don't know really. 
But why make this to me? This is drive me crazy!! I can't recognize why or were is the problem!! Hope that some body can give me some good idea to solve or to understand.

andrea271068, 2021-03-15:
The launchxl arrive, also the pcb, but is missing the most important in truth to solve previous quality problem the Led&key! But I think even if I will replace all the elements, I will face some problem at 3- 1000 the lead screw not turno 1:1 with the spindle!

andrea271068, 2021-03-15:
Lost! I hope that James can intervene to help me. Really I follow everything he say, and also my set-up, there is something in the set-up or is calculation problem to be solve??? 
This is the dilemma!

dsgroninger, 2021-03-15:
I would take a step back and look at the basics of the setup.  As cad-quark points out, if the RPM on the Clough42 panel matches the RPM on your lathe spindle, then the encoder for the spindle is set up correctly.

Now, your stepper motor and its controller together need some number of pulses to get one turn of the motor shaft.  Typical motors will take steps of 1.2 or 1.8 degrees per step.  There will be a way to set the motor controller to take a certain total number of steps per revolution of the motor.  This might be set with DIP switches, or you may need a programming cable to talk to the controller.  My particular setup has no switches and must be programmed with a serial cable and PC software.  I got my serial cable from the same place I bought the motor and controller kit, Automation Technologies, http://www.automationtechnologiesinc.com/  .  I downloaded the PC program from their web site if I remember correctly.

Anyway, there is some method to set your motor controller for some number of steps per turn, and 1000 seems to be a good number if you can get that set.  James's math in the software handles that well.  From all your notes, it sounds like this number is much higher than 1000 for your actual motor/controller combination.  If that's true, either the math won't work out well (doubtful using floating point), or the frequency limit of about 100 kHz will be reached between the Launch board and the motor controller.  This is what causes the overrun behavior that should be fixed in James's release 1.4.

Now, once you get the motor to take 1000 steps per turn, think about the gear ratio between your motor and the lead screw on the lathe.  If you have a 1:3 ratio, perhaps a 24 tooth gear on the motor and a 72 tooth on the lead screw, then set the micro steps in configuration.h to 3.

That just leaves the actual pitch of your lead screw to set in configuration.h and things should work.

I was able to set my motor controller to 1000 steps per turn of the motor, and my lead screw is 8 turns per inch.  I have the 1:3 pulley ratio (24:72 teeth).  Here's my configuration.h setup (no other changes to James's basic file):

// For Imperial leadscrews: pitch in Threads Per Inch (TPI)
#define LEADSCREW_TPI 8

// For metric leadscrews: pitch in hundredths of a millimeter (HMM)
// Example: 200hmm = 2mm
//#define LEADSCREW_HMM 200

//================================================================================
//                                STEPPER/SERVO
//
// Define the number of steps and microsteps for your stepper motor, the pin
// polarity of the driver, and whether to use additional features, like servo
// alarm feedback.
//
// NOTE: If you are using a servo drive with something other than a 1:1 drive
// ratio, you can use the STEPPER_MICROSTEPS to compensate.  For example, if you
// have a servo configured for 1000 steps/revolution and it drives the leadscrew
// through a 3:1 reduction, you can set STEPPER_RESOLUTION to 1000 and
// STEPPER_MICROSTEPS to 3.
//================================================================================

// Steps and microsteps
#define STEPPER_MICROSTEPS 3
#define STEPPER_RESOLUTION 1000

Now, with that setup, my tachometer on the lathe matches the tach readout on the Clough42 panel.  I can set the threads per inch to 8, which matches my lead screw, and by turning the lathe spindle one turn by hand, I can see the lead screw make exactly one turn to match.

If your tachometer matches up for spindle speed, I really think that your problem is in the motor controller setup.  You need to get that to some reasonable number of pulses per turn of the motor, then things should work out.  It doesn't need to be 1000, but somewhere in that general ballpark is where you want to be for all the rest of the system to work well with it.

andrea271068, 2021-03-15:
Many thanks, yes the spindle run at same speed of the encoder, both tachometer run at almost same speed, some little difference up and down during spinning. 
My servo nema24 close loop step angle is 1,8. It Is 200 khz.  32 bit DSP. My lead-screw is 2 mm, and I set configuration H at 200. For the rest I'm lost. I try all possible configuration but no-one look to work.

dsgroninger, 2021-03-15:
Going up several steps in this thread, and looking at the DIP switch chart for your motor controller, it looks to me like the switch settings should be SW3, SW4 and SW5 ON, with SW6 OFF to get 1000 pulses per turn.  If all four are off, that looks like 40,000 pulses per turn, and that would explain a lot of the problems that you are having.  With it set that high, you would only be able to get to 2.5 turns per second of the motor before you'd hit the 100 kHz limit on the signal (40k x 2.5 = 100,000 pulses per second).

Sorry, I looked back through the thread here and didn't see a picture or where you stated your actual switch settings on the motor controller.

andrea271068, 2021-03-15:
[image]

andrea271068, 2021-03-15:
Wow!! I just check the photo zoom, I see now the  arrow, down for On!!! But the stepper arrive with all the switch in the right position, I assume that was on, the arrow is so small that you can't see clearly by eye, now I will go immediately to reverse all switching and try!!! CHINA!!!!

dsgroninger, 2021-03-15:
Ahh hah!  OK, it looks like you have the switches set exactly the opposite of what I suggested.  Note the lettering on the switch in your picture, there's a tiny little arrow pointing to the left, so ON is with the switch closest to the heat sink side of the controller.  In the pictures, you have SW1 through SW5 off, and only SW6 on.  If I read the chart right, that would be 51,200 pulses per turn.

JonWoellhaf, 2021-03-15:
I concur with dsgroninger. Good advice, I think.

andrea271068, 2021-03-15:
Yes, let me see, I'm in front the machine now

[25 more replies]