clough42/electronic-leadscrew #137: New firmware update not showing new number

xfuzz, 2021-03-01:
I just updated my TI to the new firmware version 1.4. The flash completes and when I reboot the TI the control panel still comes up as CLOUGH42  1.3.   I don,t get the new version 1.4 showing. Sodid it update or not?

clough42, 2021-03-01:
Make sure you have "Release" selected (under the hammer icon).  The "Debug"
profile loads only into RAM, so it is lost on reset or power cycle.

James

mac651955, 2021-03-01:
It did not update.  Don’t forget the last step is resume.  I flashed first time without resume and it didn’t work. Rewatched the video and last step with resume worked fine.

xfuzz, 2021-03-01:
I do in fact have it set to release.. in CC in the left pane I have listed
electronic leadscrew 1.40 , then ,els-f280049c [active release} then the
remaining of the files. under the hammerI have 1 debug 2 release with
release highlighted and checked.
I click the hammer and it complies in the lower pane many files scroll by
and the last line says finished. i cycle power and it still says 1.3.

xfuzz, 2021-03-01:
Oh one more thing, I have never had a red light icon when ever I do the
flashing. no resume at all.

clough42, 2021-03-01:
Clicking the hammer just compiles the code.  You have to click the Debug
(bug icon) to actually program the device.  It will program, and then run
to the start of main() where it will stop and wait for you to press
resume.  At that point, it will be programmed.

James

xfuzz, 2021-03-01:
Not working, I downloaded the 1.4 file from get hub, I  load file into CC,
in left pane it shows els-electronic lead screw 1.400 then under that in a
sub directory it says same thing, Then under that it showes in highlight
els-f280049c {active release} i then click hammer and it compiles I then
click the bug icon, it starts and i get several pop up windows( erasing
then programing) now I get red icon, I click it red icon , finishes, with a
lot of code in mid window ending at line 177, below that in left small
window i get "C28xx_CPU1: GEL Output: ... DCSM Initialization Done" ...the
lower right window has optimizing advice. i power  cycle and hane no change
still 1.3 uggggg.

clough42, 2021-03-01:
Let's do a crazy sanity check.  Take a look at UserInterface.cpp around
line 31.  You should see this:

const MESSAGE STARTUP_MESSAGE_2 =
{
  .message = { LETTER_E, LETTER_L, LETTER_S, DASH,* ONE | POINT, FOUR |
POINT, ZERO, ZERO* },
  .displayTime = UI_REFRESH_RATE_HZ * 1.5
};

Do you see this, or do you see 1.3.01?

James

xfuzz, 2021-03-01:
Got it. Here is what I did. uninstalled cc, deleted all files relating to
1.3 and 1.4 firmware. reinstalled CC, re downloaded 1.4, did the open from
file system as I had before and whamo, success.

xfuzz, 2021-03-01:
I apologize for my poor manors. Thank you for helping me out, I truly
appreciate it. I know Iam a problem child I hope to grow up someday.

clough42, 2021-03-02:
No worries.  This stuff is complex.

closed 2021-03-02 by xfuzz