clough42/electronic-leadscrew #138: Question re Code Composer Studio

Geoff-McCombe, 2021-03-04:
hi team,

most parts are now here so trying to get CCS working. 

The main page for CCS sent me down a version 10.2.2.00009 path.

This all seems slightly different to shown in James' original video.

Should I continue with 10.2 or try to source version 9 somewhere?

Thanks

Geoff

Zymurgy42, 2021-03-04:
Code Composer Studio Downloads (ti.com)
<https://software-dl.ti.com/ccs/esd/documents/ccs_downloads.html>

Scroll down and version 9 is there. As to your question, it may be the
easiest first step.

Geoff-McCombe, 2021-03-04:
Thanks for the quick response and the link!!!

Wilco

Geoff

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2021-03-04 by Geoff-McCombe