clough42/electronic-leadscrew #139: Not precise leadscrew

poulbo, 2021-03-06:
Not precise leadscrew

Am I the only one who has discovered that 1: 1 position A
on Weiss-WM280V-F, (Warco-280VF, PM1127VF-F), is not exactly 1: 1,
when making very long threads they do not become precise. You can
correct on leadscrew parameter in Configuration.h.
I have a Weiss 280VF with 3mm lead screw instead of 300 I have written 293
and it should be 292.5, but then you have to specify 1/10 of a mm = 3000.
Is there a better way to put a correction factor into the program.
I use  ELS ver. 1.4.0

     #define LEADSCREW_HMM 293 // (300) White VM280
     #define STEPPER_MICROSTEPS 3
     #define STEPPER_RESOLUTION 1000
     #define ENCODER_RESOLUTION 4096
     #define USE_FLOATING_POINT

Yours sincerely, Poul Bo Nielsen

cad-quark, 2021-03-16:
Have you checked the RPM with a digital tach?  My BEI encoder is 1000 even so its 4000 pulses per rev rather than 4096.  Thats 2%.  Just like 7/300.

poulbo, 2021-03-20:
Hi cad-quark
You just hit the problem what I thought was 1024 / rev taco counter was a 1000 / rev
It was marked incorrectly. Thank you very much for your inquiry, it all now makes sense.
Also could not understand that lathe was not 1: 1 in position A.
Thanks again for your help.
Yours sincerely
Poul Bo Nielsen

cad-quark, 2021-03-20:
The 2% error was too much of a coincidence.  The clough42 ELS acts like an electric gearbox. Any lost or gained revolutions is either input or output.  Since most users got good threads, it had to be your input setting that did not match.

One thing I have wondered is since you wire up the I (or Z) index, which gives you one pulse per revolution,  TI processor knows when the revolution is complete, but the program must not use that index.

clough42, 2021-03-20:
Correct.  It doesn't use the index pulse.

James

clough42, 2021-03-20:
Actually, allow me to elaborate further.  It doesn't use the index pulse
because the index pulse doesn't necessarily indicate one full revolution of
the spindle if the drive between the spindle and the encoder is not 1:1.

cad-quark, 2021-03-20:
I had no criticism of the ELS design.  I am thrilled we users are able to use the weeks of engineering and code writing that make up ELS you have given us..  I just suspect that somewhere buried in the TI documentation they explain how to use GPIO59, QEP1 to zero GPIO 35,37.

cad-quark, 2021-03-21:
Great,  If you cannot get 1:1 at the spindle, then the number of pulses for one revolution ( calculated with your actual ratio ) can be set in Configuration H and the program math still works.  Wow, what a killer design.