clough42/electronic-leadscrew #142: I try to load the 1.4 but I can’t make active

andrea271068, 2021-03-13:


dsgroninger, 2021-03-13:
I struggled as well when moving from 1.3 to 1.4 in dealing with TI Code Composer.  It sounds like the same issue that you're having.  In the end, I went to the project directory (with Code Composer closed), deleted everything related to 1.3, then brought in 1.4 source and did the import in Code Composer again.  That did the trick for me.  After that, I could build and deploy to the Launch without any problems exactly as James described in his video on programming the LaunchXL.

cad-quark, 2021-03-16:
Another thing you may want to do is load CCS version 10 and drop 1.4 in with it.  Ti says that different CCS can be loaded to a computer.  
I now have two old PCs with ver 9.1 and the M1 Mac with ver 10 CCS.    Big Sur with Apple processor will not open 9.1 setup.  No amount of sudo xattr dash anything magic will take it out of quarantine.  I will let you know if 10 is backwards compatible with 9.1. Don’t hold your breath while I build up the courage to do so.

Easy Lift R210 ( 21 meter ) has been put back in service.  Every function request must pass through at least 5 sets of relay ladder logic sets of contacts due to all the safety built in.  They use some 4 pole relays no bigger than your thumb.  I built a relay tester and found two bad relays.  They arc and corrode since the single cylinder Hatz diesel is hammering 10 cm away.  This summer - after Covid19, I will replace relay sockets and trace all the low voltage problems.  I will send photos to your business mail.  Sorry Git-folks for off topic content.

Tobi75, 2021-03-17:
I think you have to switch the workspace. Menue: File->switch workspace then you can choose in workspace_v10 either "electronic-leadscrew-1.3.01" or "electronic-leadscrew-1.4.00" ...if you have more versions, you should be able to select more...that works for me fine.

cad-quark, 2021-03-17:
That seems right just add another workspace folder.  Soon we be expert: which might mean just a fool away from home.