clough42/electronic-leadscrew #143: Preconfigure start values for thread / feed

Tobi75, 2021-03-16:
in the "userInterface.cpp" you can set the start values as follows:
    this->metric = false; // start out with imperial
    this->thread = false; // start out with feeds
    this->reverse = false; // start out going forward
I understood the code to be used to determine the behavior at startup by selcting the mode but not the values to start with. 

It would be very user friendly, if you could set the start values within each table, e.g. by specifying a row from the tables or the Values (Tables.cpp)
e.g.
inch_thread -> 14 (TPI_FRACTION(140))
inch feed -> 0.008 (THOU_IN_FRACTION(8))
metric thread -> 0,75 (HMM_FRACTION(75))
metric feed -> 0,20 (HMM_FRACTION_FEED(20))

Is it possible to pre-configure these initial values and if so how? I haven't found a place in the code where you can set this yet.
This would be very user friendly.

JonWoellhaf, 2021-03-16:
Tobi75,

An easy -- but not elegant -- way to change the starting values for feeds and threads is to add a line above the first line in the tables in Tables.cpp.

For example, in inch_thread_table, copy the 14 TPI entry at line 55 and paste it immediately above line 48.

So

46 const FEED_THREAD inch_thread_table[] =
47 {
48 { .display = {BLANK, BLANK, BLANK, EIGHT}, .leds = LED_THREAD | LED_TPI, TPI_FRACTION(80) },
...

becomes

46 const FEED_THREAD inch_thread_table[] =
47 {
48  { .display = {BLANK, BLANK, ONE,   FOUR},  .leds = LED_THREAD | LED_TPI, TPI_FRACTION(140) },
49  { .display = {BLANK, BLANK, BLANK, EIGHT}, .leds = LED_THREAD | LED_TPI, TPI_FRACTION(80) },
...

This should cause 14 TPI to appear on power up (I haven't yet tested it), but the first + button push will jump down to 8 TPI.

You can similarly stack several preferred values at the top of the table.

In a much more elegant solution, I think James plans to have the last selected values saved in the eeprom and restored on power up.

Best to do this in the latest version, 1.4.00.

Jan (Yan)

Tobi75, 2021-03-17:
Hello Jan,
thanks for your answer. I think that should work, but wouldn't solve the problem, because its still necessary to click up and down to the preferred value, if not exactly the first line ist meant. 

I actually had two solutions in mind: 
1. initialize variables so that the correct values are set at programstart.
2. digital  the pressing the buttons at program start by calling the function that is activated by pressing button up several times.
both of them I don't know how to program.

"selected values saved in the eeprom and restored on power up" is just that what I want and for now without eprom, just by code. Unfortunately I do not know which values I would have to set at programstart.

haiqu, 2021-03-24:
A better solution would be to use the unused memory chip to save the current value, and restore that on restart.