clough42/electronic-leadscrew #146: LCD Flickers

jonmessenger, 2021-03-24:
I got everything up and running, no problems, works like a dream.  but the LCD flickers intermittently.  It doesn't seem to affect anything.  

anyone else had this issue?

clough42, 2021-03-24:
This is almost always an electrical noise or power issue.

Try to keep the wiring as short as possible and keep the control panel
wiring away from motor wiring.

You should also make sure the GND for your 5V electronics power supply is
not connected to the ground (-) of the motor power supply.

James

jonmessenger, 2021-04-03:
First of all, you were right.  I had made the original display cable long because I didn't know exactly how I was going to arrange things, once I shortened it up, its all good.  Thanks
Secondly I made a brief video overview of the completed project so you would know that you did good.
Clough42 ELS Implementation

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Clough42 ELS Implementation

 |

 |

 |

Thanks again James, "you da man"!

"The stature of a man is not measured by the shadow he casts, but by the shade it provides"John Messenger

Tobi75, 2021-04-21:
I had the same problem. In my case there was a connection between the cable shielding (to the display) an body of the lathe. I used aviator connectors to connect the display an an aluminium box for the display. After mounting the display by screws on the lathe, the display flickered. Isolating the connection between aviator an box solved the problem.