clough42/electronic-leadscrew #148: Omron encoder cost reduction

haiqu, 2021-03-24:
I'm finding that the Omron encoder specified is quite spendy on eBay compared to the 1000 P/R version. Using this one should make little difference to the operation, requires only a minor confiruration change and saves AUD $16.00

tgirard, 2021-03-25:
I got this one off of Ebay and it works fine
https://www.ebay.com/itm/1x-OMRON-1024P-Incremental-Rotary-Encoder-1024p-r-5-24v-DC-E6B2-CWZ6C-NPN/153609282849?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

haiqu, 2021-03-26:
Here's the 1000 P/R version. https://www.ebay.com.au/itm/OMRON-E6B2-CWZ6C-Rotary-Encoder-1000P-R-New/323002258800?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649

I couldn't see the one you listed on eBay Australia, all the 1024 P/R versions were USD $30 (AUD $40) and higher.