clough42/electronic-leadscrew #150: 1mm Metric threading incorrect

GrayTechX, 2021-03-25:
Hi, my imerial threads track perfectly but 1mm metric does not track in same thread, its off by 0.5mm. Could there be an issue with the software calculations as my imperial threading works. 
I'm using v1.3

GrayTechX, 2021-03-25:
Just tried a few more metric threads, all track wrong by just less than half the thread pitch on the second pass. Imperial works correctly.

clough42, 2021-03-25:
That would be pretty shocking, unless there's a defect that somehow only
affects your configuration.

What's your leadscrew pitch?

Can you upload your Configuration.h file?

dsgroninger, 2021-03-25:
Gray,

Are you disengaging and re-engaging your half nuts the same way with metric as imperial?  If so, and you have an imperial lead screw on your lathe, that won't work.  If you have an imperial lead screw, once you cut the first pass of a thread, you can't disengage the half nut until you've completed all your passes or the threads won't line up.

James covers metric threading in this video: https://youtu.be/10Ketpqq-eM

Cutting things this way without being able to release the half nuts can be tricky ending in a groove or at a shoulder.  It's a lot easier to use the upside down and reversed method as shown by 
Joe Pi at Advanced Innovations: https://youtu.be/Z-dqOi_z5bk and by 
Quinn Dunki at BlondiHacks (I think it's the third method shown):  https://youtu.be/q7scadYptTI  

Using that method, there's no need to stop a pass on a spot, it just runs off the end of the part and you can stop or slow to a stop without worrying about a crash.

GrayTechX, 2021-03-26:
Ah, ok. My mistake. I will try metric again without disengaging.