clough42/electronic-leadscrew #151: Another ELS up and running

RebelJRD, 2021-03-26:
I have completed the installation of the ELS on my Smithy Granite 1324.   While this machine does have quick change gear box I like the convenience of the ELS and the spindle RPM display.  And, I already had the Stepper Motor, drive and power supply.  I'm using a NEMA 34 Stepper with 4.5Nm of torque geared 2 to 1 to the lead screw.    This machine also has separate feed from the half-nut so I used the additional setting in the config file.  I did experience some missed steps at high spindle speeds and high feed rates, but those are not practical for normal operation so I'm not concerned.   I mounted the stepper on the tail-stock end of the machine using the existing mounting holes on the end of the base.

The encoder is mounted inside the enclosure on the headstock.   I 3D printed a matching gear and a mounting bracket which worked out nicely.  I do not have a lot of noise from the gears.  I mounted the bracket on the existing gear arm which made setting the backlash on the encoder gear really easy.

I made some fairly heavy test cuts and everything works fine.   Thanks to James for doing all the heavy lifting on the hardware and software design, and making the kits available to us mere mortals.  

[image]
[image]
[image]
[image]

haiqu, 2021-03-26:
Sweet installation! I was considering going this way as well, since I own a 7 x 12 mini-lathe and there's just no room in the housing at the left end for the stepper motor. Good to see it's a workable solution. I'll also be using a standard (i.e. non-hybrid) NEMA 34 stepper, although mine will be 3 Nm as it's a tiny machine.

tgirard, 2021-03-29:
Looks great. I love those Smithy Combos. Cabinet is Nice and tidy as well.  Hoping to have mine done soon.