clough42/electronic-leadscrew #152: Adding a stop switch.

ofer1973, 2021-04-19:
How can I add a micro switch that will stop the stepper motor, after the distance I choose ?

JonWoellhaf, 2021-04-19:
What stepper driver are you using?

From: ofer1973 
Sent: April 19, 2021 15:41
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] Adding a stop switch. (#152)

How can I add a micro switch that will stop the stepper motor, after the distance I choose ?

ofer1973, 2021-04-20:
[image]

ofer1973, 2021-04-20:
The micro switch should connect to the driver, or to the board ?

dsgroninger, 2021-04-20:
A couple of thoughts here:
1.)  A normally closed switch in series with the Enable line between the TI board and the stepper controller should stop the stepper when it opens.  That would essentially cut the enable line.  Now, when that switch closes again the stepper will start running once again.  This could potentially be dangerous if the operator still had the half-nut or drive gear engaged.

2.)  A better solution would be to leave the enable lines alone, and add a microswitch connected to an input pin on the TI Launchpad.  Then modify the code to monitor that switch and stop driving the stepper controller when the switch opens.  I would have the code require a manual reset to start running again, perhaps hit the On/Off button on the control pad, or the Set button, something like that.  The point being to require some positive action beyond just moving the saddle off of the positional stop switch, so that there isn't accidental engagement when the saddle is moved manually.

3.) Either of those stop switch arrangements would be OK to stop movement when feeding, but not when threading.  Feeding can be started and stopped asynchronously from the spindle, but when threading, it is critical that the lead screw is kept in synch with the spindle.  Any interruption in turning the lead screw with the spindle and the tool won't line up with the thread on the next cut.  In theory, it might be possible with option 2 above for the TI Launch to keep track of where on the rotation of the spindle it stopped driving the lead screw and start the screw turning again at the exact angle on the spindle to match back up, but that will require some thought and calculation.

I believe that James stated in one of the videos, or maybe in the notes here, that this project is simply an electronic replacement for the change gears, not the start of a CNC conversion for the lathe.  A stop on position switch is a more complex design problem than it seems on the surface.

ofer1973, 2021-04-20:
Thank you very much for the detailed explanation :-)

Tobi75, 2021-04-21:
maybe the adaption of Kent VanderVelden could solve this https://www.youtube.com/watch?v=AzoYlNzwKkY. He addded a stop switch that stops the Spindel (probably by opening the enable line?). Probably like option 1 of dsgroninger. For threading that wouldn't work.

ofer1973, 2021-04-21:
This is exactly what I want to do ,But in this video he does not explain exactly where to connect the micro switch.
And what needs to change in the code.

Tobi75, 2021-04-21:
the code is on GitHub available: https://github.com/kentavv/electronic-leadscrew/tree/feature-ccs10 
detailed information are also available: https://www.notion.so/Touchscreen-for-Clough42-ELS-99899e746e064c45b51a9397c1d289a5 
maybe it's also possible to contact Kent?

ofer1973, 2021-04-21:
Thank you very much, I will check it out !

mortonhoby, 2021-07-12:
Hi.  I want to try adding a stop switch when in Feed Mode only and have tried an NC switch in series with the enable line (as suggested above) but this doesn't appear to work.   It looks like the ELS implementation switches the line high to +5v when disabling the drive using the PWR button on the display (so an open switch is still at 0v).    I have tried inverting the enable line in the configuration.h file by changing line 78 to #define INVERT_ENABLE_PIN false but this doesn't seem to change anything (at least in this specific regard).  I cant see anyway of changing the logic on the stepper drive to disable on 0v.   I could possibly switch to a NO switch and provide a 5V source to it in parallel to the ELS Boost board but would this be a sane solution?  Any thoughts much appreciated.

clough42, 2021-07-12:
What driver are you using?

mortonhoby, 2021-07-12:
I am using a Hybrid stepper and driver sourced from Amazon UK through a company called Vevor).    The drive is labelled as an HSC57A but I cant find much info about who actually manufactures.     In terms of connections it looks much like any other driver.

 

Connections are:-

Power

DC+

DC-

NC (apparently stands for No Connection)

 

A+

A-

B+

B-

 

PB+

PB-

PA+

PA-

+5V

GND

 

ALM-

ALM+

ENA-

ENA+

DIR-

DIR+

PUL-

PUL+

 

Plus DIP switch for Motor Type (i.e 42/57/60 or 88 Frame size) and Pulses per/rev.   There  also an RS232 connection port but the company say they don’t supply any software.

 

All functions of ELS seem to work perfectly and using the PWR button on the ELS display indeed disables the drive.

 

Roger

clough42, 2021-07-12:
Okay, usually the software would have the ability to change the sense of
the enable line from active-high to active-low.

The enable line on the boost PCB is actually a sinking output, meaning the
ENA+ line is permanently connected to +5V and the ENA- line is either
switched to GND or left to float.  So if enable is active-low, that means
that turning on the LED would disable the drive.  You could connect a N.O.
switch from ENA- to GND, and this should disable the drive, independent of
what the ELS is outputting.

This is how many drivers are configured from the factory (ACTIVE-LOW
enable, a.k.a. LED OFF for enable, a.k.a. ENA- left to float for enable).
This makes it so the motor will run if you don't connect anything to the
ENA lines.

mortonhoby, 2021-07-12:
Many thanks.  I will take another look and try your suggestion.  Thanks for your time and input.  It’s very much appreciated.  

Roger

mortonhoby, 2021-07-13:
Hi James,

Just wanted to let you know that your suggestion worked well and I am now happily using the ELS.  The ELS really does transform the usability of a basic lathe specially when threading but even for general lathe operations the ability to ‘tune’ the feed is amazing.  So glad I stumbled on your channel and opted to try the ELS implementation.

Regards
Roger

clough42, 2021-07-13:
Awesome!  I'm glad it's working for you.

James

LabRatJason, 2022-03-22:
I know I'm resurrecting an old thread, but I think it's pertinent: I recently found a hall effect proximity sensor on Amazon that operates at 5v (most operate at higher voltage).  From the discussion above, it would seem I could connect to the alarm port directly with this and have it function, but the issue would be the inability to resync (I'd really like to thread to a stop for example).

James, is there any reason I couldn't commandeer the alarm pin in the code (I'm using a bog standard stepper, so I don't have an alarm hooked up anyway) and repurpose it to try to enforce the stop, but also note the encoder position to allow for resync?  There are of course many caveats, but I happen to be running my encoder 1:1 with my spindle so for me it would work.  Any advice if I try to take this on?  I've coded many a script language over the years, but never hardware embedded code, so I'm wondering what pitfalls may await me.