clough42/electronic-leadscrew #153: Magic Blue Smoke came out of ELS Boost Board V2.1

steelbarz, 2021-05-06:
Has anyone else had this problem?  I had everything all set up and it was working and just about ready to install on my lathe (Craftex  701).  I was checking out the different feed rates and threading pitches and how it would spin my hybrid servo/stepper (57HB22 With HBS57 Driver).  Setting the display to MM and cycling through the different options until I got to 6.  Spun the encoder by hand and heard a pop, the magic blue smoke came out from the ELS Boost board (V2.1) now both the TI board and ELS Boost are dead.  The power supply for the TI board and ELS Boost was set to exactly 5V and verified with a multimeter.

My setting I changed in config.h

#define STEPPER_MICROSTEPS 20
#define STEPPER_RESOLUTION 200

#define ENCODER_RESOLUTION 4000

[image]
[image]

dsgroninger, 2021-05-07:
Steel,

It looks like the voltage regulator on the interface board let go.  On your second picture, take a close look at the solder joints on that head right beside it where the little flags call out 3V3 on one side and 5V on the other.  Is there a solder short between those pins?  If not, then I'm guessing that something else supplied by 3.3 volts is shorted to ground somewhere.

Dan

steelbarz, 2021-05-07:
Thanks for the reply, I know my soldering job is not the best I blame the flux...lol, but before I applied power to the board I checked continuity as described in the troubleshooting section and made sure there were no shorts, so that was not it.  As I mentioned everything was working great until I cycled through to the highest setting in the MM pitch setting.

[image]

clough42, 2021-05-08:
I don't know.  It's possible it was a bad part, but it's hard to tell at
this point.

I went ahead and soldered the new board just to eliminate that variable.

You might check to make sure the inputs on the servo driver are truly
isolated.  If something is wrong there and it fed high voltage back through
the step/direction lines, that would cause trouble.

James

steelbarz, 2021-05-08:
Thank you James for your help and your efforts.  I also won't be using the same power supply set up to eliminate that variable as well.  This is truly a great project and I know when I get it up and running it will be well worth it.  I just hope I can do it without any more glitches, which never happens....lol  I have a mill CNC project I want to start :)

steelbarz, 2021-06-07:
IT'S ALIVE!

Thank you James for this amazing project.  After a bumpy start, it got all hooked up today and works wonderfully!  I can say this I never wanted to cut threads because of the pain of change gears.  1st thing I did was check out that feature and wow now I think my lathe is even more useful than it was before.

clough42, 2021-06-07:
Awesome!  I'm glad it's working for you.

James