clough42/electronic-leadscrew #154: Help !?

DrNdB, 2021-05-06:
Hi,

I am a complete novice with no skills or knowledge in this area.
I think I have followed the instructions and examples. Flashed the firmware onto the launchpad, but the led&key board does not respond. Checked continuity and have 5 V power. Do you have any suggestions?

JonWoellhaf, 2021-05-06:
Hi,

Have you watched James’ troubleshooting video at https://www.youtube.com/watch?v=84O8iWemqxY&list=PLDlWKv7KIIr90ZZ7Zqt-ge5nVVdS3WVgg&index=21?

Regards, 

Yan

From: DrNdB 
Sent: May 6, 2021 11:16
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] Help !? (#154)

Hi,

I am a complete novice with no skills or knowledge in this area.
I think I have followed the instructions and examples. Flashed the firmware onto the launchpad, but the led&key board does not respond. Checked continuity and have 5 V power. Do you have any suggestions?

DrNdB, 2021-05-06:
Yan,

I had not, and it looks very useful.
Thanks for the suggestion.

Norbert

LabRatJason, 2021-05-07:
I had this problem just yesterday (I'm bench testing my rig right now) and I was able to resolve it:

Re-watch the video on flashing the firmware.  Make sure you've selected "Release" mode rather than "Debug" mode, then click the "debug" icon... It's very unintuitive, and I'm a software developer who has used many other IDEs... I've never seen one that works this way (but also, I don't develop for embedded hardware, so what do I know?)

Good luck.

DrNdB, 2021-05-07:
Thank you, I’ll give it a try!

DrNdB, 2021-05-07:
I have followed everyone’s advice, thank you. This is what I have: two leds on the boost board lighted, continuity from pins on boost board to led&key board and beyond. When I flash the code, either in debug mode or in release mode, the led&key does not respond ( I have actually bought a second board without success)

JonWoellhaf, 2021-05-07:
I’m confident we will be able to get your system working.

Don’t feel too bad. I have 70 years experience working with electronics and computers and software, and I still get confused. 

After you click the bug icon, do you see several messages appearing on the screen? I assume you do.

Are you remembering to click the green right-pointing arrowhead icon? In either Debug or Release mode, the program halts and waits for that icon to be clicked. Of course, this does not happen in Release mode on power up with the computer disconnected.

From: DrNdB 
Sent: May 7, 2021 12:55
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Help !? (#154)

I have followed everyone’s advice, thank you. This is what I have: two leds on the boost board lighted, continuity from pins on boost board to led&key board and beyond. When I flash the code, either in debug mode or in release mode, the led&key does not respond ( I have actually bought a second board without success)

DrNdB, 2021-05-07:
Jon,
Thanks, I have done all that. I am “relatively” sure I have flashed the firmware onto the board. I strongly suspect that there is something amiss with the led&key board, and I have ordered yet another one.
I so much want this project to succeed, because it would vastly increase the utility of my otherwise nice bench top lathe.
Thanks again for your help.

Norbert

JonWoellhaf, 2021-05-07:
If you want to send me your LED&KEY boards, I will be happy to test them.

Are  you a member of PracticalMachinist.com?

From: DrNdB 
Sent: May 7, 2021 13:28
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Help !? (#154)

Jon,
Thanks, I have done all that. I am “relatively” sure I have flashed the firmware onto the board. I strongly suspect that there is something amiss with the led&key board, and I have ordered yet another one.
I so much want this project to succeed, because it would vastly increase the utility of my otherwise nice bench top lathe.
Thanks again for your help.

Norbert

DrNdB, 2021-05-08:
Jon, 

That is a very nice offer. I can send them to you if only for interest to see if they’re any good. I have ordered a third one ( since they’re dirt cheap), but I would be quite interested in the outcome.
Thanks for the offer.

Norbert

steelbarz, 2021-05-08:
One thing that I missed when flashing the update was after it did the upload hit the tiny "play" button to get the software to run on the board.  If your in debug mode and update if you cycle the power all the data gets wiped.   as JonWoellhafand mentioned make sure that it's set to release mode.  Also triple-check the wires from the ELS Boost to the LED board are going to the right pins.  In my case when resoldering the pins that you have to reposition the traces are so weak that I wrecked one board by accident when I pulled the original pins out, I wrecked one of the little solder pads. Yes, good choice on Jame's part they are dirt cheap I had spares on hand.  and his troubleshooting guide identifying where continuity should go helped identify that one for me.

JonWoellhaf, 2021-05-10:
New thought:

Verify the position of the switches on the TI board, including the ones under James' Boost board. Even if they are in the correct position, move them back and forth a couple times. One person reported this fix the problem he was having.

What do you see on the LED&KEY board? Is it completely dark?

Email me at jon@summitaviation.com and I'll send you my mailing address.

JonWoellhaf, 2021-05-18:
Hi, Norbert.

I received your two LED&KEY boards today. As you reported, the red LED to the lower left of the seven-segment displays remained dark when I powered the boards. However, I was able to get the LED to work on one of the boards by resoldering the 5-pin connector. Unfortunately, the board still did not work. I verified signals were getting to U1.

I believe both boards are damaged beyond repair. Fortunately, they're nearly dirt cheap -- or at least they were the last time I bought them. I thought I was ordering two spares, but I actually ordered two sets of two. So I have four extra.

Unsoldering J1 and replacing it with the header in James' kit is definitely not a project I'd recommend a beginner in electronics soldering attempt. I suspect the board was damaged during this process. Did you check the boards before replacing J1?

You said you are a "complete novice" when it comes to electronics and software, so I assume you are not aware of the sensitivity of many modern devices to static discharge. That is possibly what killed your boards, though you should have little static problems in your area. You packaged the boards you sent to me in clear bubble wrap. That stuff is death to electronics. You must use special anti-static bags (usually pink) or bubble wrap or foam, or metalized bags like the one the LED&KEY board comes in. When you handle the boards, grip them by the edges and avoid contacting any pins or traces.

I soldered the connector to one of my spare boards and tested it. I’ll send it to you tomorrow. No charge.

I am concerned your boost board may be similarly damaged. If this LED&KEY board does not work, then I’ll have to suspect the boost board is bad.

So, same deal. If the replacement LED&KEY board doesn’t fix the problem, then send me your boost board and I’ll test it. I’ll include an anti-static bag in case you no longer have the one James used.

Good luck with this!

Jan

DrNdB, 2021-05-18:
Hi Jan,

Thank you so much for helping me so much.
I am (was) a cardiac anesthesiologist and later an orthopedic surgeon volunteering in Africa, so I really don’t know a thing about this stuff.
If my booster board LEDs light up does that imply that it is OK?
Anyway, thank for your generosity in testing the boards and even sending me another one.
I am absolutely determined to get this project done (especially since I have already done the mechanical changes on my SC4 bench lathe).
I am going to wait doing the electronics work until my son comes for a visit in two weeks. He has much more experience with this sort of stuff.
Again, thank a lot and I will keep you posted on the progress.
Best wishes

Norbert

clough42, 2021-05-18:
If the LEDs light up, that means that 5V and 3.3V are present.  It might be
worth connecting the board to power without putting it on the LaunchPad
board just to test the voltages first and make sure they're correct.

James

DrNdB, 2021-05-18:
Wilco

CincyFlyer, 2021-05-20:
Note that the LED&Key PCB is _extremely_ sensitive to long cables; mine will work with a 14" cable, but not with an 18" (and I tested multiples of each!) The power LEDs may light up, but the display may not work, if the cable is unsuitable.
Try with _more than one_ short cable to see if there's an issue.
It's best to test with a LED&Key that you've not altered, fresh out of the box. I trashed my first one, lifted a pad, and didn't see it. Time waster!

LabRatJason, 2021-05-20:
Just as an additional data point, I sacrificed an old 1394 FireWire cable to drive my display and it’s about 60” long. In my bench testing I’ve had no issues with it so far.  We’ll see how it holds up after the install, but it is definitely possible to run longer cables for the display. I think it’s a combination of shielding and capacitance in the cable.  My line of thinking is that I don’t have all the testing capabilities that James has, but I stand a better chance of success if I start with a cable that is made for high speed data transfer (also, it’s a pretty much defunct standard now anyways)

Sent from my iPhone

> On May 20, 2021, at 6:44 AM, CincyFlyer ***@***.***> wrote:
> 
> ﻿
> Note that the LED&Key PCB is extremely sensitive to long cables; mine will work with a 14" cable, but not with an 18" (and I tested multiples of each!)

DrNdB, 2021-05-20:
Thank you both for your input. Gratefully received, and I will keep it in mind if and when I get to test the control box.

CincyFlyer, 2021-05-20:
I wonder if I have any FW cables; probably not at this point. I had some four-conductor with individual shields, and other weird stuff, that I may try. Otherwise, the next iteration will have the display and micro controller board in the same box, at working height.

DrNdB, 2021-05-25:
Hi Jan,

I just received your package! Thank you so much.
I will wait until next weekend when my son will be here to try things out. Just to ensure « adult supervision »
I will let you know the outcome.
BW

Norbert

JonWoellhaf, 2021-05-25:
OK. Great.

Another thought I had is that the cable from the boost board to the LED&KEY board may be defective. I wish I’ would have thought to have you send that, too.

Please keep me informed. I’m very interested in getting your system working! 

Jan

From: DrNdB 
Sent: May 24, 2021 19:05
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Help !? (#154)

Hi Jan,

I just received your package! Thank you so much.
I will wait until next weekend when my son will be here to try things out. Just to ensure « adult supervision »
I will let you know the outcome.
BW

Norbert

1jdupton, 2021-05-30:
> 
> 
> Note that the LED&Key PCB is _extremely_ sensitive to long cables; mine will work with a 14" cable, but not with an 18" (and I tested multiples of each!) The power LEDs may light up, but the display may not work, if the cable is unsuitable.
> 
> 

Every cable on mine is at least 5 feet and working fine. Be sure you have good crimps and solder joints because 4 more inches of cable is definitely not the problem.

dsgroninger, 2021-05-30:
In a system like this, between the parts of the ELS and the lathe itself, there are plenty of electronic noise sources to mess with the serial link between the Launch and keypad/display.  Proper grounding and shielding of the cables is important, and keeping the cables short helps.  When grounding the cable shields, ground at one end, and tie the shields to the system safety ground and not to the digital electronic ground on the boards.  On my system, I tied the safety ground from the mains cord to the metal box where I have the power supplies, motor control and other electronics.  I used a terminal bar system that includes a terminal set for taking the ground through the terminal mounting bar to the box.  I used connectors with shells that ground to the metal box, and tie my cable shields to the connector shells too.  It's not perfect, but the idea is to form a "star", all grounds coming to one common point and radiating out from that common point while avoiding any loops.

All that said, with my lathe setup, it turned out that I could put my control box behind the splash shield of the lathe and it extended just enough above the shield that I could mount my display box on the top edge of the big box.  That let me use a  display cable less than a foot long and shielding wasn't an issue.  There are a couple of pictures in this writeup showing how I arranged the parts in the box and the short cabling.
[Clough42 Electronic Leadscrew.pdf](https://github.com/clough42/electronic-leadscrew/files/6566110/Clough42.Electronic.Leadscrew.pdf)

Hope this is helpful.

CincyFlyer, 2021-05-30:
Yeah, my error was using multiple copies of the same longer cable. I made a new cable, somewhat over half a meter in length, with shielding, and it works properly, so I can move my control panel out of the way of chips!

DrNdB, 2021-06-01:
Hi Jan,

Well,  my adult supervisor has arrived and after a lot of fiddling found a problem with the ground on the boost board. We tried one of the new key & display boards I had bought and lo and behold everything seems to work. I am flabbergasted.
So I am keeping the board you send me to put in the hopefully final iteration, because it is the only one I trust. I have ordered a new boost board because working with a bypass like my son made does not appeal to me.
Anyway, everything seems to work. Next I’ll have to hook up the servo and make certain works and than I might be in business.
I am more grateful than I can express and I will let you know if and when I will get the whole thing installed on the lathe and working.
BW

Norbert

JonWoellhaf, 2021-06-01:
That’s great news! Thanks for the update.

From: DrNdB 
Sent: May 31, 2021 19:29
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Help !? (#154)

Hi Jan,

Well, my adult supervisor has arrived and after a lot of fiddling found a problem with the ground on the boost board. We tried one of the new key & display boards I had bought and lo and behold everything seems to work. I am flabbergasted.
So I am keeping the board you send me to put in the hopefully final iteration, because it is the only one I trust. I have ordered a new boost board because working with a bypass like my son made does not appeal to me.
Anyway, everything seems to work. Next I’ll have to hook up the servo and make certain works and than I might be in business.
I am more grateful than I can express and I will let you know if and when I will get the whole thing installed on the lathe and working.
BW

Norbert

DrNdB, 2021-06-06:
Hi Jan,

It is working :)
I am over the moon. Thank you so much for your help.
We checked and resoldered all the connections. And with your board it now all works!
So I am about to mount it all on my lathe and test it.
Thank you so much for your help.
Best wishes

Norbert

JonWoellhaf, 2021-06-06:
That’s great news! Thanks for the update.

Jan

From: DrNdB 
Sent: June 5, 2021 18:19
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Help !? (#154)

Hi Jan,

It is working :)
I am over the moon. Thank you so much for your help.
We checked and resoldered all the connections. And with your board it now all works!
So I am about to mount it all on my lathe and test it.
Thank you so much for your help.
Best wishes

Norbert

DrNdB, 2021-06-10:
Hi,

I have the system installed on the lathe, and it appears stable and it is running. It is running way too slow though.
I am using the same servo James uses and have a pulley ratio of 1:3 to the lead screw.
In configurations I set micro steps to 1 ans steps to 1000 as in his example, and I am wondering if that could be the problem. 
Do you have any idea?
Thanks for your help.
BW

Norbert

CincyFlyer, 2021-06-10:
Cut a thread (light pass) and see what pitch you get. Metric is easiest to manage. If you want 2mm but get 0.5mm, then it's possible that one of your components (can be either the encoder or the stepper) isn't doing what you want. Check the stepper controller for microsteps, it could have a weird default. Or your encoder is putting out 1K instead of 4K (which will get you my example result.) Easiest fix: you can change any of the parameters so that it works out; if it's running 1/3 speed, undo the adjustment that James suggests. It doesn't really matter what you change, as long as the ultimate ratio comes out right.

dsgroninger, 2021-06-10:
Have a look at the page in here that talks about changing the stepper controller setup:
[Clough42 Electronic Leadscrew.pdf](https://github.com/clough42/electronic-leadscrew/files/6634655/Clough42.Electronic.Leadscrew.pdf)

If you got your stepper and controller from Automation Technologies, the controllers seem to come from the factory set up for 4000 pulses per revolution, and all the math in James's code is set up by default for 1000 pulses per revolution.  You can set micro steps in the code from say James's default 3 to 12 (multiplying by 4) to test things, and if that works, that' probably what's going on.  While you can leave the code set that way and things will work at low spindle speeds, it might trip out at high spindle RPMs.  There's a speed limit of about 100 kHz on the pulses from the TI LaunchPad to the stepper controller, so at 4000 pulses per revolution, times 3 for the belt pulley gearing, gives 12000 pulses to make one turn of the lead screw.  Taking 12 k pulses into 100 kHz limit, that's 8.3 turns per second or 500 RPM.

Better to get the stepper expecting 1000 pulses per second, then you can run things to the full speeds for feeding at top spindle RPM.  James covers that in his videos, and in my document above I have links to the controller programming software and cable.

If your stepper controller is set using DIP switches, be sure of which direction on the switches is 1 (on) and 0(off).  At least one guy on here had his set from the factory at almost the exact opposite of what he wanted, but it wasn't obvious until he took a close up picture and several of us realized at the same time the switches were backward!  The on/off labeling was so small he just didn't notice without the close up.

LabRatJason, 2021-06-10:
I would also add, please validate your encoder is moving at a 1:1 relationship with your chuck. Or if it is intentionally not 1:1 that you have the correct pulses per rev in the code.  On my machine the output shaft that drives the gear train has a selectable 1:4 or 1:2 ratio with the chuck (it has a gear selector inside the head) so I had to correct this by reversing the gear ratio from that output shaft to the encoder.

Good luck,
Jason

DrNdB, 2021-06-11:
Thank you all. I am trying to understand all of this and will try some of your solutions in the coming days.
I’d be delighted to come back and report a resounding success.

DrNdB, 2021-07-28:
Well, I’m delighted to report that it is all working, albeit it at low speed when cutting threads as predicted.
I had significant trouble flashing the code onto the launchpad, but my adult supervisor sorted it all out, and now I’m good.
Thank you all for all your advice and help, and special thanks to James for putting in so much effort to help this community. It has given me much pleasure and I am learning a lot.
Best wishes

N

CincyFlyer, 2021-10-20:
You can change the software on the board, or change the hybrid stepper controller. I just saw where mine was, and adjusted a setting to make the 2:1 change I needed. Doesn't really matter which setting, as long as it affects the pulse in : pulse out ratio in the right amount.

DrNdB, 2021-10-21:
That is great and it worked. Thanks so much.
Now I have another issue: the feed direction won’t reverse, which I need to cut metric threads.
The spindle reverses, but the lead screw does not. Any ideas why not, and what to do about it?
Again your help is greatly appreciated. I feel I’m so close!

SonnyMadlangbayan, 2022-07-12:
> It continues to run forward when the spindle reverses. I know the encoder pin out is correct because all was fine with stepper motor and only failed when I switched to the servo motor.
> […](#)
> Sent from my iPad

SonnyMadlangbayan, 2022-07-12:
Hi, I finally received the launchpad after 4 months of waiting. the systems works except it also not reversing even the spindle is in reverse. it always forward even when I switch the wire A & B. any idea why? thank you in advance.

DrNdB, 2022-07-12:
Hi,

I have exactly the same problem. Have not been able to figure out why, and have been out of the country for a few months. I’ ll try and sort out what is going on, and will let you know what I find.
If you figure it out please let me know what you found.
NdB

clough42, 2022-07-13:
> Hi,
>
> I did try to swap new encoder but it still not reversing. it seems the
> issue is from the microcontroller or maybe the software. I will try to
> troubleshoot further and will let you know if I'm able to fix it.

If the lead screw is not reversing, I would suspect the direction pin on
the stepper driver first.  You can test easily.

If you switch directions using the button on the control panel, and you see
the direction lights change, the screw should then turn the opposite
direction for forward spindle rotation.  If it doesn't, the issue is with
the driver it DIR circuit.

If it changes direction with the button, but does not change direction when
you turn the spindle back and forth by hand, this would be an encoder
issue.  Because of the way quadrature encoders work, this would be a
strange failure I've never seen before.

SonnyMadlangbayan, 2022-07-13:
Hi James and thank you for the response.
I did also try to change the direction pin on the servo driver but it still not reversing. and yes the switch direction on the button change the light between forward and reverse but it is not changing the direction either. I have two encoder and two integrated servo, I test both with other program to confirm it is reversing and I confirm it is working/reversing but not when installed on ELS.
Regards,

SonnyMadlangbayan, 2022-07-13:
I take it back on my previous comment "it seems the issue is from the microcontroller or maybe the software". 
I don't think it's the software if several or many have the same settings (stepperonline integrated servo 180w, 1024 quadrature encoder and latest ELS oftware 1.4.00) and not having the same issue. maybe the problem is the micro controller since I already test two encoder and servo. I order another micro controller and will see the result.
Regards,

SonnyMadlangbayan, 2022-07-15:
it's not fix yet but I measure the voltage of the DIR pin to make sure it is inverting and I confirm it is. when spindle is forward the voltage reading on the DIR and EN pin is 5 to 6 volts and at least negative 1 volt when it is reverse. this telling me that the motor should reverse the direction but for some reason it is not reversing. any idea James? thank you in advance.

SonnyMadlangbayan, 2022-07-15:
I'm a bit confuse on the stepper/servo pin connection. the integrated servo motor from stepper online doesn't have enable pin. and the connection from the interface pc board is using -en pin. the interface pc board has pin "+, - STEP, - DIR, - EN, +ALM, - ALM".
 I did not connect the -EN since there is no enable pin on the servo driver. but I was wondering which to connect the  +DIR pin from the servo driver? also I assume the "+" pin from the interface pc board is directly connected to  +STEP pin of servo driver.
I did all the possible swapping of the wires from encoder (A,B) and servo/stepper wires but still not able to make the motor to reverse direction. it seems that the microcontroler or the motor driver is always in default (forward) mode.

DrNdB, 2022-07-16:
I had exactly the same question (and no answer)

SonnyMadlangbayan, 2022-07-19:
> it's not fix yet but I measure the voltage of the DIR pin to make sure it is inverting and I confirm it is. when spindle is forward the voltage reading on the DIR and EN pin is 5 to 6 volts and at [...]

I measure the voltage again on the servo pulse and direct using the negative side of the DC power supply as reference ground ( I think the first measurement I took is wrong reference since I measure it from the positive reference (+ pulse pin) to negative direct pin. now using 5v DC supply negative as reference I'm measuring +5volts on -pulse and -direct pin during forward and reverse spindle rotation.  I believe both pulse and direct pin voltage should go from 4-5vots (forward) then 0.5volts(reverse). I'm waiting for the second microcontroller f280049c (estimated delivery mid August) but I also have a doubt on the pc interface boost board not inverting the polarity of the voltage signal.

SonnyMadlangbayan, 2022-07-21:
> I had exactly the same question (and no answer)
> […](#)
> Sent from my iPad

> I measure the voltage again on the servo pulse and direct using the negative side of the DC power supply as reference ground ( I think the first measurement I took is wrong reference since I measure [...]

I can't wait for the second microcontroller and I also think it will not fix the issue. so I find other solution. I notice that when putting a jump wire between -DIR and -PULSE the servo will reverse the rotation (also when spindle is forward or reverse the rotation is reverse) so I connect a wire to the spindle toggle (forward/reverse) switch connected to the -DIR and -PULSE. now when I push the toggle switch to spindle reverse direction the wire is now connecting the -DIR and -PULSE reversing the servo rotation. I test it by cutting 1.5mm thread and it is very clean.

SonnyMadlangbayan, 2022-08-10:
out of curiosity I ordered a new ELS booster board (received today). I removed the temporary fixed jumper (connecting -DIR and -PULSE when I toggle the switch to reverse position) and YES the issue is the previous ELS booster board. now with the new ELS booster board the reverse and forward is working properly without putting jumper on -DIR and -PULSE.
[image]
[image]

SonnyMadlangbayan, 2022-08-11:
> Thanks again, I am looking forward to your solution. Best wishes
> […](#)
> Sent from my iPad

I'm not sure if it is the same problem causing with your els but I replaced the els booster and reverse is now working properly. the reverse button is also now working. you can verify this buy checking the voltage between + supply pulse and -Dir pin on servo motor then run the spindle forward and reverse. the voltage should be different when it is forward and reverse (should be negative and positive between the forward and reverse. if the voltage are the same for both forward and reverse then the els booster is the issue (at least this is on my case).

DrNdB, 2022-08-11:
It’s been so long, I’ve forgotten what all the issues were.
I had it working, except for the reverse, and the motor was giving me trouble.
To be honest, I am about to start a new project and I am tempted to put the gears back on….too bad.

SonnyMadlangbayan, 2022-08-11:
its the reverse rotation not working when spindle is in reverse rotation and also the reverse button from the control panel even when reverse led is lit it is not also reversing the lead screw rotation. but after I replaced the els booster board all is working well.
Rgds,

clough42, 2022-10-11:
How do you have the DIR pin wired?  The DIR- of the servo should be
connected to DIR- on the ELS, and DIR+ from the servo must be connected to
the + (+5V) of the ELS.  If you missed the + wire, you would see what
you're seeing.  I assume you've been through all of the continuity checks
in the wiki?

If Q5 fails on the ELS boost board, this could cause behavior like this,
too, though I've not seen it.  You should see the voltage switch
between +3.3V and GND on the pin labeled 1 on the boost board when changing
direction.  Note that just switching the direction button won't toggle the
output until the encoder moves.  The best way to test is to hold the meter
on that pin and move the encoder back and forth to see if it toggles.  And
if everything is wired properly and working, you should see the DIR+ pin
switching between GND and something higher, like 3-5V.  If it's stuck at
3-5V, while pin 1 is toggling, the transistor is probably bad.  If it's
stuck at GND while pin 1 is toggling, it could be a shorted transistor, or
a wiring issue (like a missing or broken + side wire).

James

clough42, 2022-10-11:
The wiring diagrams are shown in the wiki.

The + pin on the interface board is connected to *both *DIR+ and STEP+ of
the servo or stepper driver.
The STEP- pin of the interface is connected to the STEP- pin of the driver.
The DIR- pin of the interface is connected to the DIR- pin of the driver.

Since you have no enable on that driver, you can leave EN- unconnected.
You can also leave the ALM pins unconnected.

James

SonnyMadlangbayan, 2022-10-11:
> How do you have the DIR pin wired? The DIR- of the servo should be connected to DIR- on the ELS, and DIR+ from the servo must be connected to the + (+5V) of the ELS. If you missed the + wire, you [...]

Hi James, 
yes the -DIR pin of the servo is connected to the -DIR pin of the ELS and +DIR is connected to the +5V of the ELS. as I mentioned, it is not reversing the servo motor so I put a jumper switch between -DIR and -Pulse of the servo and the motor reverse its direction ( I connect the jumper switch to the reverse switch of the lathe spindle so when I togle to reverse the the jumper will also connect the -DIR and -Pulse reversing the servo direction). but since I also notice before doing this temporary fix the voltage is not changing and always stay in 4-5 volts when I rotate the spindle forward and reverse I suspect the booster board component is the issue. so I purchase the second booster board and sure enough the issue was solve (I remove the jumper switch on -DIR and -Pulse).