clough42/electronic-leadscrew #155: Electronic leadscrew failure

mistersee, 2021-06-04:
I was testing my ELS on my Hardinge HLV-H today and most of the buttons are no longer functional. The only two buttons that still work are the power button and the forward/reverse button. It was fully functional when I started. I have the latest firmware installed.

JonWoellhaf, 2021-06-04:
Does the rpm readout work?

From: mistersee 
Sent: June 3, 2021 19:53
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

I was testing my ELS on my Hardinge HLV-H today and most of the buttons are no longer functional. The only two buttons that still work are the power button and the forward/reverse button. It was fully functional when I started. I have the latest firmware installed.

mistersee, 2021-06-04:
The rpm readout does work.

JonWoellhaf, 2021-06-04:
Do the LEDs show an invalid configuration? Like MM/IN or FWD/REV both dark or both lit?

From: mistersee 
Sent: June 4, 2021 14:42
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

The rpm readout does work.

mistersee, 2021-06-04:
It is stuck on .005 feed. It does, in fact, feed and I can reverse the feed with the forward/reverse button. Beyond that, I can't change anything.

dsgroninger, 2021-06-04:
Do you have it mounted in a box or to the front panel with the flex buttons in place?  Is there a chance that one of the buttons has a switch stuck down?

It sounds like  communications between the switch/display board and the TI board is working OK and the problem is a stuck button switch or problem in the wiring on the switch/display board itself. 

If the the board is mounted to the black front panel part with the flex buttons in place, I'd pull that off, take out the red flex buttons and make sure all the individual buttons move and pop back out.  If it works like that, then check the fit of the red flex buttons and the whole mechanical stack up and make sure everything goes in and out as it should.

JonWoellhaf, 2021-06-04:
What do you see when power is first applied?

From: mistersee 
Sent: June 4, 2021 14:42
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

The rpm readout does work.

mistersee, 2021-06-04:
Jon: Looking at the display and leds there is nothing out of the ordinary. The power led is lit, the fwd/rev led is lit and I can change the direction and it is running the motor. The feed led is also lit as well as it inch led. The rpm is displayed and the feed rate is displayed. The only buttons that work are the power and fwd/rev.   Carey

From: JonWoellhaf 
Sent: Friday, June 04, 2021 4:55 PM
To: clough42/electronic-leadscrew 
Cc: mistersee ; Author 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

Do the LEDs show an invalid configuration? Like MM/IN or FWD/REV both dark or both lit?

From: mistersee 
Sent: June 4, 2021 14:42
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

The rpm readout does work.

JonWoellhaf, 2021-06-04:
I agree with ds that it’s most likely a mechanical problem with the nonworking buttons.

mistersee, 2021-06-04:
It is mounted in the same aluminum box that James uses. All of the buttons click on and click again when released.  I was practicing cutting threads and was using the fwd/rev button and the power button. Then it quit recognizing most of the buttons. It still runs the stepper and I can still reverse direction. I just can’t change anything else.

From: dsgroninger 
Sent: Friday, June 04, 2021 5:09 PM
To: clough42/electronic-leadscrew 
Cc: mistersee ; Author 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

Do you have it mounted in a box or to the front panel with the flex buttons in place? Is there a chance that one of the buttons has a switch stuck down?

It sounds like communications between the switch/display board and the TI board is working OK and the problem is a stuck button switch or problem in the wiring on the switch/display board itself.

If the the board is mounted to the black front panel part with the flex buttons in place, I'd pull that off, take out the red flex buttons and make sure all the individual buttons move and pop back out. If it works like that, then check the fit of the red flex buttons and the whole mechanical stack up and make sure everything goes in and out as it should.

mistersee, 2021-06-04:
The buttons click when touched and click again when released. All of them.

From: JonWoellhaf 
Sent: Friday, June 04, 2021 5:23 PM
To: clough42/electronic-leadscrew 
Cc: mistersee ; Author 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

I agree with ds that it’s most likely a mechanical problem with the nonworking buttons.

mistersee, 2021-06-04:
It powers up normally. Clough42 then the firmware version number 1.4.00

From: JonWoellhaf 
Sent: Friday, June 04, 2021 5:16 PM
To: clough42/electronic-leadscrew 
Cc: mistersee ; Author 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

What do you see when power is first applied?

From: mistersee 
Sent: June 4, 2021 14:42
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

The rpm readout does work.

mistersee, 2021-06-04:
If it was a stuck button, I couldn’t change direction or power off. James made it so that pushing multiple buttons wouldn’t work in the later firmware.

From: JonWoellhaf 
Sent: Friday, June 04, 2021 5:23 PM
To: clough42/electronic-leadscrew 
Cc: mistersee ; Author 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

I agree with ds that it’s most likely a mechanical problem with the nonworking buttons.

dsgroninger, 2021-06-04:
Hmm, good point.  I forgot that James had multiple button pushes blocked.

One other thought - James also blocks many of the keys from working when things are running.  If the spindle is turning, most of the keys are ignored to prevent asking for something potentially damaging like a sudden reversal of the lead screw while it's spinning.  If things work with the spindle dead stopped and not when it's turning, that's normal operation.

If that wasn't it, that pretty much leaves us with a problem on the display board, and I don't think that there's a schematic for that anywhere in the project since it's an off the shelf eBay thing.  The easiest thing is probably to find a match on eBay and grab another.    https://www.ebay.com/itm/401080292536 

It's hard to tell how the buttons connect to the micro controller on that board - if there were 8 open input ports, the buttons probably each go to a separate pin.  Often times keypads are laid out in a grid electrically, with an output port driving a column and the rows going to an input.  

Anyway, if some switches work, all the display stuff works, that just leaves a problem between the keys and micro on that display board or James's software blocking your requested action.

mistersee, 2021-06-04:
I just bought a couple on Amazon. $6 apiece. I will give that a try. They will be here tomorrow. At least I will be able to rule that out if I still have problems. Thanks.  Carey

From: dsgroninger 
Sent: Friday, June 04, 2021 6:10 PM
To: clough42/electronic-leadscrew 
Cc: mistersee ; Author 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

Hmm, good point. I forgot that James had multiple button pushes blocked.

One other thought - James also blocks many of the keys from working when things are running. If the spindle is turning, most of the keys are ignored to prevent asking for something potentially damaging like a sudden reversal of the lead screw while it's spinning. If things work with the spindle dead stopped and not when it's turning, that's normal operation.

If that wasn't it, that pretty much leaves us with a problem on the display board, and I don't think that there's a schematic for that anywhere in the project since it's an off the shelf eBay thing. The easiest thing is probably to find a match on eBay and grab another. https://www.ebay.com/itm/401080292536

It's hard to tell how the buttons connect to the micro controller on that board - if there were 8 open input ports, the buttons probably each go to a separate pin. Often times keypads are laid out in a grid electrically, with an output port driving a column and the rows going to an input.

Anyway, if some switches work, all the display stuff works, that just leaves a problem between the keys and micro on that display board or James's software blocking your requested action.

mistersee, 2021-06-05:
Today I had some work to do on the lathe and was using the tach feature. I finished up my job and had some time to check the buttons again. It got worse. I lost the fwd/rev button leaving only the power button functional. Amazon delivered a couple of new led/key boards today. I unbagged one and hooked it up to the header and BINGO. Everything is working again. For the sake of science, I took the other board out and tested it again by itself. Still bad. 
Thanks for your help. I am glad it was a $6 fix.

dsgroninger, 2021-06-06:
Good deal, glad that the replacement took care of things.  After I got my system together and working, I ordered a second set of boards from James and have those on a shelf for spares.  I keep  meaning to lay in a spare LaunchPad too,  but haven't gotten to it yet.  All this stuff has should last a long time, but has a limited commercial life and may not be there when I need replacement parts.  The rest of the parts are generic enough to be able to replace easily, but those three parts are unique enough and cheap enough to get the spares now.

paroikoi, 2024-12-07:
@mistersee: I have an HLV-H and am interested in doing this ELS conversion. I haven't been able to find anyone else who has done it, but you have per above. Any chance you can share with me any tips, what it cost you, etc.? My Google email is the same as my user ID here. Thanks in advance.

Josh

Turner75, 2024-12-08:
Hi Josh

I built and fitted the ELS, plus a variable frequency drive to a new
Grizzly G6020 lathe, about three years ago. No more change wheels grinding
and clanking! The parts from Clough42 back then were a bit iffy, the
interface board had a few issues, the readout / push button unit, I think
was a bought in item and had a couple of diodes wired wrongly. The 22
videos were the key to understanding how it all works. Most of the threads
reporting troubles centered around the Ti  DSP board/encoder/stepper/servo
drive programming. DSP unit price has increased quite a lot. A lot of the
threads reported troubles with encoder reliability, suggestion, buy the
best you can afford, not the $25 - $40 Ebay stuff. I have used Swiss Baumer
encoders over the years,  very good long term reliability. Getting the
programming environment, IDE, setup can be a bit tricky, but not too
onerous. Far more queries are reported on servo motor setups than the
simpler stepper motor/drive that I used. Final comment. When it's working,
its a a joy to use, never go back to gears.

That my two penno'th, for what it's worth.

Graham

paroikoi, 2024-12-08:
Hi Graham,
Thanks for all the info. I'm just learning about ELSs now. I was hoping to
not have to reinvent the wheel, so to speak, as @mistersee has done this on
the same lathe as me, Hardinge HLV-H.

Josh

mistersee, 2024-12-10:
paroikoi:  I took my Hardinge HLV-H as partial payment for my airplane when I sold it. The lathe was inoperable due to missing axis motors and control. That was perfect for me since I was going to do a CNC conversion anyway. The Z-axis has a ballscrew mounted on the backside and the X- axis has a direct drive stepper motor on the front. 
This year I upgraded from Mach 3 to a Centroid Acorn controller. Big job with lots of wiring and failsafes. That presented another problem. My original conversion implemented Vexta 5phase stepper motors @ 1000 steps/revolution. Very precise for Mach 3 but choppy for Centroid. Centroid wants 1600steps/rev minimum . I tricked it into running the 5 phase motors but it just wasn’t smooth. So I had to change motors. I went with hybrid steppers with encoders. 
The end goal was to be able to do threading. Mach 3 never was able to properly implement that feature. So I no longer needed the ELS for that purpose and it is now out of the system. I will probably use it on one of my other lathes. 

From: paroikoi 
Sent: Saturday, December 7, 2024 2:07 PM
To: clough42/electronic-leadscrew 
Cc: mistersee ; Mention 
Subject: Re: [clough42/electronic-leadscrew] Electronic leadscrew failure (#155)

@mistersee: I have an HLV-H and am interested in doing this ELS conversion. I haven't been able to find anyone else who has done it, but you have per above. Any chance you can share with me any tips, what it cost you, etc.? My Google email is the same as my user ID here. Thanks in advance.

Josh

paroikoi, 2024-12-15:
mistersee: Thanks for the helpful info.  Sorry for taking so long to
reply.  Would you happen to have any photos of how you mounted the hardware
(stepper, spindle encoder, etc.) on your HLV-H?  Again, if direct emailing
is easier or preferred, my Google email is the same as my user ID here...

Turner75, 2024-12-16:
Hi Josh
Not being fortunate to be in possession of a Hardinge, i would think think
the physical placement are going to differ quite substantially.
My leadscrew drive motor was mounted in a gap in the headstock casting
below the spindle bearing housing. The timing pulleys in line with the
toothed timing belt with a tensioner pulley riding on the belt to maintain
tension and prevent it slipping a tooth.
The encoder was on a bracket that was spaced off the headstock caseing to
line up with the other driven wheel off the spindle rear extension, means
provided by slackening off the screws holding the encoder on a plate that
can swing way 😳thereby providing a means of putting tension on the belt,
or decoupling the encoder if not using the ELS function.