clough42/electronic-leadscrew #156: Step count issue

Tactilenuffin, 2021-06-07:
I just set up my whole system and everything powered on and worked but when I went to test the movement of the leadscrew it was barely moving. I put it into thread mode to cut a 8 tpi thread and it ended up cutting something very different. I adjusted the step count from 3 (3to1 reduction) to 12 and then tried the 8 tpi cut again and it came out to something like 24 tpi. So I changed the step count in the code again (I don't remember the exact number I put but after trial and error I got it right) the encoder is showing the right rpm so why is the step count wrong.

Any information would be super helpful.

JonWoellhaf, 2021-06-07:
Hi, Tactile.

In Configuration.h, what do you have for 
    #define LEADSCREW_TPI,
    #define STEPPER_MICROSTEPS, and
    #define STEPPER_RESOLUTION 200

Jan (Yan)

dsgroninger, 2021-06-07:
There is a strong chance that your stepper motor controller is set up to need way more than 1000 pulses from the Launchpad for each revolution of your stepper motor.  Please check the stepper controller manual.  You can set the code on the LaunchPad to deal with other than 1000 pulses per revolution, but there is a speed limit of about 100,000 pulses per second, so going too high here can be an issue.

Some controllers use dip switches to set this, others use a programming cable and software from the controller manufacturer.  It's not hard to do, but you'll need the right cable.  The software can usually be downloaded from the manufacturer's web site.  Please see James's notes in this thread below:  Factory settings on Hybrid Servo #140 for the setup used on some controllers that use the cable and software to set.

One other person on this site had a controller that used switches to set up, but the problem he had was seeing which way was on (1) or off (0).  It was marked in a tiny font on the switch, but the factory settings were almost the exact opposite of what he needed and it wasn't obvious till he used a close up picture to magnify things.  I'm sorry, I can't find the thread on here that had a whole discussion (with pictures) where we figured out the backwards switch issue.

Tactilenuffin, 2021-06-07:
> Hi, Tactile.
> 
> In Configuration.h, what do you have for
> #define LEADSCREW_TPI,
> #define STEPPER_MICROSTEPS, and
> #define STEPPER_RESOLUTION 200
> 
> Jan (Yan)

LEADSCREW_TPI 12
#define STEPPER_MICROSTEPS 12
#define STEPPER_RESOLUTION 1000
i was able to change the settings to this to eventually get it to work and cut the 8 tpi and 24 tpi threads but i dont know why this worked if you have any insight.

if the other commenter is right and i neeed to buy the cable and program the stepper i guess i will do that because i would rather have it be 100% then just ckinda right.

Thanks

dsgroninger, 2021-06-07:
If 12 works for micro steps, that implies that your stepper controller is expecting 4000 pulses per revolution.

If your lead screw truly is 12 threads per inch, and you have the 3:1 belt drive reduction that James used, then micro steps of 3 and 1000 would work.  If the stepper controller is expecting 4000 pulses per revolution, then 4x3=12 will work.  If I remember right, my controller was set for 4000 steps per rev when I got it from Automation Technology.

The settings that you have for now will be fine for threading at slower spindle speeds, but you may find that it trips out at higher feed speeds at higher spindle RPM.  That's when he LaunchPad will be clocking the stepper driver fastest, and that will trip out the software at 100,000 pulses per second.  Good news is that you can still have fun with it till your programming cable arrives.

Tactilenuffin, 2021-06-07:
do i need a rs232 cable in order to change the settings? is the software for it free

sorry for the stupid and easy questions i am very new to steppers and you guys are being super helpful

dsgroninger, 2021-06-07:
For the stuff that I got from Automation Technology, here's the link to the cable I needed:  https://www.automationtechnologiesinc.com/products-page/nema23-closed-loop-stepper-motor-system-hybrid-servo-kit/calibration-cabel

I put this document together on how I implemented everything for my Precision Matthews lathe.  There's not much in here that James didn't cover better in his videos, but I tried to condense it all into one document with the notes on things that I needed to do to fit it to the PM1030.
[Clough42 Electronic Leadscrew.pdf](https://github.com/clough42/electronic-leadscrew/files/6612532/Clough42.Electronic.Leadscrew.pdf)

Links are in there for all the parts that I used, and the configuration.h settings that I needed to make things right.  I have notes on the stepper controller setup too, and links to where James covered that in detail.

Tactilenuffin, 2021-06-08:
thanks so much for this. its super helpful.

Tactilenuffin, 2021-06-08:
so that cable is to rs232 so i would also need a rs232 to usb cable right if i dont have the connection on my computer?

dsgroninger, 2021-06-08:
Yes, you'll need an RS232 adapter if your computer doesn't have a port with a D-sub 9 RS232 port.