clough42/electronic-leadscrew #157: [Project/Execution] Another Mini 7x14

AlpineWhite, 2021-06-14:
#25 inspired me to look at using existing mounting solutions for minis. The loudest , most dreadful, least-square part of my geartrain was the fwd/reverse and 45T/20T shelf mounted pulley. That sucker's stepped spindle was in a bore entirely too large for it.

Additional BOM:
Stepperonline CL57Y digital driver
Stepperonline NEMA24 4Nm 24HS39-4204D
2x 40T GT2 pulleys for stepper/leadscrew
1x SDP/SI A6A51M074NF0908 72T GT2 9mm pulley (no flange) for spindle, bored to 28mm w/ 4mm hand-filed keyway
1x SDP/SI A6A51M032DF0606 32T GT2 6mm flanged pulley for encoder. (6mm bore)
Standard Omron 1024P/R encoder 
1x SDP/SI A6R51M124060 124-tooth GT2 belt for spindle<->encoder
200mm GT2 Closed-loop belt for motor<->leadscrew
1/4" 6061 plate for mounting
100mm 6mm diam steel/Alu rod for stepper spacers
48V 7A Meanwell PSU 
1x 10mm M3 cap head screw for top encoder mount
2x 15mm cap head screw for bottom 2 encoder mount holes
4x 30mm M4 cap head screw for motor mount (40mm if you want additional nuts on the back of mount plate)

- Encoder holder is printed out of PETG. Encoder is rated for 20Nm of axial load and 10Nm of thrust load. Encoder holder replaces previous shelf gear arrangement on a 5* pivoting mount for belt tensioning. 
- Motor is 1:1 with leadscrew since it's 4Nm and it's just a 7x14 which won't be doing very deep cuts anyway. If more gearing is needed, cheap pulleys are available on Amazon. 
- CL57Y driver set to 1600uStep/rev for balance of noise:speed with 100k step limit
- Mounting plate is slotted at the banjo mount and at the pillow block bearing to facilitate belt tensioning with 6mm fore-aft travel
- Motor is 60mm between centers to use commonly-available belts from Amazon. 
- Spacers are bored to 4.5mm and are 21mm in length
- Clearance preserved for outboard spider on spindle 
- Encoder:spindle ratio is 2.3125:1. For quadrature mode, that's 9472 pulse/rev with a 1024 P/R encoder. Chosen to avoid floats.
- ZIP file attached of STL for the mount cutting guide and encoder mount. 
- Mounting plate is tapped with M4 thread for motor mount.
- 2-point mount on motor plate allows for mfg variange in leadscrew and banjo positioning since motor, leadscrew, and pillow are all collinear. Mount will tilt/pivot about banjo stud but will maintain distance between drive components.
[minilathe.zip](https://github.com/clough42/electronic-leadscrew/files/6649024/minilathe.zip)

[image]
[image]
[image]

How to pronounce "minilathe":  MuH-niL-UH-thUH (see https://www.youtube.com/watch?v=05vUCdzhoe4)

I have made everything except the standoffs and mount plate. I plan to get a larger lathe at some point in the future, so I will rout a pocket in my workbench for the control panel and fasten the electronics to the underside (it's wood, so no ground wonkiness).  Big thanks to James for working the code and electromech side of this project!

AlpineWhite, 2021-06-21:
The 47.14mm dimension between mounts on the adapter plate is for a NEMA23 motor. Please note the actual radius of mounting holes for the NEMA24 StepperOnline motor is 50mm and the clearance hole for the shaft should be 10-11mm, not 9mm (for an 8mm shaft).

RyanH122, 2021-09-27:
Looks nice!  I basically built a very similar mount out of aluminum to use the existing hole in the bed. 

Slightly off topic but I'm trying to get a StepperOnline CL57T to work. I'm getting an error (error light flashing) that according to the PDF, says "The acceleration time setting is too small,increase the value of appropriately".  StepperOnline says that it needs changed in the controller. Did you have to change anything in the software?  Can you please tell me your dip switch settings on the CL57T?  I'd appreciate the help. I really want this to work. :-)

AlpineWhite, 2021-09-28:
I use the CL57**Y** so it may not be apples to apples. I set the uSteps to 1600, set tracking mode to enable (so the driver checks for lost steps during motion), set to PUL/DIR mode (pulse = step and direction hi/lo).

Checking the CL57T driver manual, I am seeing a few possible solutions for you before wiring an RS232 adapter to interface with the driver software IF you have the V4 (blue text on driver nameplate)
- Increase the current trimpot S1 to max of your motor minus 10% (for both the NEMA23 and NEMA24 from OMC set to 8-9 on the trimpot for about 5A peak. Peak current will be held at standstill where it may sit for a long time if you forget to hit PWR on the control display, so make sure not to overprovision)
- Increase PSU output. I use 48V and it has no problem. 
- A warped or misaligned leadscrew (or bearings) can cause unnecessary torque requirements. I bored out the headstock end support and put in an NKIA5901 from SKF (combined needle roller and ball thrust bearing) and then at the tailstock end put in an 8mm needle thrust bearing. My bearing supports were not bored axially straight, so the leadscrew was almost binding straight from factory. If your tailstock end is too tight, it will also bind like crazy without a thrust bearing at each end.  Installing the NKIA requires turning down the front end of the leadscrew to 12mm in the lathe and also boring the support in the lathe. 
- Check for bearing block misalignment on the leadscrew. Mine was WAY off. I shimmed it (attache
[image]
d
[image]
[image]
)
- On the V4 CL57T controller, SW8 set to 'ON' will enable s-curve acceleration. This means you CANNOT start the lathe immediately at the thread, but if you start half an inch before the threads, it will come up to speed more easily. Both potentiometer current control and s-curve are not present on V3 CL57T nor are either available eon CL57Y (all digital).

RyanH122, 2021-09-28:
Thank you for all the information!  Unfortunately even though I just got the kit from Amazon but I got a V3 (I think) controller so no pot and my PSU is set at 48V.  I'm trying to get it to run off the lathe now.  I had it mounted up but needed to get to a computer so it's on the floor right now, LOL.  All I can do is get it to buzz mostly and maybe jiggle then kick out.

StepperOnline says I need to increase the acceleration time in the host software but I'm hesitant to mess with anything in there and I know less than nothing about code.

That bearing setup you did looks good!  I got lucky on the blocks on my lathe and just had to open up one of the mounting holes to let it rotate a bit and the leadscrew spins by finger nice and easy.