clough42/electronic-leadscrew #158: Omron Encoder Problem

mortonhoby, 2021-06-22:
I am trying to adapt the ELS to my SIEG-SC4 lathe.  Everything looks good apart from the fact that I cannot get any response from the encoder.   Indeed I have sourced 2 OMRON E6B20-CWZ6C from 2 different sources on eBay (Chinese copies I presume).   The quality looks good.   I have wired as per the instructions i.e Brown(5v), Black(A), White(B), Orange(Z), Blue(0V).

I have tried changing from J12 to J13 (and changing the Config parameters to match) but still nothing.

The ELS display appears to work perfectly so don't think I have damaged the Launchpad PCB.   I am trying to get hold of an oscilloscope to check the encoders are OK but I find it a bit strange that both appear defective and therefore think I must have missed something.

If anyone has any ideas please let me know.  Hopefully someone can point me in the right direction.

clough42, 2021-06-22:
Definitely check the positions of the switches on the LaunchPad board.
Flip them and flip them back to be sure they're fully seated.

James

mortonhoby, 2021-06-22:
Thank you so much James.   All sorted.   Just shows to double check everything.   I was convinced I had set the jumpers but because they were under the boost board I didn't re-check.   Thanks again and keep posting.  You certainly have a great presentation style.

Roger

closed 2021-06-22 by mortonhoby