clough42/electronic-leadscrew #161: RPM and direction issue

maxjenius, 2021-06-25:
Hello, I'm having an issue where my spindle is running at 300 RPM, but the panel says its going 1400 RPM or thereabouts.  The servo runs fine, speeds up and slows down as it should, but trying to use REV, it spins the same direction.  Connecting to the driver through serial, I can spin it either direction just fine.  I just can't figure out why...all my connections are correct and continuity for everything is good.  Anyone see this problem?

maxjenius, 2021-06-27:
Update:  Downgraded to 1.3.0.1 and back.  This fixed the RPM issue, but still cannot get the leadscrew to run in reverse....odd.

JonWoellhaf, 2021-06-27:
In James' project Wiki at <https://github.com/clough42/electronic-leadscrew/wiki/Continuity-Checks> he details how to check end-to-end continuity by probing board pads. Have you tried that for the DIR line? I found one board that had an unsoldered pin that didn't make contact with the pad. This process can also diagnose a defective solder joint.

Do you have an oscilloscope, or access to one?

maxjenius, 2021-06-28:
Thank you so much for helping!  Yes sir, I went over that section and checked all lines for continuity:  they all passed.  I do have a scope, what should I probe and look for?

JonWoellhaf, 2021-06-28:
Are you able to probe the boost board while your lathe is running?

maxjenius, 2021-06-28:
Sure thing.

JonWoellhaf, 2021-06-28:
OK. Please try this:
1) In the Wiki article, scroll down to the Output Pins section and locate transistor Q5, then find it on the board.
2) Set your lathe to run at a couple hundred rpm.
3) Ground your scope lead and probe Q5's top terminal. It should be either 0 or about 5 volts.
4) Stop the lathe.
5) Push the FWD/REV button. (In the latest software, FWD/REV is inactive when the lathe is running.)
6) Verify the panel FWD/REV LEDs swapped.
7) Start the lathe.
The voltage should be the opposite of what it was before.
If it isn't opposite, probe either end of R8 (below Q5) and make the same test.
Let us know what you see.

maxjenius, 2021-06-28:
OK, so I probed the top of Q5 while the lathe was running...I get 0V in either FWD or REV (LEDS swap fine).  Probing R8 resister gives me 3.3V in FWD and OV in REV!  The transistor the issue?

JonWoellhaf, 2021-06-28:
It sure looks like Q5 is bad. Sorry.
It's a long shot, but probe the left-hand bottom lead of Q5 and see if the voltage changes.
If it does, then it's pretty certain Q5 is bad.
If the voltage doesn't change at the left lead of Q5, but DOES change at the left end of R8, then it's an open in the very short board trace between the two.
Another thought: Can you examine the board around Q5 with a 10x loupe? Compare the connections and components around Q4 and Q6 with those around Q5. All three should look identical.

JonWoellhaf, 2021-06-28:
I just had another thought.
It may be that the DIR output is shorted to ground somewhere.
To check this,
1) Power off.
2) Remove the wire from the DIR pin on the driver connector.
3) Measure the resistance between ground and the connector screw.
I get an open circuit.

maxjenius, 2021-06-28:
OK, confirmed everything.  Left bottom lead of Q5 changes along with R8.  No shorts detected when testing as you suggested.  Inspecting with a loope looks good, traces are all identical.  Looks like a bad Q5, which is good since I can easily replace it.  I'll check my spare parts bin to see if I have a 7002 transistor and swap it out.  I'll report back.

JonWoellhaf, 2021-06-28:
OK. It's great that you can replace Q5!
I thought of one last test to confirm Q5 is bad.
1) Remove the driver wire from DIR on the driver connector.
2) On the driver connector, put a resistor (1K to 10K or so) between +5 and DIR.
2) Probe DIR and see if stays at 0 volts when you do the test.

maxjenius, 2021-06-28:
I ended up replacing Q5 before I saw your last message.  Looks like you nailed it, everything works now.  Thank you so much Jon for the help, I never would have suspected that component.  Time to make some screws!

closed 2021-06-28 by maxjenius

JonWoellhaf, 2021-06-28:
Excellent! A bad transistor was about the last suspect on my list, because they're usually so reliable -- unless you let the magic smoke out, of course. Did you see any physical damage? I'm delighted you got it working. I haven't heard of another bad transistor on one of James' Boost boards.

James, if you're listening, did you find any bad ones during your board testing?

maxjenius, 2021-06-29:
No physical damage at all, board was pristine.  I was not handling it while grounded though, so it may have been my fault.  I've built plenty of boards without, but hey, it does happen.  Thanks again for the help!