clough42/electronic-leadscrew #163: Programming launchHXL

GrayOneAus, 2021-07-12:
Hi James
Have recieved your Electronic lead screw control panel and ELS Boost board v 2.1
So followed your very good youtube on how to program the board (current version of ccs_setup 9.3)
All seemed to go well (brand new laptop no programs loaded) tried debug mode program loaded (ver 1.3) latest
However the only thing the control did was to light up ie all leds on and display showing 7 and the rest are 8's buttons do not respond. no error reported by the loader.
Have gone through all your board tests and double checked all the connections all appear to be OK all power light are as they should be. (only have the launchHXL and control panel connected as per the youtube leads are 4" long only change I made to the config file was to make it metric at 300 for 3mm pitch
[image]
Would you have any idears as to the problem ?

clough42, 2021-07-12:
That is really odd.  Usually, it either works completely or doesn't work at
all.  Have you checked continuity using the instructions on the GitHub wiki?

Does the display change at all if you press any of the buttons?

GrayOneAus, 2021-07-12:
Yes I have gone through all the instructions (all check out as required)
No the display does not change when you press the buttons.

I believe it is some where in the programing that I have not got right (like other programmers did you have to load a driver for the display?) It would be nice if you could do a youtube on how the debugger works as you explain things very well
Oh yes changed the connections to screen cable to see if that was an issue, no change still the same problem.

GrayOneAus, 2021-07-14:
Hi James
Re above have watched your Youtube a number if times, and suddenly relised 
The led 5 is flashing on my TL LaunchPad after pressing the run key in the debugger. 
It does not in your youtube. does this give you a clue as to what is wrong with my setup!

GrayOneAus, 2021-07-20:
Hi James
Finally got it working!
It was that nasty Code composer (getting your version was real hard kept getting some sort of corruption) any way after trying the third computer  finally it clicked.
Just awaiting the for the encoder to arrive to try the next part.

clough42, 2021-07-20:
Awesome!  Glad you got it working.

James

GrayOneAus, 2021-07-22:
Hi James
Can finally report that the unit works on the bench, (having just recieved the rotary encoder). 

One comment I would make, see attached pic cables would be better off in protected flexable conduit may also help with screening.

And finally you need a DRO on your lathe (would make a good project, and yes I have one fitted make turning a breeze)
[image]
Keep up the great Youtube stuff

closed 2021-07-22 by GrayOneAus

clough42, 2021-07-22:
The flex shield is a good tip.  It'll work best if you connect the shield
to the chassis ground at one end only.

James