clough42/electronic-leadscrew #164: Connecting to motor to motor tuning software.

IanScout, 2021-07-13:
I'm nearly there. My electronics are working. Code loaded. Configuration done.

However the motor and encoder are not turning in the correct ratios to each other. I'm pretty sure it is because the pulses per revolution of my motor is not an appropriate amount.

I'm trying the integrated servo motor which James recently tested - Servo Motor - NEMA 23 Integrated Easy Servo Motor 180w 3000rpm 0.6Nm(84.98oz.in) 20-50VDC Brushless DC Servo Motor. I think it defaults to 4884 pulses per revolution. Because when I look into the parameters in the tuning software in the offline mode it shows this.
[image]
I'm wanting to change this to 500 pulses/rev as James did in his video.

So my issue is that I'm having trouble connecting the tuning software to the motor to change the pulses/rev parameter.
"No find related driver!"
[image]
"Connect the drive fails."
[image]

I've updated the driver for the USB to Serial adapter and done a continuity test on the cable that came with the motor and that is fine.

IanScout, 2021-07-13:
Is there a way to check the USB to serial adapter?

clough42, 2021-07-13:
I had exactly the same problem.  StepperOnline ultimately sent me one of
their USB serial cables and it worked.

https://www.omc-stepperonline.com/cable-and-connector/rs232-adapter-cable-to-usb-2-0.html

IanScout, 2021-07-13:
Okay. Thanks James. I'll order one.

IanScout, 2021-07-19:
The connection issue was solved with the adapter cable from Stepperonline as suggested by James. Finding the correct driver was difficult. Eventually I found that this adapter uses a Prolific driver and the reliable source for the driver software was found on the Prolific website. Hope this is useful for others.

patl1, 2023-06-11:
I just purchased the BestBuy prolific cable ($20 and no shipping cost) and it worked for me to connect to the stepper online serevo driver. (the cable I had on hand from 2012, even though it was Prolific did not work)

Chaney101, 2023-08-13:
Did anyone ever determine the best driver for the cable? I ordered a prolific cable with the same specs and installed the driver from the benfei website but my computer is still not recognizing it? I just ordered the ugreen (same one stepperonline sells) but specs look the same as the one I have? Thanks

patl1, 2023-08-14:
I bought mine at BestBuy and it worked perfectly...nsignia™ - 1.3' USB-to-RS-232 (DB9) PDA/Serial Adapter Cable, with Prolific Chipset - Black