clough42/electronic-leadscrew #165: Power Feed not using half nuts

BL-Shopwork, 2021-07-17:
Another Clough42 ELS up and running! Thank you James for putting this together in a great video series and how to. I am running the "Clearpath-Killer" you recently posted, its working great.
[image]

[image]

I managed to tuck the Servo and encoder into the head below and above the ways. next up is mounting the interface on the front.

Feature I would like to put into the code:
I am fortunate to have a Power feed carriage and I prefer not to use the half nut unless specifically threading. My carriage has a 2:1 reduction for this power feed. I would like to have the "Feed" mode of the ELS base change feedrate on an additional 2:1 reduction. When in "Thread" mode it will for obvious reasons run without the additional gearing.  

[image]

(PLEASE let me know if this has already been done by one you fine folks, hate to re-invent if not needed)

dsgroninger, 2021-07-17:
BL,

It's really easy to set up for separate feed rates if your carriage has a gearbox feed.  My Precision Matthews 1030 has this setup, and it's just one or two lines of code to change in configuration.h.  Here's a writeup that I did on my implementation.  There's a page in there on configuration.h changes and the math that I used to figure out my settings.
[Clough42 Electronic Leadscrew.pdf](https://github.com/clough42/electronic-leadscrew/files/6835349/Clough42.Electronic.Leadscrew.pdf)

Good luck,

Dan

BL-Shopwork, 2021-07-26:
Thanks for the writeup Dan, I don't know how I missed that line of code regarding feed micro stepping but it did exactly what I needed. For me I had to double the micro step value and its working great.

With respect to the feed stop I started down a path that was turning overly complex (for what I was trying to do). I was trying to setup new inputs to disable via software, then a magnetic switch to trigger said input and around I went for the past week until I just decided to put a simple NC momentary limit switch (button) in series with the Step/Dir + wire from the launchpad breakout to the servo...... Maybe not the most elegant, but its works really well. 

Now when the limit is hit, the servo stops dead. Simple, effective and super easy. I also put a switch in parallel to the feed stop switch so that I can disable the feature for example when threading or its just not needed.

SoManyHobbies, 2021-07-27:
I was curious if anyone had run into cases where the ratio between the half nut and power feed couldn't be accounted for by just adjusting the STEPPER_MICROSTEPS_FEED and STEPPER_RESOLUTION_FEED values?
I have a PM 1228 lathe and have been struggling to figure out how to account for the difference in power feed rate.

I've got an 8 TPI lead screw, and the 180W Integrated Servo, and plan to use a 6:1 ratio (3:1 motor to lead screw input, and 2:1 in the gear box similar to how Cough42 had his setup in his video).

I'm using a STEPPER_RESOLUTION value of 1000, and STEPPER_MICROSTEPS of 6.

However the problem comes in with the carriage power feed. If, for example, I have the gear box set to thread 16 TPI (0.0625 in/rev), that same gear box setting would drive the carriage 0.0072 in/rev. Which if expressed as a ratio of (thread feed)/(power feed) would be 1/0.1152, or if "reduced" to the smallest whole values is 625/72.

Maybe I'm just thinking about it all wrong, but I don't see how I can set the STEPPER_MICROSTEPS_FEED and STEPPER_RESOLUTION_FEED to account for a difference like that.

I am comfortable writing in C++ so if there are any suggestions of a change I could make to use a different calculation when in feed mode I am willing to give it a shot.

dsgroninger, 2021-07-27:
SoMany,

Please look up two items in this thread where I linked to my writeup on my implementation on a PM1030.  It deals with exactly the situation that you have in using the separate gear ratio of the carriage feed rather than the half nut for feed.  For the 1030, I think the gearbox ratio was 3.33:1, so the math worked out to round integers for me.  I think that if you find this line in configuration.h:
#define STEPPER_MICROSTEPS_FEED

and set that to twice what his line is set to:
#define STEPPER_MICROSTEPS 3 

If that's 6 for you and the threading feed is correct, then set STEPPER_MICROSTEPS_FEED to 12.
#define STEPPER_MICROSTEPS 6
#define STEPPER_MICROSTEPS_FEED 12
 

Anyway, I have the lines of configuration.h that I changed listed in my writeup.

Good luck!

Dan

SoManyHobbies, 2021-07-27:
I did read through your writeup, and it was really helpful. But I think the part that I got hung up on was that the ratio in my case didn't work out so nicely to result in whole numbers.

I do have STEPPER_MICROSTEPS set to 6 currently.
But if I understand things correctly setting STEPPER_MICROSTEPS_FEED would basically double the feed per rev when in feed mode. But in my case the power feed rate isn't half the threading feed rate, it's closer to 8 times slower (8.680555555......) to be more precise.

If I did just assume it was 8x, I could set STEPPER_MICROSTEPS_FEED to 48 (ie. 6 x 8). If my math is right that would result in a difference of about 8% between the calculated feed rate and true feed rate. Probably not a big deal at low feed rates, but might become a concern at higher feed rates.

Does that seem right, or am I just making this too complicated? :)

dsgroninger, 2021-07-27:
Wow, 8.68:1 seems like a lot.  I spent a lot of time turning the lead screw by hand and watching feed motion on the DRO to come to my measurement of 3.33:1 on the PM1030.

If it really is 8.68, then if your threading micro steps are 6, then 8.68 x 6 = 52.08.  For feeding, what's it matter if the rate is just slightly off?  Set STEPPER_MICROSTEPS_FEED to 52 and you're only off by less than 0.2%.

Now, the only problem with all this is that at high spindle RPMs, you're going to exceed the 100 kHz limit on clocking the stepper controller.  James has the code set up to trip out if the clock to the stepper goes above 100 kHz.  For me, it's more  of a problem at high feed rates, but the rates are higher than I'd try with a lathe like the 1030 anyway.  I have the math around that in my writeup too.  I'd expect you to hit that limit at 0.040" per turn at 1/5.2 of what I did, so something like 180 RPM.  If that's a problem, I'd change the stepper controller to use fewer pulses per revolution, maybe 300 or 500 rather than 1000 pulses per rev.  There's a couple of places to play with settings to trade off where one hits that 100 kHz.

SoManyHobbies, 2021-07-27:
Good point about setting MICROSTEPS_FEED to 52, I think I was hung up on it being an even multiple of the thread feed because I was actually defining it as
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS * 9, but makes more sense just to set it to 52 and put in a comment about where the number came from.

Yeah, I am really worried about running out of spindle RPM as well as the 100 kHz limit. I may actually decide to set the gearbox different between threading and power feed. If I max out the feed the gearbox can provide I can get up to 0.0115 in/rev, which puts the ratio at 5.43... Which isn't quite so bad, but you're right I may end up needing to drop the steps/rev count on the motor in order to support feeding at higher RPM even if I change the gearbox between the two operations.
I might even add some code to flash the gearbox setting on the display for a few seconds when changing from feed to thread and vice-versa.

Initially I came up with the ratio just using the threading/feed chart on the lathe, but I did validate with and indicator and manually rotating the spindle and measuring the movement.

SoManyHobbies, 2021-07-28:
Thank dsgroninger for the help. I think using the existing mechanism might work well enough even for odd ball gear boxes like the one on my PM 1228.

But one suggestion I might have would be to possibly include the option to just set a floating point value that is a ratio of power feed to threading feed. So we could plug in values like 8.6805, and assume it is then a multiplier to the existing feed rate. Then if we carry that value until as late as possible in the calculation it might help minimize rounding errors.
Note:
- This assumes we are in the #ifdef USE_FLOATING_POINT case
- This would replace STEPPER_MICROSTEPS_FEED and STEPPER_RESOLUTION_FEED. I dunno which one would be more useful to others.

Seems like the feedRatio() function in Core.h might be a good candidate since it's already treating its product as a float?

But I don't fully understand the math well enough to know if it would work or not. Or if it's worth keeping this issue open for such a change. I didn't intend to hijack the OPs issue if the original issue has been resolved.

BL-Shopwork, 2021-08-03:
Solved with code edit to change feed rate Microsteps, hardware feedstop used.

closed 2021-08-03 by BL-Shopwork