clough42/electronic-leadscrew #168: Did I fry my Launchpad

IamTomzo, 2021-07-24:
Greetings - I apologize if Github is not the right place for this sort of inquiry.   I can't find the right forum for Clough42 ELS communications.

I got everything soldered up, did the continuity checks, and connected it to my Win10 PC.   The device was recognized, but after I loaded the code it stopped being recognized.  I tried disconnecting the Clough42 board and reconnecting the jumpers to use just USB power to see if the board was recognized but no luck.

I was pretty careful about static, although I must admit that I was not using my antistatic mat and wristbands - even though James said he fried one - duh.

Anyway, if anyone has any tips about these boards not being recognized I am all ears.

JonWoellhaf, 2021-07-24:
When you plug the USB connector into the TI board, does the computer beep? Are you sure the other end of the cable is still plugged into the computer?

IamTomzo, 2021-07-24:
It recognized the board at first and had the requisite beep.  The code seemed to load - it went through all the steps.    I have done a lot of arduino projects so I am familiar with editing config.h files and the like.   I tried different cables, checked the device manager, etc, but windows says there is a hardware problem.   I failed to take proper static precautions (stupidly) so it looks like TI is going to get another $30 from me...

clough42, 2021-07-24:
When I fried mine, it was a connection issue with the power pins, causing
5VDC to end up on the 3.3V power rail.

I would suggest checking all of your support joints and testing the boost
interface board by itself, not connected to the new LaunchPad to make sure
the 5 and 3.3v supply pins are correct before you begin.

Also, remember that the three power jumpers on the LaunchPad board must be
in place of you're just connecting USB for testing, and off if you're
providing power through the booster pack board.

If you are powering through the booster pack, you can still program through
USB with the power jumpers off.

James

IamTomzo, 2021-07-24:
Thanks James (and BTW this project and your channel area awesome).  I did take off the power jumpers before sending 5V through your board although to test things out I also removed your board and reinstalled them to see if it would work.  No change.

I got both of the lights (3.3 and 5V) on your board and all the continuity checks were good.   Are you saying I should measure the voltage as well?  If there was a solder bridge across the 3.3 and 5V pins on your board would both lights still light up?

I did notice one LED on the TI board that was on before it fried and then was off.  It was either LED4 or LED5 (I did not note this when it was on but noticed it was off when things were not working).  Reading the documentation from TI those are not power LEDs but are related to GPIO 23 and 34 and the only pin that this project uses was 34 (if I am reading the documentation correctly).   I am spitballing here so forgive me - I should have taken more careful notes when I was doing all of this but excitement got in the way.   When I powered it up the PC recognized it and connected.   After I loaded the code is when it failed.   Could I have wired something wrong so that the code sent power the wrong way?  

I have another TI board on the way - they are cheap and TI does 2 day shipping, so my weekend will be freed up for some honeydos (which will please my wife) so all is not lost.   Heck - I don't even own a lathe (yet) but love projects like this.  I figured that when I do get my lathe it will either be off Craigslist or a Chinese import - either way it is getting torn down for cleaning, etc and I want to install this as things go back together.

Thanks for all you do!

clough42, 2021-07-28:
Yeah, I would definitely check the voltages of the 5v and 3.3v supplies
with the boost board disconnected from the launchpad.  If there is a bridge
or a damaged regulator, the 3.3v might output 5v, and that would kill the
launchpad.  The 3.3v led would still light, but would be brighter.

Also be sure the three power jumpers are of the launchpad before powering
it through the boost board.

James

IamTomzo, 2021-07-28:
[image]
Thanks James - I powered up your board and measured 5.21V and 3.34V across the 5V and 3.3V pins (red oval in this image) using the ground pin in the green circle so that seems OK.  The pins on the display connector are reading 5.21V.  Are there other pins to measure or does this mean the board is OK?

Of course, it could be that my observations earlier were flawed and I killed the board through a lack of static protection.   The $30 for the new board is not as bad as having to wait to get the next one, so I don't really want to brick another one...

IamTomzo, 2021-07-28:
And by the way, I realize this is not your day job and you have better things to do with your time.   My replacement launchpad should arrive today so I will give it another go - with better safeguards for static and better control/documentation of conditions - and see how that goes.

clough42, 2021-07-28:
That looks good to me.

James

IamTomzo, 2021-07-28:
Thanks.   Again, I appreciate what you do for us hobbyists.   BTW, you have also inspired me to drop some of the COVID weight that added to the other extra weight I already had when it started.  No fads - just watching calories and moving more.  Down about 15 pounds so far with another 40 or so to go!

clough42, 2021-07-28:
Awesome!  It's all about making a choice and taking control.

James

IamTomzo, 2021-08-02:
It's alive!!  Finally managed to get this done properly without letting any of the magic smoke out.   I am converting an old desktop PC case into an ELS electronics enclosure.   Still need to finish all my cables, but the tips on the wire termination tools was super helpful.  I am using the StepperOnline integrated servo, so I need to figure out the precise settings on that, but seeing that I don't quite have a lathe yet, I am going to do all the final software configuration once I know my reductions, etc.   I am leaning toward the G0602 but the ire from my wife for dropping $2500 on another one of my "toys" is the main obstacle.   She already has to deal with ~$10K worth of gear in my observatory....
[image]

IamTomzo, 2021-08-07:
Well, since this last post I took the project from the desktop and installed it in my enclosure.   I think I failed to set the boost board properly - it appears to have been set on only one row of the pins which could move 5V onto the 3V section.   Result - another dead TI board - UGH.

As I waited for my new board to arrive (oddly TI would not let me add a third one in two weeks to my cart so I had to buy from Mouser), I rechecked all the continuity tests and tested the cables through the enclosure all the way to each device.   All were in the right places.   However, I still have something wrong going on.

I carefully removed the jumpers and set the switches in full anti static mode (anti static mat, wristband, etc).   I then made sure the boost board was set properly this time and went to load the firmware.   I used the TI supplied USB cable and just a light in the area right next to the USB connection lit up which is what was expected.   I then plugged in 5V power to the boost board and nothing.   The TI board was not recognized by the PC.   I then took off the boost board, reset the 5V jumpers so I had just a bare TI board and connected via USB and presto - the board comes alive and can be flashed.   This is odd because it is the same boost board used in the image above.

Anyway, after I flashed the TI board I took it back out and installed it into my enclosure with the boost board.   When I powered everything up the display flickered a bit followed by a POP and the familiar smell.   I immediately powered down and upon inspection the TM1638 chip had fried.  This is the same display board pictured above, so it had worked previously.   I stripped the TI board down, reset the 5V jumpers and it still connects to the PC, so it does not look like I fried that board (again).

I have another of the LED&KEY boards on order but was wondering if any of you have any thoughts on (a) why the TI board won't connect to the PC when using power from the boost board and (b) what error on my part would cause me to fry the display?

Thanks

Tom

IamTomzo, 2021-08-10:
OK - I got the new LED&Key board and the one I received this time was clearly made better than the last two - even though I got them from the same source.   Odd.   I also noticed that I fried the voltage regulator on the boost board so I ordered a 10 pack of those and replaced it successfully.   I have everything in the enclosure and it powers up fine to the Clough boot screen but I am getting nothing from my encoder as RPM on the display.  Consequently, the servo does not spin of course.   I tested the encoder with an ohmmeter (black/blue, blue/white) and they read the same so I am pretty sure the encoder is OK.  I flipped the switches back and forth a few times to make sure they were set correctly on the TI board.

I also tried both of the encoder connections EQEP 1 and 2 (and reflashed the code accordingly) same result.   I do not have an oscilloscope to do more detailed tests on the board, but in reading a closed post here on github from another user it may be the TI board (again).   Are there any good tests that can be done without a scope?  If there is a bad resistor can I just get one and solder it onto the board rather than buying a whole new board?

Also, on a side note, I used a laser thermometer to look at the temps on the boost board just to see what was going on there.   The only thing that stood out to me were resistors R7 and R8 which seemed warmer than the rest heating up to ~125F pretty quickly.   Not sure if this means anything but it is a data point.

Thanks

Tom

IamTomzo, 2021-08-19:
I got it figured out - and posted a video on my "need help wiring Clearpath Killer" issue.   As always I was the problem but want to thank all of you who helped me out.

closed 2021-08-19 by IamTomzo

clough42, 2021-08-19:
I saw your previous message saying R7 and R8 were getting warm.  Do you
know what was causing that?  They're connected to the gates of MOSFET
transistors, so they should only carry brief spikes of current at 5mA max
during level transitions.

James

IamTomzo, 2021-08-19:
I am not sure but it might be the fact that I had 12V going through there due to my stupidity.   When I was building my enclosure I bought a 12V and a 5V wall wart.  I guess I must have had another 12V on my workbench (I power stuff in my observatory on 12V regularly) and I actually printed a label that said 5V and put it on the wrong wall wart.   It got installed into the enclosure and I only realized it when I was troubleshooting the encoder and found 12V on pin A.   I am quite embarrassed by the mistake, but that goes for nearly all of my mistakes.

I just fired it up and checked with my laser thermometer and they are all running just a hair above ambient temperature.   I suppose that temperature increase should have made me check that power supply earlier, but hindsight is always 20-20.

The real question is when I get to buy a Clough42 T-shirt!   Thanks for all you do!