clough42/electronic-leadscrew #170: Non-integer related spindle/encoder relationship

EatSand, 2021-07-31:
I used different sized pulleys and an oddball encoder (500ppR), giving me ENCODER_RESOLUTION as 4444.4444444'. This fails the sanity check due to being non-integral, but being a 'clever' sort, I commented that out and it seems to work perfectly...until it doesn't - I'll be cutting a thread and it will suddenly jump into the wrong place. Can anyone confirm if this is an unacceptable config and needs to be integer related?

clough42, 2021-07-31:
I needs to be integral.  The value you're entering is converted to an
integer and the extra fraction is discarded by the compiler.  You can run
with 4444 as the ratio, but there will be some error, and it will
accumulate as it runs, leading to messed-up threads.

James

EatSand, 2021-07-31:
Thanks, I assumed as much but c++ isn't my strong point. I have different pulleys and gears on order already.

Really appreciate the project - I'll put up some pictures once I've tidied it up! Completely transforms the lathe.

EatSand, 2021-08-01:
PS: I retract my prior comment about the jumping - I had a partially sheared leadscrew shear pin. New lathe teething!