clough42/electronic-leadscrew #171: Feature idea: report spindle angle

EatSand, 2021-07-31:
Having fitted an encoder to the spindle, it would be neat if the spindle angle could be reported to the display. With a live tool and a spindle lock (or some tolerance/faith...) this would give a convenient indexing capability for bolt hole circles. 

If I could link the spindle angle to my cross slide DRO I could convert polar to Cartesian and go way beyond simple bolt circles.

Late to the party, but I couldn't see this mentioned elsewhere...

johnsonm, 2023-04-03:
You aren't alone; this is also feature request #145

tomvikse, 2023-04-04:
> You aren't alone; this is also feature request #145

https://github.com/kwackers/electronic-leadscrew

"Added thread to shoulder, custom thread pitches and **angular position of spindle.**"

johnsonm, 2023-04-05:
Thank you! I hadn't found that yet when I commented. I've since found #231