clough42/electronic-leadscrew #173: Wiring the "Clearpath killer" -  iSV57T-180

IamTomzo, 2021-08-03:
Greetings,
Inspired by James' testing of the integrated servo that cost only $100 I bought one but can't seem to figure out exactly how to wire that motor up.  I realize that the enable pin will not be used, but I am not sure how to deal with the DIR + connection on the motor.  I have read through the documentation to the extent of my ability and tried to zoom in and see how James wired it in the video, but don't want to let any more magic smoke out as I complete this project.  He appears to have some of the pins jumped in the video but it is hard to tell.

The only possible matches I can find are:
STP -> PUL-
"+" -> PUL+
DIR- -> DIR-
ALM+ -> ALM+
ALM- -> ALM-

I may be way off here. Has anyone else wired this up (well, I know James did)?  If you would be so kind, could you give me some information on the correct way to wire this?  One image is from the manual showing the connections and the other is the Clough 42 board.

Thanks

Tom
[image]
.
[image]

IamTomzo, 2021-08-03:
Doing a bit more reading I see that the jumper that I see in James' video appears to be connecting PUL+ to DIR+ according to this chart from the manual for the motor.   Of course, it could also be going from PUL- to DIR-, depending on which of these diagrams is chosen.  I would just try each of them but don't want to fry anything.

Also, there are inline resistors in this diagram but I don't see anything like that in any of the documentation from the wiring schematics - perhaps those are already on the boost board.  I see some on the board and perhaps that is their function, but I am unsure.

I apologize for my glaring lack of knowledge in this area, but that is one reason I took on this project as by doing these things I always learn a few things.

[image]

BL-Shopwork, 2021-08-03:
+        -->   PUL+ and Dir+ (orange with red jumper in photo below)
STP-   -->   PUL - (White)
DIR-   -->   DIR- (Blue)

Not using Alarm
36v power supply for my setup.

[image]

I set my microstep dip switches S1,2,3 for 1600, did not change the stiffness (70).

IamTomzo, 2021-08-03:
Awesome - thanks!   That is just what I needed to know!  I may just wire up the ALM connectors anyway - I suppose it could not hurt.

IamTomzo, 2021-08-04:
Of course, even though I had the board flashed and working with the display, when I got it all into the enclosure it seems that I have fried another TI board.   SMH.   I sounded out all my connections again and all of the continuity checks come out perfect.   Not sure where I went wrong this time, but I have a third TI board on the way.

IamTomzo, 2021-08-19:
First off I want to thank all of you who took time out of your day to help me along the way.   As usual, the problems I had were entirely self inflicted, but three dead TI boards later I got it working on the fourth!  I also want to thank James for all his work on this very interesting project.   I don't even own a lathe yet (coming soon) but the project was too cool to pass up!   Here is a video showing how I cobbled it together in an old desktop PC case.

Thanks again!

Tom

https://youtu.be/JKiJtWTBcc0

closed 2021-08-19 by IamTomzo